esphome/issues #3878: GPIO pin states not being restored as expected

kbx81, 2022-12-06:
### The problem

I have [a config](https://github.com/kbx81/esphome-configs/blob/c7fc70e947a5b970cb42d76212a6cf11118dc948/dev/esp-sprinkler-nextion-controller.yaml#L865-L954) with a number of GPIO pins that have `restore_mode: ALWAYS_OFF` in the switch config. After the ESP resets, they are instead restored to the ON state.

Simple case of incorrect behavior. 😄

I did some backtracking and found that the issue began after [this PR](https://github.com/esphome/esphome/pull/3648) was merged.

### Which version of ESPHome has the issue?

2022.12.0-dev (Dec 6 2022, 01:09:32)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.5

### What platform are you using?

ESP32

nagyrobi, 2022-12-06:
Having the same issue

ssieb, 2022-12-06:
I know what the issue is and I'm working on a PR.

jpeletier, 2022-12-06:
@ssieb Thank you and my apologies for all. I am traveling and will be back on Friday to also take a deeper look.

closed 2022-12-06 by jesserockz