esphome/issues #3881: Deep Sleep constantly wakes up on its own when it went to sleep when no wifi

comet424, 2022-12-08:
### The problem

i having issues with deep sleep

if your connected to wifi and home assistant..  and it goes to sleep  it works fine and the wake up button works fine

but 

if your connected to wifi and home assistant  first.. and you leave  the wifi range and then the esp32  goes into deep sleep.. something glitchs and keeps re awaking it.... so say its set for 15 min..   15 min goes  it then restarts  so its i guessing going into deep sleep the instantly re awakes...    randomy it will goto sleep...  but the wake up pin will not work.. and then it randomly  will re start up again coming out of deep sleeep

once  it gets wifi again  it will connect to wifi and home assistant.. buttons i have seem to work again but isnt able to talk to home assistant properly.. you gotta cycle the power..    so it can connect to wifi to work.. but glitchs after it goes to sleep  at the time its no longer connected to wifi

but when its connected to wifi and it goes to deep sleep.. it doesnt have an issue and doesnt randomly wake up etc

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.0

### What platform are you using?

ESP32

### Board

esp32 wroom nodemcu

### Example YAML snippet

```yaml
deep_sleep:
  run_duration: 15min

######################
### Wake up Button ### 
######################
  wakeup_pin: 
    number: 25
    inverted: True
    mode: INPUT_PULLUP
#  esp32_ext1_wakeup:
#    pins: 
#      - 14
#      - 27
#      - 26
#      - 25
#    mode: any_high
    
################
### Switch's ###
################
binary_sensor:

##############
### Switch 1 #
##############
  - platform: gpio
    name: 'Vehicle1 Temp Switch #1'
    pin: 
      #number: 14
      number: 25
      mode:
        input: true
        pullup: true
```

### Anything in the logs that might be useful for us?

```txt
cant see logs as its in the car
```

comet424, 2022-12-10:
ive tried a couple esp32s now.. both experience the same issue.. if its connected to wifi  and goes to sleep.. it will stay asleep

but if it leaves wifi while on...  and when it activates deep sleep.. it instantly turns back on...  so i have it set for 15 min  and it will constantly restart.. the odd time  it will sleep longer  but then randomly wakes back up again without a button press..  so 2 esp32s  experience the same issues when no wifi 

the 2nd esp32s i not sure why it is? on the esp32s  when you first plug it in  the red and blue led on the board and turns on my external blue led i have too and it doesnt boot.. you have to press the one button on the esp32s  to get it to boot then the blue led is off... so not sure this happens when you first plug it in..  not sure if its some hanging issue in the code or not

not sure if i need the change the board type?  its a esp32s
and i have it set for
```
esp32:
  board: nodemcu-32s #esp-wrover-kit
  framework:
    type: arduino```

and 
serial logger doesnt work with ```ap:``` mode  using the ```web_server:```  be nice if that could be added  only  ota update and wifi setting  so i cant read the logs in the car when main wifi is not connected

ssieb, 2022-12-11:
I don't know what you mean about the serial logger not working with the web server.  The serial log is what you get from the USB port.  Maybe you mean that the web server doesn't work with ap mode?

comet424, 2022-12-12:
well i only using ur words "serial logger"  as the logger in home assisnant/esphome as  "usb" and the "wifi logs"

web_server: and ap:  when you goto the website all you get is the  OTA update and wifi options..  no logger

when i mentioned  when you first install esphome from the esphome website.. you get the dark/day mode and the logger on the right side..  

thats where i comparing it..  and called it a serial logger  cuz you mentioned  the word "serial logger"  figured thats ment for logger in wifi and connected to home assistant esphome

comet424, 2022-12-13:
oh forgot.. if you needed to know which  esp32s model i have  here is the amazon link if it helps at all
https://www.amazon.ca/gp/product/B086MJGFVV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

comet424, 2023-01-12:
so from what i found i fixed the reboot  i think  cuz of the mqtt   reboot_timeout: 0s

but the issue now is  Deep sleep never works if there is no wifi
looks like  esp32  is stuck in wifi searching and prohibits the Deep Sleep from ever running..    a check or multi process option

so it can search for wifi  and  deep sleep starts... so if your connected to wifi it starts the counter...  u leave the wifi area..  it shuts down.. .  u re awak the esp32  while out of the wifi range  it then stuck..  deep sleep never re intiates.

and an option so if you press a button  that would re set the deep sleep counter... as if i press a button and its close to the deep sleep  it shuts down on me when i about to press a second button

berkayozg, 2023-12-19:
Any solutions?