esphome/issues #3882: DHT don't work any more with dev branch

Philippe12, 2022-12-08:
### The problem

I try to use for test the dev branch and I can't communication with DTH (sonoff SI7021).

### Which version of ESPHome has the issue?

2023.1.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

sonoff thr316d

### Component causing the issue

DHT one wire

### Example YAML snippet

```yaml
sensor:
  - platform: dht
    pin: GPIO25
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
I configured a standard DHT sensor.
The communication work perfectly with 2022.11.5 but the I use the 2023.1.0-dev no more communication.

I look the signal with oscilosop and I can see the first pulse when the GPIO is in output (the timing look like good) but after nothing. I hop that the GPIO not switch in input mode.
```

### Additional information

I can do some test.

Philippe12, 2022-12-09:
I thing that I find the problem.
The gpio pilot function take more time than older version, and the Sonoff SI7021 timing is strick.
For it I must to modify the code of esphome/components/dht/dht.cpp like this (line 86)  :
``` c
  if (this->model_ == DHT_MODEL_DHT11) {
    delayMicroseconds(18000);
  } else if (this->model_ == DHT_MODEL_SI7021) {
    delayMicroseconds(360);
    //this->pin_->digital_write(true);
    //delayMicroseconds(40);
  } else if (this->model_ == DHT_MODEL_DHT22_TYPE2) {
    delayMicroseconds(2000);
  } else if (this->model_ == DHT_MODEL_AM2302) {
    delayMicroseconds(1000);
  } else {
    delayMicroseconds(800);
  }
```

Philippe12, 2022-12-14:
same way with 2022.12.0

paulius2k, 2022-12-16:
I have the same issue on ESP32 with Sonoff Si7021 sensor. 
I get this in my logs, irrespective if I use the pull-up resistor or not:
```
[16:15:38][W][dht:169]: Requesting data from DHT failed!
[16:15:38][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

I upgraded ESPHome to `v2022.12.1` - same issue there.

However, it **does work fine** on the Sonoff TH16 hardware (`board: esp8285`) flashed with ESPHome `v2022.12.0`.

Philippe12, 2022-12-17:
I work on. But the timing look like treky.