esphome/issues #3883: Wemos d1 mini pro not booting in esphome

varadujagan, 2022-12-08:
### The problem

When I try to use Wemos d1 mini pro in esphome, Wemos d1 mini pro is not booting it is showing some error in the log, which I cont understand.

### Which version of ESPHome has the issue?

2022.11.5 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.5 

### What platform are you using?

ESP8266

### Board

Wemos d1 mini pro

### Example YAML snippet

```yaml
esphome:
  name: esp-eb

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "aVLsH8jfp0NAZQ8DY5dlmk0wcGomQl7SgGTJ0WuAo6M="

ota:
  password: "ea9f8000236dbc42be9ac14a4753d208"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Eb Fallback Hotspot"
    password: "LcoDGAvIYAwu"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
rl l��| �l�|�l�b|����r�b�b��nn�lnn���bp��lrlrlp�n��l��bn�|���b��nn� l��l`�nn�lnr���nrr�`p�n�r������bn�|�b��nn� �l`�nn�lnr���nrl`r��nrl`�쎎�l`��n�l[I][logger:258]: Log initialized
[C][ota:469]: There have been 5 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: D8:BF:C0:FD:25:1A
[D][wifi:386]: Starting scan...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40257f83 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[... 1021 lines cut ...]
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00074f80
~ld
```

### Additional information

I didn’t edit anything in the code. it came out when I create a new device by giving the device name and board type.

Note:
What I tried:

Tried with different Wemos d1 mini pro (i tried 2 Wemos d1 mini pro with the same code)
no use same error in both devices.
I tried different both ESPhome flasher and ESPhome web Tools to install the code. I think everything installing the code properly, the problem is in booting only.
I change the board type to “d1_mini” and installed it in Wemos d1 mini. no problem with D1 Min working properly.

leodrivera, 2022-12-08:
It seems a problem with platformio:
https://github.com/platformio/platform-espressif8266/issues/281

cucamacuca, 2022-12-17:
I have the same issue. Installing tasmota works fine, but not esphome.

varadujagan, 2022-12-18:
> I have the same issue. Installing tasmota works fine, but not esphome.

yes for me also works with tesmota, but not in esphome. if u got any solution please share me.

[image]

raphael303, 2022-12-18:
Same problem here. D1 mini pro ESP8266. I have not change anything in the initial yaml:

```
`esphome:
  name: espd1test

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "WFVROpHyI6GF1gxpjPn50g6LHK+nhf2flOmYZHJp+Y0="

ota:
  password: "0a49bb5de73a59513b052dbbce24022a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espd1Test Fallback Hotspot"
    password: "VWomHeBo0tEE"

captive_portal:
    `
```

I get this in the log on esphome in google chrome.:

```
`r$l����$�|�l�#|�����|b�b��on�$oN���cp��l{lslp�n��l��bn�|���#�|~�o�l��$`�Nn�s$N{���osr�`p�o�r��ܜ��"N�|�c��on��$`�no�{lnr���n{l r��n{l �䎐�܎l`��N�l[I][logger:258]: Log initialized
[C][ota:469]: There have been 3 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: 48:3F:DA:60:18:03
[D][wifi:386]: Starting scan...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00074f70
~ld`
```

amon-ra, 2023-02-04:
You cold try:
```
`esphome:
  name: espd1test

esp8266:
  board: d1_mini_pro
  framework:
    version: 3.1.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "WFVROpHyI6GF1gxpjPn50g6LHK+nhf2flOmYZHJp+Y0="

ota:
  password: "0a49bb5de73a59513b052dbbce24022a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espd1Test Fallback Hotspot"
    password: "VWomHeBo0tEE"

captive_portal:
```