esphome/issues #3886: DSMR: I only get "Missing unit" errors on a T211 DSMR5 meter running on a SlimmeLezer+

iruoy, 2022-12-09:
### The problem

The only values I can read out of my new SlimmeLezer are the "ESPHome Version" and "SlimmeLezer IP Address" fields (and other SlimmeLezer * fields). All the others stay at NA. And I get the same error every second saying "Missing unit".

[image]

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.21.1

### What platform are you using?

ESP8266

### Board

SlimmeLezer+

### Component causing the issue

DSMR

### Example YAML snippet

```yaml
substitutions:
  name: slimmelezer
packages:
  zuidwijk.slimmelezer: github://zuidwijk/dsmr/slimmelezer.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
17:23:13 [E] [dsmr:265] 0-1:24.2.1(632525252525W)(00000.000)
                                                           ^Missing unit

17:23:14 [E] [dsmr:265] 0-1:24.2.1(632525252525W)(00000.000)
                                                           ^Missing unit

17:23:15 [E] [dsmr:265] 0-1:24.2.1(632525252525W)(00000.000)
                                                           ^Missing unit

17:23:16 [E] [dsmr:265] 0-1:24.2.1(632525252525W)(00000.000)
                                                           ^Missing unit

17:23:17 [E] [dsmr:265]	0-1:24.2.1(632525252525W)(00000.000)
                                                           ^Missing unit
```

opaulzen, 2023-01-26:
Dit you ever manage to solve this issue? I have exaclty the same thing.

RoadXY, 2023-02-22:
I have the same issue.
`[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:96.1.1(4530303632303030303130353334303232)\r\n"
[D][uart_debug:158]: <<< "0-0:96.14.0(0001)\r\n"
[D][uart_debug:158]: <<< "1-0:99.97.0(0)(0-0:96.7.19)\r\n"
[D][uart_debug:158]: <<< "1-0:52.36.0(00000)\r\n"
[D][uart_debug:158]: <<< "1-0:31.7.0(002*A)\r\n"
[D][uart_debug:158]: <<< "1-0:61.7.0(00.000*kW)\r\n"
[D][uart_debug:158]: <<< "0-1:96.1.0(0000000000000000000000000000000000)\r\n"
[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(221006204622S)\r\n"
[D][uart_debug:158]: <<< "0-0:96.1.1(4530303632303030303130353334303232)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.1(000003.270*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.1(000000.747*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "0-0:96.14.0(0001)\r\n"
[D][uart_debug:158]: <<< "1-0:32.32.0(00002)\r\n"
[D][uart_debug:158]: <<< "1-0:52.36.0(00000)\r\n"
[D][uart_debug:158]: <<< "1-0:21.7.0(00.355*kW)\r\n"
[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(221006204623S)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:32.32.0(00002)\r\n"
[D][uart_debug:158]: <<< "1-0:52.36.0(00000)\r\n"
[D][uart_debug:158]: <<< "1-0:72.7.0(220.0*V)\r\n"
[D][uart_debug:158]: <<< "1-0:41.7.0(00.181*kW)\r\n"
[D][uart_debug:158]: <<< "0-1:96.1.0(0000000000000000000000000000000000)\r\n"
[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(221006204624S)\r\n"
[D][uart_debug:158]: <<< "0-0:96.1.1(4530303632303030303130353334303232)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.1(000003.271*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.1(000000.747*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "0-0:96.14.0(0001)\r\n"
[D][uart_debug:158]: <<< "1-0:1.7.0(00.552*kW)\r\n"
[D][uart_debug:158]: <<< "0-0:96.7.9(00003)\r\n"
[D][uart_debug:158]: <<< "1-0:72.36.0(00000)\r\n"
[D][uart_debug:158]: <<< "1-0:72.7.0(220.0*V)\r\n"
[D][uart_debug:158]: <<< "1-0:42.7.0(00.000*kW)\r\n"
[D][uart_debug:158]: <<< "0-1:24.2.1(632525252525S)(00000.000)\r\n"
[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(221006204625S)\r\n"
[D][uart_debug:158]: <<< "0-0:96.1.1(4530303632303030303130353334303232)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.1(000003.271*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.1(000000.747*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "0-0:96.14.0(0001)\r\n"
[D][uart_debug:158]: <<< "0-0:96.7.9(00003)\r\n"
[D][uart_debug:158]: <<< "1-0:32.36.0(00000)\r\n"
[D][uart_debug:158]: <<< "1-0:72.7.0(220.0*V)\r\n"
[D][uart_debug:158]: <<< "1-0:41.7.0(00.159*kW)\r\n"
[D][uart_debug:158]: <<< "0-1:24.1.0(003)\r\n"
[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(221006204626S)\r\n"
[D][uart_debug:158]: <<< "0-0:96.1.1(4530303632303030303130353334303232)\r\n"
[... 219 lines cut ...]
[D][uart_debug:158]: <<< "1-0:52.32.0(00002)\r\n"
[D][uart_debug:158]: <<< "0-0:96.13.0()\r\n"
[D][uart_debug:158]: <<< "1-0:31.7.0(002*A)\r\n"
[D][uart_debug:158]: <<< "1-0:42.7.0(00.000*kW)\r\n"
[D][uart_debug:158]: <<< "!5D3F\r\n"
[D][uart_debug:158]: <<< "/Ene5\\T211 ESMR 5.0\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(221006204643S)\r\n"
[D][uart_debug:158]: <<< "0-0:96.1.1(4530303632303030303130353334303232)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.1(000003.274*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.1(000000.747*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "0-0:96.14.0(0001)\r\n"
[D][uart_debug:158]: <<< "1-0:1.7.0(00.546*kW)\r\n"
[D][uart_debug:158]: <<< "1-0:2.7.0(00.000*kW)\r\n"
[D][uart_debug:158]: <<< "0-0:96.7.21(00009)\r\n"
[D][uart_debug:158]: <<< "1-0:32.36.0(00000)\r\n"
[D][uart_debug:158]: <<< "1-0:32.7.0(220.0*V)\r\n"
[D][uart_debug:158]: <<< "1-0:71.7.0(000*A)\r\n"
[D][uart_debug:158]: <<< "1-0:42.7.0(00.000*kW)\r\n"
[E][dsmr:265]: 0-1:24.2.1(632525252525S)(00000.000)
                                   ^
Missing unit`

RTVRTV, 2023-02-25:
I have te same isue

CK0y0TE, 2023-03-07:
I got the same after the meter was replaced for a 3 phase version.

opaulzen, 2023-03-07:
Apparently you have to wait a few days fot the new smart meter to be configured by your provider. It took 7 days before mine started working. The slimmemeter code does not like the values not increasing,  hence the error.

CK0y0TE, 2023-03-07:
thx for your lightning fast response!

What I also noted is the "wM-BUS” printed on the new meter and the note from marcel ’s webshop on my smartmeter that my version doesnt work with “m-bus”… ugh. 
Maybe time to upgrade to an ethernet version that also does m-bus. 
But, I ll wait a week to see what happens.

Cheers,
Camiel

RoadXY, 2023-03-07:
> Apparently you have to wait a few days fot the new smart meter to be configured by your provider. It took 7 days before mine started working. The slimmemeter code does not like the values not increasing, hence the error.

Yup, same for me. After a few days it started working

jeffreykog, 2023-04-24:
I'm having the same issue. Only difference is that i have had my new meter for a month, and the service provider does not seem to be able to get my gas meter linked. That also means that i'm not able to read my meter values using esphome, even though most of it is in the payload.

It also fails on this line:
```
0-1:24.2.1(632525252525S)(00000.000)
          ^
Missing unit
```

jeffreykog, 2023-04-25:
As per the DSMR spec, this value type should include a unit. It is the smart meter not complying to this, possibly due to the gas meter simply not being linked, returning an empty value without a unit as a result. The error is thrown by the DSMR parsing library esphome uses, as can be seen here: https://github.com/glmnet/arduino-dsmr/blob/master/src/dsmr/parser.h#L220

The value on which the error is returned (`0-1:24.2.1`) technically should always return m3 as unit. The long sequence of numbers is the ID of the gas meter itself. The string `632525252525S` looks like a default identifier for an unknown gas meter. Another DSMR reader I manage at another location does read it correctly:

```"0-1:24.2.1(xxxx25070000S)(14336.324*m3)\r\n"```

The DSMR parser library linked above searches for the star and checks after it what the value is, but it is missing and it returns an error. Most likely, the upstream parser library needs to be changed in order to allow values without an unit. Maybe even let it detect the `632525252525S` identifier as an unlinked gas meter, and handle that gracefully. It wouldn't even surprise me that this error causes a gasless home to be unable to use esphome DSMR at all.

Elangco, 2023-04-25:
Thank you for this very clear and elaborate answer.

For gasless home owners: don't worry, if you leave the gas_delivered out of the yaml, you can work with the electricity data without a problem.

SamuraiCoder, 2024-02-21:
Thanks to @Elangco I was able to fix it by just commenting the `gas_delivered` but any idea when this will be solved?