esphome/issues #3887: Lambda in http_request.post body resets ESP32 (same lambda with manual HTTPClient request works)

manuelsteiner, 2022-12-09:
### The problem

Hi,

when using a somewhat elaborate lambda snippet for a `http_request.post` snippet, the ESP32 resets itself before/while executing the request.

When I just a plain lambda and create the HTTP request manually with an `HTTPClient` instance, the request works fine.

I am not sure but it might be relevant that I am calling the requests from a script with quite a few parameters.

I am not sure how to correctly debug this situation. Any hints as to the steps to successfully get to the root cause would be appreciated. I am happy to further investigate this issue.

### Which version of ESPHome has the issue?

2022.12.0b2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

adafruit_qtpy_esp32s2

### Component causing the issue

http_request

### Example YAML snippet

```yaml
Not working with `http_request`.

- id: influxdb_write
  parameters:
    influxdb_url: string
    influxdb_token: string
    influxdb_organization: string
    influxdb_bucket: string
    influxdb_measurement: string
    influxdb_tag_names: string[]
    influxdb_tag_values: string[]
    influxdb_float_field_names: string[]
    influxdb_float_field_values: float[]
    influxdb_int_field_names: string[]
    influxdb_int_field_values: int[]
    influxdb_unsigned_int_field_names: string[]
    influxdb_unsigned_int_field_values: int[]
    influxdb_bool_field_names: string[]
    influxdb_bool_field_values: bool[]
    influxdb_string_field_names: string[]
    influxdb_string_field_values: string[]
    influxdb_timestamp_provided: bool
    influxdb_timestamp: int
    influxdb_precision: string
  then:
    if:
      condition:
        wifi.connected:
      then:
        - http_request.post: 
            url: !lambda |-
                std::string url;
    
                url.append(influxdb_url);
                url.append("/api/v2/write");
[... 176 lines cut ...]
        
            if(influxdb_timestamp_provided) {
                body.append(" ");
                body.append(std::to_string(influxdb_timestamp));
            }

            WiFiClient client;
            HTTPClient http;

            http.begin(client, url.c_str());
            http.addHeader("Authorization", auth.c_str());
            int httpCode = http.POST(body.c_str());
            http.end();

            if(httpCode != 204) {
                ESP_LOGI("influxdb_write", ("HTTP Request failed; URL: " + url + "; Code: " + std::to_string(httpCode)).c_str());
            }
```
```

### Anything in the logs that might be useful for us?

```txt
No logs related to the http request, the ESP32 resets with cause `Software Reset Digital Core`
```

ferrettinico, 2023-02-21:
I'm experiencing the same issues

nliviu, 2024-10-30:
I think I know what causes the device to reset.
The strings url, auth and body are local variables and you return the `str.c_str()` which is not valid anymore after exiting the `lambda`.
Make them `static` and it will work.
Also pay attention to http_request's `buffer_size_tx`.