esphome/issues #3890: Upon some reboots, BMP280 sensor randomly gives randomly offset readings several times an hour

melyux, 2022-12-10:
### The problem

I didn't have this problem the first time I started ESPHome with my BMP280. But after a few days, I restarted it. This time, I was met with seemingly slightly offset readings multiples times an hour at seemingly random intervals. This goes for both pressure and temperature

Pressure: 
[image]

Temperature: 
[image]

* I tried with other BMP280 units but all produce the same result (on some boots).
* The problem persists regardless of oversampling values or not.

I'm wondering if this is the same problem that the BME280 had with #3383 that was fixed by @dbaarda in #3615 for the BME280. I don't know if the register logic is the same between these sister boards so I haven't tried replicating the changes to `bmp280.cpp` out of caution.

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

ESP8285

### Component causing the issue

bmp280

### Example YAML snippet

```yaml
esphome:
  name: xxxxxxx

esp8266:
  board: esp8285

# Enable logging
logger:

ota:
  password: "xxxxxxx"

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxxxxx"
    password: "xxxxxxx"

captive_portal:

mqtt:
  broker: xxxxxxx
  username: xxxxxxx
  password: xxxxxxx
  discovery: false
  topic_prefix: xxxxxxx

i2c:
  - id: bmp_i2c
    sda: GPIO4
    scl: GPIO5

sensor:

  - platform: bmp280
    address: 0x76 # Have to set this for some reason...
    temperature:
      name: "BMP280 Temperature"
      oversampling: 16x
      accuracy_decimals: 2
      filters: # Convert to Fahrenheit
        - multiply: 1.8
        - offset: 32.0
      unit_of_measurement: °F
    pressure:
      name: "Pressure"
      oversampling: 16x
      accuracy_decimals: 5
      filters:
        - multiply: 0.02952998330101 # Convert from hPa to inHg
      unit_of_measurement: inHg
    update_interval: 60s

switch:

  - platform: restart
    name: "Restart"
```

### Anything in the logs that might be useful for us?

```txt
Logs are similar in both cases, only the readings differ
```

melyux, 2022-12-11:
I found that the BMP280 uses the same registers and commands for this, and the BMP280 API from Bosch on GitHub does the same soft reset thing that BME280 uses. (Actually, that BMP280 API seems to be deprecated and replaced by a new API on GitHub from Bosch, but that API doesn't do this soft reset.)

I made the analogous changes to `bmp280.c` that @dbaarda did for the BME. Let's cross our fingers

melyux, 2022-12-11:
No, that doesn't seem to have solved it... 
[image]

dbaarda, 2022-12-12:
Note when I was debugging the bme280 I had a bunch of theories about what might be the cause and added logging to check them (all my theories were wrong), which might be worth doing here too.

The behaviour here seems a bit different to what I was seeing; when the BME280 gave a bad reading it got "stuck" and stayed like that until a power cycle.

In your case the bad readings seem transient and typically only give a bad reading for one cycle. This to me suggests a different mechanism to what I was seeing. I would suspect some kind of racy init or read implementation that is not correctly waiting long enough or checking the status is right. Note the reference implementations seem to include a hell-of-a-lot of repeated retries for just about everything, which makes me suspect they are not entirely reliable, and you always need to check and retry.

melyux, 2022-12-12:
Yeah, was about to report that the same fix didn't work. (Though it should probably be PR'd anyway to solve the separate deep sleep issue, right?)

I'm not sure about how ESPHome deals with that kind of stuff; can we block to do these retries on every read? My first thought was that it was getting interrupted in the middle of a read by a hardware interrupt.

dbaarda, 2022-12-12:
In my fix for `setup()` I poll `BME280_STATUS_IM_UPDATE` up to 5 times every 2ms waiting for the NVM to load. I remember agonising a bit over whether this would block other stuff from running and not being sure if the `delay()` and/or i2c `read_byte()` or `write_byte()` would let other "processes" run (I don't think they do), but I remember seeing other stuff that did it so I guess it's OK. I did consider trying to do the right thing and using `this->set_timeout()` to schedule callbacks, but it seemed like this was not best-practice and I was worried it might cause other problems with things like reads happening before `setup()` was completed.

I kept my changes to a bare minimum, but I do remember thinking that `update()` which reads everything didn't really adhere to the reference implementation much either. I think there are status registers you can read which will tell you when the data is ready to read, and the `update()` implementation just waits what should be long enough to do the read.

When I was debugging I dismissed insufficient wait as a cause of the problem because my logging showed we were actually waiting heaps longer than required, and the status registers were indicating the data was ready even when it was wrong.

melyux, 2022-12-12:
Maybe the deep sleep problem is caused by the `setup()` not waiting for NVM and you fixed it, and this is caused by the `update()` not adhering to the reference implementation. I do have 2 other sensors connected to the same ESP so maybe that exacerbates it. Did you ever get draft code written for the `update()` fix? I can test to see if it helps here.

melyux, 2022-12-12:
Also just saw this sector, where every anomalous value (except one) happened exactly 13 minutes after the previous one. Happened 6 or 7 times in a row. Weird

[image]

dbaarda, 2022-12-14:
I never actually attempted to fix `update()`, I just added logging to show the state of the registers and the raw adc values when the read was done. This confirmed that all the status registers were indicating the value should be good, just the raw adc value was wrong. My logging in `setup()` also showed that the NVM values were actually loaded fine and unchanged even before my fix. The only conclusion I could draw is that there must sometimes be some kind of power-grey-out during sleep that messed with some kind of internal state. Adding the soft reset to `setup()` seems to repair whatever internal state that got messed up.

dbaarda, 2022-12-14:
Oh yeah, 13mins is a prime... interesting. Often periodic glitches like this are the product of two different periodic factors aligning and interacting, but then the interval is typically the product of two or more primes. That the interval is a prime kind-of suggests there is no interaction, just one periodic cause.

Are you sure there are no error logs? In my debugging I also saw some occasional "failed to..." logs.

Also, are you using deep-sleep? I'm not sure but I think the problem only showed for me when I was using deep_sleep.

melyux, 2022-12-14:
I don't see anything anomalous in the logs, but I also don't have it at the `DEBUG` level. Let me turn it on and see if I get anything. I'm not using deep sleep, just your regular ESPHome setup. Unless I'm using it without knowing, I don't think so though. Config is above.

Do you still have the logging to show the register states? That would be very useful.

Your fix was magic for that problem. I ported it to the BMP280 and am planning to do a PR to add it here too (since the BMP280 reference implementation does the same thing), once I figure this other problem out.

I've continued to see bad values every 13 minutes, but occasionally it changes to a different interval (seemingly random) but always returns to one bad value every 13 minutes.

melyux, 2022-12-14:
VERY_VERBOSE log from a normal value (:36), bad value (:37), normal value (:38) sequence.

```
[20:36:16][VV][scheduler:195]: Running interval 'update' with interval=60000 last_execution=2388813 (now=2448813)
[20:36:16][V][bmp280.sensor:151]: Sending conversion request...
[20:36:16][VV][i2c.arduino:140]: 0x76 TX F4B5
[20:36:16][VV][scheduler:026]: set_timeout(name='data', timeout=76)
[20:36:16][VV][scheduler:195]: Running timeout 'data' with interval=76 last_execution=2448824 (now=2448905)
[20:36:16][VV][i2c.arduino:140]: 0x76 TX FA
[20:36:16][VV][i2c.arduino:116]: 0x76 RX 82A730
[20:36:16][VV][i2c.arduino:140]: 0x76 TX F7
[20:36:16][VV][i2c.arduino:116]: 0x76 RX 650990
[20:36:16][D][bmp280.sensor:175]: Got temperature=26.5°C pressure=1008.9hPa
[20:36:16][V][sensor:076]: 'BMP280 Temperature': Received new state 26.530001
[20:36:16][VV][sensor.filter:014]: Filter(0x3fff1cbc)::input(26.530001)
[20:36:16][VV][sensor.filter:024]: Filter(0x3fff1cbc)::output(47.754002) -> 0x3fff2194
[20:36:16][VV][sensor.filter:014]: Filter(0x3fff2194)::input(47.754002)
[20:36:16][VV][sensor.filter:021]: Filter(0x3fff2194)::output(79.753998) -> SENSOR
[20:36:16][D][sensor:126]: 'BMP280 Temperature': Sending state 79.75400 °F with 2 decimals of accuracy
[20:36:16][V][mqtt:414]: Publish(topic='sensor/indoor/sensor/bmp280_temperature/state' payload='79.75' retain=1)
[20:36:16][V][sensor:076]: 'Pressure': Received new state 1008.934753
[20:36:16][VV][sensor.filter:014]: Filter(0x3fff1fac)::input(1008.934753)
[20:36:16][VV][sensor.filter:021]: Filter(0x3fff1fac)::output(29.793827) -> SENSOR
[20:36:16][D][sensor:126]: 'Pressure': Sending state 29.79383 inHg with 5 decimals of accuracy
[20:36:16][V][mqtt:414]: Publish(topic='sensor/indoor/sensor/pressure/state' payload='29.79383' retain=1)
[20:36:16][V][component:200]: Component bmp280.sensor took a long time for an operation (0.11 s).
[20:36:16][V][component:201]: Components should block for at most 20-30ms.
[20:36:31][VV][scheduler:195]: Running interval '' with interval=60000 last_execution=2403199 (now=2463202)
[20:37:16][VV][scheduler:195]: Running interval 'update' with interval=60000 last_execution=2448813 (now=2508816)
[20:37:16][V][bmp280.sensor:151]: Sending conversion request...
[20:37:16][VV][i2c.arduino:140]: 0x76 TX F4B5
[20:37:16][VV][scheduler:026]: set_timeout(name='data', timeout=76)
[20:37:16][VV][scheduler:195]: Running timeout 'data' with interval=76 last_execution=2508827 (now=2508907)
[20:37:16][VV][i2c.arduino:140]: 0x76 TX FA
[20:37:16][VV][i2c.arduino:116]: 0x76 RX 829050
[20:37:16][VV][i2c.arduino:140]: 0x76 TX F7
[20:37:16][VV][i2c.arduino:116]: 0x76 RX 650990
[20:37:16][D][bmp280.sensor:175]: Got temperature=26.4°C pressure=1008.8hPa
[20:37:16][V][sensor:076]: 'BMP280 Temperature': Received new state 26.420000
[20:37:16][VV][sensor.filter:014]: Filter(0x3fff1cbc)::input(26.420000)
[20:37:16][VV][sensor.filter:024]: Filter(0x3fff1cbc)::output(47.556000) -> 0x3fff2194
[20:37:16][VV][sensor.filter:014]: Filter(0x3fff2194)::input(47.556000)
[20:37:16][VV][sensor.filter:021]: Filter(0x3fff2194)::output(79.556000) -> SENSOR
[20:37:16][D][sensor:126]: 'BMP280 Temperature': Sending state 79.55600 °F with 2 decimals of accuracy
[20:37:16][V][mqtt:414]: Publish(topic='sensor/indoor/sensor/bmp280_temperature/state' payload='79.56' retain=1)
[20:37:16][V][sensor:076]: 'Pressure': Received new state 1008.761719
[20:37:16][VV][sensor.filter:014]: Filter(0x3fff1fac)::input(1008.761719)
[20:37:16][VV][sensor.filter:021]: Filter(0x3fff1fac)::output(29.788717) -> SENSOR
[20:37:16][D][sensor:126]: 'Pressure': Sending state 29.78872 inHg with 5 decimals of accuracy
[20:37:16][V][mqtt:414]: Publish(topic='sensor/indoor/sensor/pressure/state' payload='29.78872' retain=1)
[20:37:16][V][component:200]: Component bmp280.sensor took a long time for an operation (0.12 s).
[20:37:16][V][component:201]: Components should block for at most 20-30ms.
[20:37:30][VV][scheduler:195]: Running interval '' with interval=60000 last_execution=2463199 (now=2523203)
[20:38:16][VV][scheduler:195]: Running interval 'update' with interval=60000 last_execution=2508813 (now=2568813)
[20:38:16][V][bmp280.sensor:151]: Sending conversion request...
[20:38:16][VV][i2c.arduino:140]: 0x76 TX F4B5
[20:38:16][VV][scheduler:026]: set_timeout(name='data', timeout=76)
[20:38:16][VV][scheduler:195]: Running timeout 'data' with interval=76 last_execution=2568824 (now=2568904)
[20:38:16][VV][i2c.arduino:140]: 0x76 TX FA
[20:38:16][VV][i2c.arduino:116]: 0x76 RX 829E40
[20:38:16][VV][i2c.arduino:140]: 0x76 TX F7
[20:38:16][VV][i2c.arduino:116]: 0x76 RX 650780
[20:38:16][D][bmp280.sensor:175]: Got temperature=26.5°C pressure=1008.9hPa
[20:38:16][V][sensor:076]: 'BMP280 Temperature': Received new state 26.490000
[20:38:16][VV][sensor.filter:014]: Filter(0x3fff1cbc)::input(26.490000)
[20:38:16][VV][sensor.filter:024]: Filter(0x3fff1cbc)::output(47.681999) -> 0x3fff2194
[20:38:16][VV][sensor.filter:014]: Filter(0x3fff2194)::input(47.681999)
[20:38:16][VV][sensor.filter:021]: Filter(0x3fff2194)::output(79.681999) -> SENSOR
[20:38:16][D][sensor:126]: 'BMP280 Temperature': Sending state 79.68200 °F with 2 decimals of accuracy
[20:38:16][V][mqtt:414]: Publish(topic='sensor/indoor/sensor/bmp280_temperature/state' payload='79.68' retain=1)
[20:38:16][V][sensor:076]: 'Pressure': Received new state 1008.923889
[20:38:16][VV][sensor.filter:014]: Filter(0x3fff1fac)::input(1008.923889)
[20:38:16][VV][sensor.filter:021]: Filter(0x3fff1fac)::output(29.793505) -> SENSOR
[20:38:16][D][sensor:126]: 'Pressure': Sending state 29.79350 inHg with 5 decimals of accuracy
[20:38:16][V][mqtt:414]: Publish(topic='sensor/indoor/sensor/pressure/state' payload='29.79350' retain=1)
[20:38:16][V][component:200]: Component bmp280.sensor took a long time for an operation (0.12 s).
[20:38:16][V][component:201]: Components should block for at most 20-30ms.
```

dbaarda, 2023-01-01:
I was just looking at your example and noticed that the "glitches" you are seeing are actually very small; only 0.1C and 0.1hPa in the logs.

I believe this is maybe in the order of the kinds of fluctuations you can see for transient things like doors-opening etc. I remember in the docs this was one of the reasons they suggested using oversampling and filtering to get a better "average" pressure. I've also seen fluctuations in temperature much higher than that due to CPU load near the sensor.

Are you sure it's not something environmental?

melyux, 2023-01-01:
I'm quite sure about that. The problem persists regardless of how still the environment is and the location. If it were just temperature spikes, I'd say it's some esoteric reading of the ESP heating up, but the temperature always spikes *down*, so that's probably not it, and the pressure also always spikes down. It seems more like a trim issue, or incomplete measurement. The spikes are occasionally quite large. Always downwards.

dbaarda, 2023-01-08:
Sorry, for the late followup; Christmas holidays etc.

Note that 0.1hPa is about the pressure change you get per meter of elevation, so it's about the pressure change you'd get from picking it up off the floor. The biggest drop in your graph is a bit larger at about 0.06inHg or 2hPa, or 20m elevation change. My point is these are small enough that things like an exhaust fan turning on in your bathroom down the hall or change in wind direction through an open window could trigger them. A simultaneous pressure and temperature drop are precisely what you would see for these kinds of environmental effects (https://sciencing.com/calculate-drop-due-pressure-drop-8739366.html).

The bme280 supports "oversampling" and an "IIR filter" to try and average and filter out these kinds of transient effects and noise. The data-sheet explicitly mentions "slamming doors" as a source of "short-term fluctuations" that the IIR filter for temperature and pressure helps compensate for. In https://github.com/rwaldron/johnny-five/issues/1274#issue-200767502 there is some discussion about oversampling settings seeming to fix things in another completely different driver.

So I'd recommend playing with any oversampling and filtering settings to see if they help. I notice in your logs that you appear to have some kind of extra ESPHome filter configured but I think it's just converting from C/hPa to F/inHg.

dbaarda, 2023-01-08:
Also note that in #3383 I was seeing a sudden huge 300hPa drop in pressure equivalent to 3km of elevation change, and it was persistent not transient. This was clearly something that couldn't be explained by environmental effects, unless you consider "it fell into my partial vacuum chamber" to be an environmental effect :-)

melyux, 2023-01-21:
I had the sensor in a completely empty building for many days with nothing running though... Some of the dips are quite sharp and really make the sensor pretty useless for detecting any kind of statistical changes, which is all the temperature sensor on this thing is really good for in the first place

I'm going to try two things based on your ideas.

1. I'm going to turn on the `iir_filter` option at 16x and see if that helps at all (oversampling is already at 16x but makes no difference whether that's on or off).
2. I'm going to place the device outdoors to see if theses semi-regular dips disappear (no doors being shut or HVAC starting and stopping outdoors, so the dips should either stop or become completely irregular).

Still interested in seeing the status registers you mentioned and checking them as well. I'll try to add that to the code to debug

dbaarda, 2023-01-22:
I realized I did check in the temporary logging I played with in a separate branch on my esphome fork.

It's the tst-3383-bme280-setup branch on my github fork you can pull from at https://github.com/dbaarda/esphome.git.

It's pretty ugly and was never intended for general consumption, but you might find it useful.

melyux, 2023-01-25:
I'll try that, thanks a bunch. I tested outdoors and IIR at 16x but saw no difference with either. So it's probably not environmental effects. After the last boot and bringing it back inside, the spikes became more frequent for some reason:

[image]

melyux, 2024-05-11:
This is suddenly working for both temperature and pressure after updating. Maybe it was [#6124](https://github.com/esphome/esphome/pull/6124). Works perfectly now

[image]

closed 2024-05-11 by melyux