esphome/issues #3891: BME680 BSEC - Wemos D1 - Missing Dependency

Jebsta, 2022-12-10:
### The problem

Can't compile a yaml config that worked some versions back.

Tried to use the config for another board (changed name etc.)

But the new device firmware won't compile (same board etc.)

Then I tried the old yaml config for the running board. Doesn't compile either.

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.1

### What platform are you using?

ESP8266

### Board

Wemos D1 mini Pro

### Component causing the issue

bme680_bsec

### Example YAML snippet

```yaml
esphome:
  name: bedroom-2-air-sensor
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom 2 Air Sensor"
    password: "wUYtdx4t9mor"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D1
  scl: D2
  scan: True
  id: id_i2c

bme680_bsec:
    address: 0x77
    sample_rate: ulp

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
      accuracy_decimals: 2
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/bedroom-2-air-sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bedroom-2-air-sensor (board: d1_mini_pro; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing boschsensortec/BSEC Software Library @ 1.6.1480
Warning! Please upgrade to the PlatformIO Core 6
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <Wire> 1.0
|-- <SPI> 1.0
Compiling /data/bedroom-2-air-sensor/.pioenvs/bedroom-2-air-sensor/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling /data/bedroom-2-air-sensor/.pioenvs/bedroom-2-air-sensor/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/bedroom-2-air-sensor/.pioenvs/bedroom-2-air-sensor/src/esphome/components/esp8266/core.cpp.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
src/esphome/components/bme680_bsec/bme680_bsec.h:12:10: fatal error: bsec.h: No such file or directory

**************************************************************
* Looking for bsec.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bsec.h"
* Web  > https://registry.platformio.org/search?q=header:bsec.h
*
**************************************************************

   12 | #include <bsec.h>
      |          ^~~~~~~~
compilation terminated.
*** [/data/bedroom-2-air-sensor/.pioenvs/bedroom-2-air-sensor/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
Compiling /data/bedroom-2-air-sensor/.pioenvs/bedroom-2-air-sensor/src/esphome/components/esp8266/gpio.cpp.o
```

### Additional information

I didn't change the Board or Yaml.
I wanted to copy the code and put it on another wemos but it doesn't compile. The Original Code won't compile either.
BSEC Version: 1.4.8.0 is currently on the running ESP (got Information out of logs)

enbeee, 2023-01-13:
I'm having the same issue.

oleksey, 2023-02-19:
Can confirm the issue. Took the minimal config example from [here](https://esphome.io/components/sensor/bme680_bsec.html) 

### Which version of ESPHome has the issue?
2023.2.2

### What type of installation are you using?
Docker

### What platform are you using?
ESP32

### Component causing the issue
bme680_bsec

### Example YAML snippet
Part before i2c was autogenerated by EspHome for BLE proxy
```
substitutions:
  name: asdfadfasdf
  wifi_ap_password: ''
esphome:
  name: asdfadfasdf
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: '1.0'
  build_path: .esphome/build/asdfadfasdf
  friendly_name: ''
  platformio_options: {}
  includes: []
  libraries: []
  min_version: 2023.2.2
esp32:
  board: esp32dev
  framework:
    version: 4.4.2
    sdkconfig_options: {}
    advanced:
      ignore_efuse_mac_crc: false
    source: ~3.40402.0
    platform_version: platformio/espressif32 @ 5.2.0
    type: esp-idf
  variant: ESP32
wifi:
  ap:
    password: ''
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  enable_btm: false
  enable_rrm: false
  networks:
  - ssid: ChuckNorris
    password: changeme
    priority: 0.0
  use_address: asdfadfasdf.local
api:
  port: 6053
  password: ''
  reboot_timeout: 15min
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
ota:
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
improv_serial: {}
dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
  import_full_config: false
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
    duration: 5min
    continuous: true
bluetooth_proxy:
  active: true
  cache_services: true
  connections:
  - {}
  - {}
  - {}
button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic
  disabled_by_default: false
  icon: mdi:restart-alert
  device_class: restart

i2c:

bme680_bsec:

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    humidity:
      name: "BME680 Humidity"
```

### Build log
```
INFO Reading configuration /config/asdffff.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing asdfadfasdf (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Dependency Graph
|-- Improv @ 1.2.3
Compiling .pioenvs/asdfadfasdf/src/esphome/components/bme680_bsec/bme680_bsec.o
Compiling .pioenvs/asdfadfasdf/src/esphome/components/button/button.o
Compiling .pioenvs/asdfadfasdf/src/esphome/components/dashboard_import/dashboard_import.o
Compiling .pioenvs/asdfadfasdf/src/esphome/components/esp32/core.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
src/esphome/components/bme680_bsec/bme680_bsec.h:12:10: fatal error: bsec.h: No such file or directory

**************************************************************
* Looking for bsec.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bsec.h"
* Web  > https://registry.platformio.org/search?q=header:bsec.h
*
**************************************************************

 #include <bsec.h>
          ^~~~~~~~
compilation terminated.
*** [.pioenvs/asdfadfasdf/src/esphome/components/bme680_bsec/bme680_bsec.o] Error 1
========================== [FAILED] Took 7.38 seconds ==========================
```

### Attempts to solve the problem taken
Tried [this](https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME680-on-ESPHome-using-Bosch-s-BSEC-library/m-p/13213#M3015) and [this](https://registry.platformio.org/libraries/boschsensortec/BSEC%20Software%20Library/installation)

forTT78, 2023-03-03:
I have the same problem, with Version: 2023.2.4 too. 

The configuration adapted fot arduino framework produces no error.
But i would like to use esp-idf frame work for mqtt with tls.
Any hint?

**Buid log:**

```INFO Generating C++ source...
INFO Compiling app...
********************************************************************************************************************************************************************************
Obsolete PIO Core v6.1.5 is used (previous was 6.1.6)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
********************************************************************************************************************************************************************************
Processing esp32-mobil-bme680-esp-idf (board: az-delivery-devkit-v4; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Dependency Graph
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/bme680_bsec/bme680_bsec.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
src/esphome/components/bme680_bsec/bme680_bsec.h:12:10: fatal error: bsec.h: No such file or directory

**************************************************************
* Looking for bsec.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bsec.h"
* Web  > https://registry.platformio.org/search?q=header:bsec.h
*
**************************************************************

 #include <bsec.h>
          ^~~~~~~~
compilation terminated.
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/logger/logger.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/md5/md5.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mdns/mdns_component.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mdns/mdns_esp32.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mdns/mdns_esp8266.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mdns/mdns_rp2040.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/custom_mqtt_device.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_backend_idf.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_binary_sensor.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_button.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_client.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_climate.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_component.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_cover.o
Compiling .pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/mqtt/mqtt_fan.o
*** [.pioenvs/esp32-mobil-bme680-esp-idf/src/esphome/components/bme680_bsec/bme680_bsec.o] Error 1
========================================================================== [FAILED] Took 3.28 seconds ==========================================================================
```

ssieb, 2023-03-23:
This is a problem with using proprietary software.  There is no version for the S2 and no source to compile.
https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME-680-BSEC-for-ESP32-S2/td-p/16892
The esp-idf problem might be because the library is Arduino only.

ssieb, 2023-03-24:
https://github.com/esphome/esphome/pull/4616 gives a proper error message now if you try to compile with esp-idf or one of the ESP32 variants.

closed 2023-03-24 by ssieb

forTT78, 2023-03-25:
> This is a problem with using proprietary software. There is no version for the S2 and no source to compile. https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME-680-BSEC-for-ESP32-S2/td-p/16892 The esp-idf problem might be because the library is Arduino only.

May be this helps:

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME-680-BSEC-for-ESP32-S2/m-p/60824/highlight/true#M12073 
--> "Please refer attached BSEC library for ESP32-S2."