esphome/issues #3894: Custom Sensor Component -> publish_state () - Value transfer problem

Woszu1981, 2022-12-12:
### The problem

I use Custom Sensor Component to display the value from the included library (water meter status reading).

`  
uint32_t MeterID = 1095753644;

class MyCustomSensor : public Component {
 public:                          
  Sensor *id_sensor = new Sensor();                              
  
void loop() { 
        id_sensor->publish_state(MeterID);
    }`

I test entered the value of the variable: 
MeterID = 1095753644

Unfortunately, this value does not pass correctly to yaml in esphome. The reading in yaml is standard:

`sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor->id_sensor}; 
  sensors:
  - name: "Meter Id"
    accuracy_decimals: 0`

The value is cut off. Instead of 1095753644 I get 1095753600.

If I type 10957536, then it transmits fine. 
Probably there is some limitation on the value of the transmitted data.  

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.1

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

Custom Sensor Component

ssieb, 2022-12-12:
GitHub issues are not for support requests.  Ask on the esphome discord or one of the forums.
A sensor state is a float value.  That number is too big to have full resolution.  You would be better off with a text sensor.

closed 2022-12-13 by ssieb

Woszu1981, 2022-12-13:
Now i know thenI  can't put this number into a sensor, but how can I put it into a text sensor?

ssieb, 2022-12-13:
Use `to_string()`.  Any further discussion on this needs to go elsewhere.