esphome/issues #3896: mdns ipv6: I always get a fe80 instead of an ip from my prefix

jeremyp3, 2022-12-12:
### The problem

hello,

  I use ESPHome version 2022.12.0b3 with ESPIDF and IPV6 support. 
It is a ULA prefix in autoconfiguration. The problem is that when I use mdns to look up my device, it always returns the ipv6 in fe80 instead of the ipv6 it is autoconfigured with.

Is there any way to correct this behavior on my side ?

  I specify that if I use the ipv6 without MDNS, it works as expected. 

Thanks,

Jeremyp3

### Which version of ESPHome has the issue?

2022.12.0b3

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

az-delivery-devkit-v4

### Component causing the issue

MDNS

### Example YAML snippet

```yaml
esphome:
  name: meters-hub

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf
    sdkconfig_options:
#      CONFIG_MBEDTLS_CERTIFICATE_BUNDLE: n
      CONFIG_LWIP_IPV6_AUTOCONFIG: y
```

jeremyp3, 2023-01-06:
UP ?

ignisf, 2023-01-19:
I also noticed that the esphomelib address list does not include IPv6:

```
$ avahi-browse -r _esphomelib._tcp
...
= enp1s0 IPv4 iot-bluetooth-proxy-123456                    _esphomelib._tcp     local
   hostname = [iot-bluetooth-proxy-123456.local]
   address = [10.5.3.85]
   port = [6053]
   txt = ["version=2022.12.3" "mac=84cca8123456" "platform=ESP32" "board=esp32dev" "network=wifi"]
= enp1s0 IPv6 iot-bluetooth-proxy-123456                    _esphomelib._tcp     local
   hostname = [iot-bluetooth-proxy-123456.local]
   address = [10.5.3.85]
   port = [6053]
   txt = ["version=2022.12.3" "mac=84cca8123456" "platform=ESP32" "board=esp32dev" "network=wifi"]
```

HeMan, 2023-06-19:
Could you try enabling ipv6 in network component instead of sdkconfig_options? ie
```
network:
  enable_ipv6: true
```
It should enable the flag `CONFIG_LWIP_IPV6` which adds aaaa-record to mdns.

jeremyp3, 2023-06-19:
thanks for the network section. 

 I replaced as indicated, but still no ipv6 other than fe80 as you had indeed noted

thanks for the interest in this bug

Jeremyp3

Hedda, 2023-09-13:
[HeMan](https://github.com/HeMan) wrote this is fixed by https://github.com/esphome/esphome/pull/5217 if using ESP-IDF 5.0.x (or presumably also ESP-IDF 5.1.x), not sure about 4?

HeMan, 2023-09-13:
> [HeMan](https://github.com/HeMan) wrote this is fixed by [esphome/esphome#5217](https://github.com/esphome/esphome/pull/5217) if using ESP-IDF 5.0.x (or presumably also ESP-IDF 5.1.x), not sure about 4?

Unfortunately not.

ignisf, 2023-09-15:
Confirming:
```
# avahi-resolve -6 -v -n iot-bt-proxy-123456.local
Server version: avahi 0.8; Host name: home-assistant.local
iot-bt-proxy-123456.local	fdbd:7cbb:3b06:64:a276:4eff:fe12:3456
```

This command still for some reason does not list the IPv6 addresses (it might be unrelated anyway):
```
avahi-browse -r _esphomelib._tcp
```

jeremyp3, 2023-09-21:
as discussed in #4876

fix in :
  framework:
    type: esp-idf
    version: 5.0.2
    platform_version: 6.3.2

I confirm that in mdns I now have the ip of my prefix. On the other hand, I still have fe80. the problem with that is that it ping pongs between the two ip's. I think you should remove the ip in fe80 if ESP has another ip, or at least an option to do so, because otherwise it causes unexpected results.

ignisf, 2023-09-21:
I also observed random switching between fe80 and the IP from the assigned prefix. Not sure I'd go with broadcasting just one, unless this is generally the practice in other mdns implementations.

randybb, 2023-09-21:
link local address (fe80) is mandatory for ipv6

jeremyp3, 2023-09-21:
> link local address (fe80) is mandatory for ipv6

i never asked to remove fe80 from the ipv6 interface.
i only made a suggestion to be able to hide it from mdns advertisements when another ipv6 address is present on the ESP

in some use cases, I use mdns, but I'm not on the same network segment (l2) so I can't contact fe80, only my ULA prefix.

What's more, none of the devices on my network expose their fe80 on avahi.

HeMan, 2023-09-22:
It's not sure that you could filter which addresses mdns shows, but I'll have a look.

HeMan, 2024-05-19:
After some more mDNS testing I found that `avahi-resolve` only shows one address, even if the client sends more. I'm still to find a tool (other than wireshark) that shows all the addresses returned.

jeremyp3, 2024-05-19:
personally, to see the addresses exposed in mdns, I  use dig, on the ip address of the ESP on port 5353 asking for its MDNS name 

$ dig @192.168.80.24 -p 5353 meters-hub.local

;; QUESTION SECTION:
;meters-hub.local.		IN	A

;; ANSWER SECTION:
meters-hub.local.	120	IN	A	192.168.80.24
meters-hub.local.	120	IN	AAAA	fe80::ea6b:eaff:fe30:59cc
meters-hub.local.	120	IN	AAAA	fd42:d1d2:fada:cbfd:ea6b:eaff:fe30:59cc

I don't know if this is a good way to do it, but here's how I do it :)