esphome/issues #3897: bluetooth_proxy makes wifi unstable

eduperez, 2022-12-12:
### The problem

I have a couple of Sonoff Dual R3 devices that work flawlessly, except for bluetooth_proxy: when bluetooth_proxy is enabled, both of them keep disconnecting and reconnecting to the WiFi network, constantly.

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.12.3

### What platform are you using?

ESP32

### Board

Sonoff Dual R3

### Component causing the issue

bluetooth_proxy

### Example YAML snippet

```yaml
esp32_ble_tracker:
  scan_parameters:
   active: false

bluetooth_proxy:
  active: true
```

### Anything in the logs that might be useful for us?

```txt
INFO sonoff1.intranet: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO sonoff1.intranet: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for sonoff1.intranet
WARNING Disconnected from API
INFO Successfully connected to sonoff1.intranet
```

### Additional information

Even with` VERY_VERBOSE` logs active, I could not see any other relevant message when the disconnection happens.

PetrMa, 2022-12-14:
It seems I have the similar issue.
https://github.com/esphome/issues/issues/3907

In my case it happenes only if ESP32 camera config is used together with BL proxy.

bdraco, 2022-12-16:
Is this any better with 2022.12.0 and esp-idf?

eduperez, 2022-12-18:
> Is this any better with 2022.12.0 and esp-idf?

Yes! With 20.22.12.0 (no ESP-IDF, and no need to re-partition), it has been stable for 12 hours; I think this has been fixed.

closed 2022-12-18 by eduperez

bdraco, 2022-12-18:
> > Is this any better with 2022.12.0 and esp-idf?
> 
> 
> 
> Yes! With 20.22.12.0 (no ESP-IDF, and no need to re-partition), it has been stable for 12 hours; I think this has been fixed.

Without the repartition active connections are slower but should work just fine

eduperez, 2022-12-21:
> Without the repartition active connections are slower but should work just fine

Unfortunately, these devices are now inside a hard-to-reach junction box, and flashing them over serial would take a lot of work; I think I will delay that until it is absolutely necessary.