esphome/issues #3898: List of Strapping Pins does not Match ESP32 Data Sheet

skaterdude, 2022-12-13:
### The problem

The list of strapping pins in gpio_esp32.py is:
_ESP32_STRAPPING_PINS = {0, 2, 4, 12, 15}

However, the ESP32 datasheet here:
https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf
on page 21:
• MTDI
• GPIO0
• GPIO2
• MTDO
• GPIO5
(MTDI is GPIO 12 and MTDO is GPIO 15).

It appears that the list in gpio_esp32.py should be:
_ESP32_STRAPPING_PINS = {0, 2, 5, 12, 15}

I checked the errata and there does not appear to be a change between versions of the chip.  There's also nothing in the revision to the data sheet about 4 vs 5.

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.3

### What platform are you using?

ESP32

### Board

(Variant of) Heltec WiFi Kit 32

### Component causing the issue

GPIO

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12    
      mode:
        input: true
        pullup: true
    id: gpio_12
  - platform: gpio
    pin:
      number: GPIO13
      mode:
        input: true
        pullup: true
```

### Anything in the logs that might be useful for us?

```txt
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
```

skaterdude, 2022-12-13:
Note: The warning I included is just to provide the text template from the log.  This issue is about GPIO 4 vs GPIO 5.  GPIO 12 (aka MTDI) is correctly generating a warning in the log snippet.

ed-so, 2023-06-22:
same issue here

kernelpanic85, 2023-07-13:
I am seeing the same issue. I have a nodemcu-32s board and I'm getting a warning about GPIO 4 being a strapping pin, however the pinout does not classify it as a strapping pin. Does the list just need to be updated in the code?

https://www.waveshare.com/nodemcu-32s.htm