esphome/issues #3899: esp32-c3: xtensa-lx106-elf-g++: not found

Gloomyeye, 2022-12-13:
### The problem

I am not able to compile a simple esphome config for my esp32-c3 with board esp32-c3-devkitm-1.

I bought the following esp32: https://www.reichelt.de/entwicklungsboard-esp32-c3-mini-1-esp32c3devkitm-p311730.html?&trstct=pos_1&nbc=1
Now, I created a new esp32-c3 test config file and try to compile this file. Everytime, I get the compile issue.

I am using esphome and homeassistant in a docker environment on a raspi4 (32bit os system).

### Which version of ESPHome has the issue?

2022.12.0b5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.12.4

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

riscv32-esp-elf-g++: not found

### Example YAML snippet

```yaml
esphome:
  name: esp32-c3-deltasol

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "rtNqto5QVuHDNGmwy49sAYuJF/AqmkODco6OGDqI590="

ota:
  password: "8128cdc333ff77136fdca21a774bcf70"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-C3-Deltasol"
    password: "ticdAfDllBs2"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esp32-c3-deltasol.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-c3-deltasol (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_connection.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2.cpp.o] Error 127
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 4.32 seconds ==========================
```

Gloomyeye, 2022-12-14:
I haven’t had the time to do a clean install with the 64 but version. Anyway, you think that the issue will not occur if I upgrade the raspi OS to 64 but version?

ssieb, 2022-12-14:
No one has reported an issue with that.

Gloomyeye, 2022-12-15:
I changed my raspi os from 32bit to 64bit with the help of this tutorial: https://www.tutonaut.de/raspberry-pi-von-32-auf-64-bit-umstellen/
After that I removed all my docker images/containers completly from the raspberry pi and reloaded them again. Unfortunately, the compiling issue still exists:

`
INFO Reading configuration /config/esp32-c3-deltasol.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-c3-deltasol (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_connection.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/esp32-c3-deltasol/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 6.31 seconds ==========================
`

Gloomyeye, 2022-12-15:
I startet another test with another esp32 framework. However, another issue occurs.

This ist the config:
```#https://esphome.io/components/esp32.html
esp32:
  board: esp32-c3-devkitm-1
  framework:
      type: esp-idf
      version: latest
      # Custom sdkconfig options
      sdkconfig_options:
        CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      # Advanced tweaking options
      advanced:
        ignore_efuse_mac_crc: false```

This is the log:
```
INFO Reading configuration /config/esp32-c3-deltasol-sc.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-c3-deltasol-sc (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /config/.esphome/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch3/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch3/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch3/bin/riscv32-esp-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/esp32-c3-deltasol-sc/.pioenvs/esp32-c3-deltasol-sc/CMakeFiles/CMakeOutput.log".
See also "/config/.esphome/build/esp32-c3-deltasol-sc/.pioenvs/esp32-c3-deltasol-sc/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/config/.esphome/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch3/bin/riscv32-esp-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /config/.esphome/build/esp32-c3-deltasol-sc/.pioenvs/esp32-c3-deltasol-sc/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/config/.esphome/platformio/packages/tool-ninja/ninja cmTC_028a0 && [1/2] Building C object CMakeFiles/cmTC_028a0.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_028a0.dir/testCCompiler.c.obj 
    /config/.esphome/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch3/bin/riscv32-esp-elf-gcc   -march=rv32imc -o CMakeFiles/cmTC_028a0.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /config/.esphome/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch3/bin/riscv32-esp-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:296 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 9.97 seconds ==========================
```

jesserockz, 2022-12-15:
> I changed my raspi os from 32bit to 64bit with the help of this tutorial: [tutonaut.de/raspberry-pi-von-32-auf-64-bit-umstellen](https://www.tutonaut.de/raspberry-pi-von-32-auf-64-bit-umstellen/)

This only changes the kernel, not the full userspace, which is what is actually causing this issue. If you were using the 32-bit kernel, or a full 64-bit OS then it would work as expected.

Gloomyeye, 2022-12-22:
I did a complete reinstall of my raspberry pi and installed raspberry pi os lite (64 bit). Now, I can compile with the latest esphome version 2022.12.3 the mentioned test config for my esp32-c3.

closed 2022-12-22 by Gloomyeye

lexcorp, 2023-03-02:
Tengo el mismo problema, la configuración de mi ambiente es:

Raspberry Pi 4
Versión del Sistema Operativo: 5.10.0-20-arm64
Arquitectura de CPU: aarch64

## System Information

version | core-2023.2.5
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.10.0-20-arm64
arch | aarch64
timezone | America/Mexico_City
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.31.0
Stage | running
Available Repositories | 1239
Downloaded Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2023.02.dev2801
agent_version | 1.4.1
docker_version | 23.0.1
disk_total | 219.6 GB
disk_used | 8.9 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.6.1), File editor (5.5.0), ArgonOne Active Cooling (29c), HassOS I2C Configurator (0.13a), Duck DNS (1.15.0), ESPHome (2023.2.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 1 de marzo de 2023, 00:57
-- | --
current_recorder_run | 2 de marzo de 2023, 08:40
estimated_db_size | 0.99 MiB
database_engine | sqlite
database_version | 3.38.5

</details>