esphome/issues #3904: Cannot compile with 2022.12.0

SVH-Powel, 2022-12-14:
### The problem

Compile fails with "sh: 1: xtensa-esp32-elf-g++: not found"

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP32

### Board

esp32dev

### Anything in the logs that might be useful for us?

```txt
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.2.0 has been installed!
INFO espressif32@5.2.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40201.0
INFO Installing platformio/tool-esptoolpy @ ~1.40201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40201.0 has been installed!
INFO tool-esptoolpy@1.40201.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Compiling /data/luftkvalitet1/.pioenvs/luftkvalitet1/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/luftkvalitet1/.pioenvs/luftkvalitet1/src/esphome/components/api/api_connection.cpp.o] Error 127
========================= [FAILED] Took 371.07 seconds =========================
```

ssieb, 2022-12-14:
What are you running it on?  And 32-bit or 64-bit?

leppa, 2022-12-14:
Same issue. Also using Raspberry Pi 4 with Raspbian (32-bit).

JackTalisker, 2022-12-14:
Same here on RPI 3b:

```
INFO Reading configuration /config/boxmultisensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing boxmultisensor (board: wt32-eth01; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Platform Manager: espressif32@5.2.0 has been installed!
INFO espressif32@5.2.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40201.0
INFO Installing platformio/tool-esptoolpy @ ~1.40201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40201.0 has been installed!
INFO tool-esptoolpy@1.40201.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/boxmultisensor/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/boxmultisensor/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/boxmultisensor/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/boxmultisensor/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/boxmultisensor/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/boxmultisensor/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/boxmultisensor/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/boxmultisensor/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================= [FAILED] Took 329.43 seconds =========================
```

zanna-37, 2022-12-14:
Me too, 32bit raspberry, ESPHome run in Docker.

ssieb, 2022-12-14:
Why do people run 32-bit systems?  What's the point?  Just like https://github.com/esphome/issues/issues/3899, this isn't an esphome issue.  Platformio gets confused with a 32-bit OS running on a 64-bit kernel.

leppa, 2022-12-14:
> Why do people run 32-bit systems? What's the point? Just like #3899, this isn't an esphome issue. Platformio gets confused with a 32-bit OS running on a 64-bit kernel.

Because at the time when I was setting up my Raspberry Pi, the official OS for Raspberry Pi was Raspbian and there was no (stable) 64-bit variant.

ssieb, 2022-12-14:
Then I suggest you re-install it now as 64-bit.

ssieb, 2022-12-14:
A pure 32-bit system will probably work.  It's the combination of 64-bit kernel and 32-bit userspace that causes the problem.  Maybe there was a change in the newer platformio version.

SVH-Powel, 2022-12-14:
> A pure 32-bit system will probably work. It's the combination of 64-bit kernel and 32-bit userspace that causes the problem. Maybe there was a change in the newer platformio version.

As others have written, when I installed my system, 32-bit was the only option. I am running hass-os. I have no idea of what kernel it is (64 or 32 bits)

ssieb, 2022-12-14:
Fine, but I'm telling you that there's nothing we can do about it.  This is an issue with platformio.

JackTalisker, 2022-12-14:
Seems (to me) a hasty statement since this issue appeared with the changes in 2022.12.0.
It only follows that at the moment, with this new release, the esp32 side is unusable

kevjs1982, 2022-12-14:
> Seems (to me) a hasty statement since this issue appeared with the changes in 2022.12.0. It only follows that at the moment, with this new release, the esp32 side is unusable

In particular when Home Assistant OS installs on a Raspberry Pi 4 (according to the stats the most popular hardware) are impacted too, and the installation of this broken version is allowed from the Home Assistant Add On Store - seems a bit of an oversight when one Nabu Casu project allows a broken version of another of its own products to be installed with not even a warning!

RaceAap, 2022-12-14:
I get a different error but result is the same:
`Processing minitemp01 (board: mhetesp32devkit; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling /data/minitemp01/.pioenvs/minitemp01/src/esphome/components/api/api_connection.cpp.o
Compiling /data/minitemp01/.pioenvs/minitemp01/src/esphome/components/api/api_frame_helper.cpp.o
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/wchar.h:6,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:40,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory`

en-ver, 2022-12-14:
Same for me on Orange PI Plus 2E and Banana PI with Armbian Bullseye

edenhaus, 2022-12-14:
I had also compile errors with the docker container on Ubuntu server 64-bit, which I solved by deleting the folders `build` (or the folder specified with the config option `build_path`) and `platformio` in the folder `.esphome` (this folder exists inside your config folder).
Not sure if this helps also for your issue but maybe someone can give it a try. Please stop the addon before deleting this folders

Pop70, 2022-12-14:
> A pure 32-bit system will probably work. It's the combination of 64-bit kernel and 32-bit userspace that causes the problem. Maybe there was a change in the newer platformio version.

RPi 3b 32 bit HASSOS 
Same problem

GazAt105, 2022-12-14:
Same here on Odroid XU4:
Home Assistant 2022.12.6
Supervisor 2022.11.2
Operating System 9.4
Frontend 20221213.0 - latest 

The OS is 32 bit.

ESP8266 (platform: platformio/espressif8266 @ 3.2.0)) complies okay.

ESP32 fails as per below.

> INFO Reading configuration /config/esphome/esp32-001.yaml...
INFO Detected timezone 'Australia/Brisbane'
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-001 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- EspSoftwareSerial @ 6.16.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/adc/adc_sensor.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_connection.cpp.o
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_server.cpp.o
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/list_entities.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/adc/adc_sensor.cpp.o] Error 127
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/api_server.cpp.o] Error 127
Compiling /data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/proto.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/list_entities.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32-001/.pioenvs/esp32-001/src/esphome/components/api/proto.cpp.o] Error 127
========================== [FAILED] Took 8.76 seconds ==========================

cvwillegen, 2022-12-14:
Home Assistant 2022.12.5
Supervisor 2022.11.2
Operating System 9.4
Frontend-versie: 20221213.0 - latest
RPi4 - 32bit

Same problem, using nothing but the Community Addon...

ssieb, 2022-12-15:
> RPi 3b 32 bit HASSOS Same problem

@Pop70, what does `uname -a` show for the architecture?
(or anyone else that thinks they're running a 32-bit HASSOS)

jesserockz, 2022-12-15:
Alternatively, paste the top section from here:

[image]

en-ver, 2022-12-15:
`## System Information

version | core-2022.12.6
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.80-sunxi
arch | armv7l
timezone | Europe/Kiev
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 5 December 2022 at 12:31
-- | --
current_recorder_run | 15 December 2022 at 09:13
estimated_db_size | 125.14 MiB
database_engine | mysql
database_version | 10.6.10

</details>`

RaceAap, 2022-12-15:
I updated my previous post, but i'm running hassos 64bits in a virtual machine on a Synology DS920+.
ESPHome is the standard add-on from HA so no control on which version gets installed unless there is an update.

rikkertkoppes, 2022-12-15:
armv7l here, so 32 bit

JackTalisker, 2022-12-15:
## System Information

version | core-2022.12.6
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.10.103-v7+
arch | armv7l
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.28.4
Stage | running
Available Repositories | 1166
Downloaded Repositories | 18

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 11
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 8 dicembre 2022 09:29
-- | --
current_recorder_run | 15 dicembre 2022 08:31
estimated_db_size | 573.83 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Sonoff</summary>

version | 3.3.1 (b20e33c)
-- | --
cloud_online | 5 / 5
local_online | 5 / 5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

JackTalisker, 2022-12-15:
## System Information

version | core-2022.12.6
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.10.103-v7+
arch | armv7l
timezone | Europe/Rome
config_dir | /config

ssieb, 2022-12-15:
We don't need any more information now.  It turns out that it's a problem with the 32-bit toolchain not matching debian.  Hopefully there will be an add-on update soon to workaround that.

closed 2022-12-15 by jesserockz

domcat11, 2022-12-16:
I also had the same problem, solved with new version 2022.12.1, but now I've this new one error
`Compiling /data/esp32azdel/.pioenvs/esp32azdel/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:116:10: fatal error: sys/ioctl.h: No such file or directory`

I'm on RPI3 HASSIO
Any idea?
Thanks

zanna-37, 2022-12-16:
with 2022.12.1 this error occurs to me which was not present in 2022.11

```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- Preferences @ 2.0.0
|-- ESP32 BLE Arduino @ 1.0.1+sha.adc2aee
|-- Nuki Ble for Esp32 @ 0.0.4+sha.b0f8588
|   |-- NimBLE-Arduino @ 1.4.1+sha.0d9f039
|   |-- CRC16 @ 0.0.0+20221216094630.sha.f46d86b
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/esp-nuki-bridge/src/esphome/components/nuki_lock/nuki_lock.cpp.o
Compiling .pioenvs/esp-nuki-bridge/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/esp-nuki-bridge/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/esp-nuki-bridge/src/esphome/components/socket/bsd_sockets_impl.cpp.o
In file included from .piolibdeps/esp-nuki-bridge/Nuki Ble for Esp32/src/NukiBle.h:9,
                 from src/esphome/components/nuki_lock/nuki_lock.h:9,
                 from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
.piolibdeps/esp-nuki-bridge/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pioenvs/esp-nuki-bridge/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
Compiling .pioenvs/esp-nuki-bridge/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
```

ssieb, 2022-12-16:
@zanna-37 are you using a custom or external component?  Your compile is importing a second BLE library and I can only imagine that is going to cause you trouble. 

This issue is closed.  Any other compile issues need a new issue.  And try a clean build first.

zanna-37, 2022-12-16:
> @zanna-37 are you using a custom or external component?
Yes, I am, but it was working before. I think a piece went missing, just like the `xtensa-esp32-elf-g++: not found` of this issue. The issue is the maintainer of the external component is not very responsive.
 
> And try a clean build first
Yeah, I already tried.

nliaudat, 2023-01-20:
That's related to old libs on platformio cache.

To fix it on windows : 
`cd C:\Users\%USERNAME%\AppData\Local\Programs\Python\Python310\Lib\site-packages`
delete platformio & platformio-6.1.5.dist-info
`cd In C:\Users\%USERNAME%\`
delete .platformio

run : 
```
pip install --upgrade esphome
pip install --upgrade platformio
```

lambu76, 2023-02-04:
with ESP 2022.12.8 default add-on on Homeassistant below version I still have the problem.

How to solve it ?

Home Assistant 2023.2.1
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230202.0 - latest

alpat59, 2023-02-14:
ESPHome 2022.12.8 installed as HA add-on on Raspberry and I have the same problem 
"sh: 1: xtensa-esp32-elf-g++: not found"

Could someone try to fix it ?
Thank you for support

lazlykama, 2023-02-15:
Same here, if you use ESPHome as a HA add-on the problem not solved yet. cc: @jesserockz

SVH-Powel, 2023-02-16:
> Same here, if you use ESPHome as a HA add-on the problem not solved yet. cc: @jesserockz

In that case, it would be better to open a new issue instead of writing here. I am still on the 2022.11.5 version and hold off any upgrades till this and other issues are fixed.

lambu76, 2023-02-16:
done

https://github.com/esphome/issues/issues/4178

lexcorp, 2023-03-02:
ESPHome: Current version: 2023.2.4

El problema continua tengo las versiones mas recientes

## System Information

version | core-2023.3.0
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.10.0-20-arm64
arch | aarch64
timezone | America/Mexico_City
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.31.0
Stage | running
Available Repositories | 1239
Downloaded Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2023.02.dev2801
agent_version | 1.4.1
docker_version | 23.0.1
disk_total | 219.6 GB
disk_used | 9.3 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.6.1), File editor (5.5.0), ArgonOne Active Cooling (29c), HassOS I2C Configurator (0.13a), Duck DNS (1.15.0), ESPHome (2023.2.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 1 de marzo de 2023, 00:57
-- | --
current_recorder_run | 2 de marzo de 2023, 14:08
estimated_db_size | 1.20 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

lexcorp, 2023-03-03:
Para los que tienen arquitectura de 64 bits esto soluciono mi problema: Edite el archivo "util.py", primero realice la búsqueda dentro de mi sistema de archivos y me devuelve 3 resultados, como no supe cuál de los tres es el que debía modificar, modifique los tres:

El código después de modificarlo:

def get_systype():
    system = platform.system().lower()
    arch = platform.machine().lower()
    if system == "windows":
        if not arch:  # issue #4353
            arch = "x86_" + platform.architecture()[0]
        if "x86" in arch:
            arch = "amd64" if "64" in arch else "x86"
    if arch == "aarch64":
        if platform.architecture()[0] == "32bit":
            arch = "armv7l"
    return "%s_%s" % (system, arch) if arch else system

https://github.com/esphome/issues/issues/3092#issuecomment-1412223981

root@rpi4-20230102:/# find | grep platformio/util.py

./usr/local/lib/python3.9/dist-packages/platformio/util.py

./var/lib/docker/overlay2/7dc1dc2395dd053c78fc5cd8616d1c61f5703e4e6433d46eb4ccd3cbb1bff485/merged/usr/local/lib/python3.9/dist-packages/platformio/util.py

./var/lib/docker/overlay2/c04cf2bac904287623c5baf075232b1ceae6d9be38336527a51a251bc513ec79/diff/usr/local/lib/python3.9/dist-packages/platformio/util.py

root@rpi4-20230102:/# nano ./usr/local/lib/python3.9/dist-packages/platformio/util.py 

root@rpi4-20230102:/# nano ./var/lib/docker/overlay2/7dc1dc2395dd053c78fc5cd8616d1c61f5703e4e6433d46eb4ccd3cbb1bff485/merged/usr/local/lib/python3.9/dist-packages/platformio/util.py 

root@rpi4-20230102:/# nano 
./var/lib/docker/overlay2/c04cf2bac904287623c5baf075232b1ceae6d9be38336527a51a251bc513ec79/diff/usr/local/lib/python3.9/dist-packages/platformio/util.py

Después de realizar este ajuste, pude compilar sin problemas y completar la instalación y mi dispositivo aparece en linea.

2opremio, 2023-05-05:
I am still have this problem with the most recent version of esphome and platformio on a raspberry pi 4 with raspberry pi os

Florian-Schmidt, 2023-05-07:
I still get this message on a raspberry pi 4 with 64bit OS (aarch64) too. In my case I'm using a esp8266.

Read several issues now and nothing worked. I update esphome docker container, I cleaned up build files and reinstalled home assistant integration already.

ndfred, 2023-05-18:
Exactly the same issue here, docker ESPHome on a Raspberry Pi 4, 32 bit OS, started happening two weeks ago:

    Compiling .pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_connection.cpp.o
    sh: 1: xtensa-esp32-elf-g++: not found
    Compiling .pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_frame_helper.cpp.o
    sh: 1: xtensa-esp32-elf-g++: not found
    Compiling .pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_pb2.cpp.o
    sh: 1: xtensa-esp32-elf-g++: not found
    *** [.pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_connection.cpp.o] Error 127
    Compiling .pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_pb2_service.cpp.o
    *** [.pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
    *** [.pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_pb2.cpp.o] Error 127
    sh: 1: xtensa-esp32-elf-g++: not found
    *** [.pioenvs/arthur_bedroom_esp32/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
    ========================= [FAILED] Took 100.94 seconds =========================

I tried clearing `.esphome/platformio` to re-download the toolchains but that did not help. Looking around `.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++` is present. ESPHome dashboard says it installed `toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5`.

ssieb, 2023-05-18:
@ndfred try running that program and see what happens.

ndfred, 2023-05-18:
Super weird:

    $ ./.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
    -bash: ./.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++: No such file or directory

It looks like an arm64 binary:

    $ file ./.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
    ./.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, BuildID[sha1]=abfed9e5893019faebfd4748833b6357f5cfe614, stripped

And my system is ARM32, but for some reason my kernel is ARM64 (don't remember when I made that change), that is probably what is throwing ESPHome off:

    $ uname -a
    Linux bagend 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux
    $ file /usr/bin/arm-linux-gnueabihf-g++-8
    /usr/bin/arm-linux-gnueabihf-g++-8: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=35312a560fd37e95fb3a5d385e78048673f06cf2, stripped

ssieb, 2023-05-18:
Yes, that's the problem. And it's not esphome, it's platformio.  It sees the arch is 64-bit (from the kernel), so it gets the 64-bit toolchain.

Florian-Schmidt, 2023-05-18:
@ndfred Thanks, seems to be exactly my situation too. Kernel 64-bit and binaries 32-bit.

Micha70, 2023-05-19:
I have the same issue too... Shouldn't the issue reopened? Or is there a new one existing?

Home Assistant 2023.5.3
Supervisor 2023.04.1
Frontend 20230503.3 - latest

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

ESP-Home 2023.5.1

'''

/data/kuechenlampe/.pioenvs/kuechenlampe/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/kuechenlampe/.pioenvs/kuechenlampe/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/kuechenlampe/.pioenvs/kuechenlampe/src/esphome/components/api/api_connection.cpp.o] Error 127
***
'''

ndfred, 2023-05-19:
From [this commit](https://github.com/esphome/esphome/commit/1eacbd50fadec3ea4b881d5f2dfa181e5e1f69b1) mentioned above it looks like the fix was to pull in the arm 32 linker in the docker container to work around the issue, I might play with that and see if it fixes it:

    $ ln -s /lib/arm-linux-gnueabihf/ld-linux.so.3 /lib/ld-linux.so.3

ndfred, 2023-05-27:
I tried to shell into the container and the link trick above was already done properly:

    root@bagend:/config# ls -l /lib/ld-linux.so.3
    lrwxrwxrwx 1 root root 38 May 17 01:21 /lib/ld-linux.so.3 -> /lib/arm-linux-gnueabihf/ld-linux.so.3

I then looked into disabling my arm64 kernel to see if that would help (also means I will max out at 4GB on my 8GB Pi I guess?), followed [this guide](https://medium.com/for-linux-users/how-to-make-your-raspberry-pi-4-faster-with-a-64-bit-kernel-77028c47d653) and added `arm_64bit=0` to the `[pi4]` section of `/boot/config.txt` and restarting. After a bit of a stressful reboot we are in 32 bit mode:

    fred@bagend:~ $ uname -a
    Linux bagend 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

Got the ESPHome Docker container up and tried to update one of my nodes, that then downloads a new version of the toolchain and builds / installs no problem.

So that means the issue really is the kernel, and this pretty involved workaround works. I guess I will update all my nodes, and upgrade my system to Raspberry Pi OS 64 bit, however scary that sounds. I can say that this used to work for sure, so something broke recently, either in ESPHome or in the toolchains.