esphome/issues #3905: adopt doesn't work when status_use_ping is enabled

pathia, 2022-12-14:
### The problem

I had a problem with getting an esphome node to popup in esphome for adoption (in the esphome sidebar item with the dashboard)
Turns out that I had to disable status_use_ping for it to come available for adoption over there.
I wouldn't expect this behavior since it isn't clearly written in the docs. So I assume it's a bug.

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

?

ssieb, 2022-12-14:
`status_use_ping` disables mdns use which is how detection works.  I suppose it might be possible to still try to listen even if broadcasting isn't working.  But why are you even using that if mdns works?

pathia, 2022-12-14:
I had mdns disabled because I needed that it the past. Nevertheless if I see a setting status_use_ping it's completely unexpected that autodetection of new devices is also disabled.