esphome/issues #3907: bluetooth_proxy does not work wtth ESP32 cam

PetrMa, 2022-12-14:
### The problem

Hi,

today morning I made an upgrade of esphome to the latest version [v2022.12.0].

After the upgrade (some esp32 was upgrade by OTA, some by cable) some ESP32 modules stopped working.
After each reset the WiFi module is OK and ESP32 pinging in local network but after aprox 30 second the ping is lost and device stopped responding.

My configuration of ESP32 devices includes configuration of ESP32 camera and bluetooth_proxy together.
If I delete bluetooth_proxy setting from config and leave only the camera settings, everything works fine.
Also if I delete camera settings from config and leave bluetooth_proxy settings, everything works fine.

But both of these settings, camera and bluetooth_proxy in one configuration cause the device is stopped working on network after 30-60 seconds.

Before today upgrade to version [v2022.12.0] there was no issue.

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP32

### Component causing the issue

bluetooth_proxy

### Example YAML snippet

```yaml
esphome:
  name: esp32-cam2

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password
  encryption:
    key: !secret encryption_key

ota:
  password: "....."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Cam2 Fallback Hotspot"
    password: "......"

captive_portal:

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

# Image settings
  name: Kamera předsíň
  resolution: 1600x1200

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4

light:
  - platform: binary
    output: gpio_4
    name: Kamera předsíň Light

button:
  - platform: restart
    name: Kamera předsíň Restart
    id: esp32_cam2_restart

binary_sensor:
  - platform: status
    name: Kamera předsíň Status
    id: esp32_cam2_status

sensor:
  - platform: wifi_signal
    name: Kamera předsíň WiFi Signal
    id: esp32_cam2_wifi_signal
    update_interval: 60s

  - platform: uptime
    name: Kamera předsíň Uptime
    id: esp32_cam2_uptime

bluetooth_proxy:
  active: true
```

PetrMa, 2022-12-14:
When the connection lost, I´ve got the same error as mentioned there
https://github.com/esphome/issues/issues/3897

PetrMa, 2022-12-14:
Update: Rollback back to 2022.11.5 solved the issue.

dersch81, 2023-11-05:
Any update here? I have the same issue with activated bluetooth_proxy: (not other options) on all ESP32 Cam Modules with 2023.10.6 (also before that version)

GiorgosPap777, 2024-05-03:
I have some great news! i tried to make the esp32-cam a Bluetooth proxy as i am running Home Assistant in a docker container and i don't have a Bluetooth dongle. I made it work by removing the sd card and pressing the RESET button before even you click the blue "connect" button. After it is finished you re-do the process but now there a new option has appeared which says "connect to wi-fi". you click it, connect to wifi.
REMEMBER that patience is key! 
You may have to redo this 1-2 times. NEVER GIVE UP