esphome/issues #3911: Error with component webserver: on ESPHome 22022 12.0 -   This feature is only available with frameworks ['arduino'].

adekloet, 2022-12-14:
### The problem

After update to Current version: 2022.12.0 (Changelog) I tried to update a device with a formerly working config OTA, but it fails with the following error:

INFO Reading configuration /config/esphome/esp32-bluetooth-proxy1.yaml...
INFO Detected timezone 'Europe/Amsterdam'
Failed config

web_server: [source /config/esphome/esp32-bluetooth-proxy1.yaml:15]
  
  This feature is only available with frameworks ['arduino'].
  port: 80
async_tcp: None
  {}
This feature is only available with frameworks ['arduino']

When I remove the config for webserver: then it does compile and work.

Was there a change in 12.0 that the webserver would not work on an ESP8266 Out of the box? do I need to add some config or framework?

Running HASS OS on PI with ESPHOME as Add-on. I did restart etc.

Thank you

### Which version of ESPHome has the issue?

Current version: 2022.12.0 (Changelog) 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022 12.4 core

### What platform are you using?

ESP8266

### Board

Wemos d1 mini

### Component causing the issue

Webserver:

### Example YAML snippet

```yaml
# Enable Web server.
web_server:
  port: 80
  
# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time
```

### Anything in the logs that might be useful for us?

```txt
error during install: 

This feature is only available with frameworks ['arduino'].
  port: 80
async_tcp: None
  {}
This feature is only available with frameworks ['arduino']
```

adekloet, 2022-12-14:
found it's caused by this chnage in the framework type last week:

Switch proxies to use esp-idf instead of arduino -> https://github.com/esphome/bluetooth-proxies

added to the config:

esp32:
  board: esp32dev
  framework:
    type: arduino

Then it compiles again with the webserver present as I do want to see the output at the device level via IP ast port 80. Doesn't the esp-idf type framework have no webserver: at device level? Or is there siomething I could change to make that work?

adekloet, 2022-12-14:
More info https://github.com/esphome/esphome-docs/pull/2469 I guess the webserver component interferes with active BT connections. Strange but true apparently. Framework Arduino obviously still works. type: esp-idf does not work with webserver: in there. The latter framework seems more stable then Arduino, but I do hope that we can get the webserver: back into esp-idf then...

bdraco, 2022-12-14:
There is an effort to get the webserver working with idf here https://github.com/esphome/esphome/pull/3500

adekloet, 2022-12-15:
> There is an effort to get the webserver working with idf here [esphome/esphome#3500](https://github.com/esphome/esphome/pull/3500)

Thanks for the info, I will keep an eye on it as the webserver component is very memory hungry on the arduino framework but still like the functionality.

closed 2022-12-16 by adekloet