esphome/issues #3912: SDM meter modbus CRC errors

depuits, 2022-12-14:
### The problem

SDM meter modbus is no longer working. Entities are unknown in home assistant and the esphome logs keeps giving CRC errors. Reverting back version 2022.11.5 fixes the problem.

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Component causing the issue

sdm_meter

### Example YAML snippet

```yaml
sensor:
  - platform: sdm_meter
    phase_a:
      current:
        name: "Apartment phase 1 current"
      voltage:
        name: "Apartment phase 1 voltage"
      active_power:
        name: "Apartment phase 1 power"
    import_active_energy:
      name: "Apartment total energy"
    update_interval: 15s
```

dxxxm, 2022-12-14:
Can confirm the issue for the ESP32 platform.
For both a SDM120 and a SDM630. ea on a different type of esp32 board (esp-wrover-kit and mhetesp32minikit)
A rollback to version 2022.11.5 fixed it.
```
Processing it-cabfan (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
...
[15:44:29][D][uart_debug:158]: >>> "\x01\x04\x00\x00\x00P\xF06"
[15:44:29][W][modbus:105]: Modbus CRC Check failed! C071!=00
[15:44:29][W][modbus:105]: Modbus CRC Check failed! C071!=00
[15:44:29][W][modbus:105]: Modbus CRC Check failed! C071!=00
[15:44:29][W][modbus:105]: Modbus CRC Check failed! C071!=00
[15:44:29][W][modbus:105]: Modbus CRC Check failed! C071!=4842
[15:44:29][D][uart_debug:158]: <<< "\x01\x04\xA0Ceff\x00\x00\x00\x00\x00\x00\x00\x00?\t7L\x00\x00\x00\x00\x00\x00\x00\x00B\x84\xCC\xCD\x00\x00\x00\x00\x00\x00\x00\x00B\x97\x90\x18\x00\x00\x00\x00\x00\x00\x00\x00\xC2\x11\x99\x9A\x00\x00\x00\x00\x00\x00\x00\x00?`c\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
...
```

kaareseras, 2022-12-14:
Confirm

drled, 2022-12-14:
confirm

kaareseras, 2022-12-14:
Installed on 8266, it works with no issues

janneman001, 2022-12-15:
It looks like same prob here:

**I get this after upgrade esphome 2022.12.0**
[19:27:45][W][modbus:105]: Modbus CRC Check failed! C071!=00
[19:27:45][W][modbus:105]: Modbus CRC Check failed! C071!=00
[19:27:45][W][modbus:105]: Modbus CRC Check failed! C071!=00
[19:27:45][W][modbus:105]: Modbus CRC Check failed! C071!=00
[19:27:45][W][modbus:105]: Modbus CRC Check failed! C071!=4842

**Instead of in esphome 2022:11.5:**
[19:30:28][D][sdm_meter:043]: SDMMeter Phase A: V=230.900 V, I=1.729 A, Active P=268.500 W, Apparent P=296.883 VA, Reactive P=-126.600 VAR, PF=0.904, PA=0.000 °
[19:30:28][D][sensor:127]: 'SDM120M Voltage': Sending state 230.89999 V with 0 decimals of accuracy
[19:30:28][D][sensor:127]: 'SDM120M Current': Sending state 1.72900 A with 0 decimals of accuracy
[19:30:28][D][sensor:127]: 'SDM120M Active Power': Sending state 268.50000 W with 0 decimals of accuracy
[19:30:28][D][sdm_meter:069]: SDMMeter: F=50.000 Hz, Im.A.E=699.068 Wh, Ex.A.E=148.747 Wh, Im.R.E=32.208 VARh, Ex.R.E=222.870 VARh, T.P=0.000 W

**relevant data from config:**

esp32:
  board: firebeetle32
  framework:
    type: arduino

# uart to ttl->rs485 circuit
uart:
  id: rs485_uart
  rx_pin: GPIO03
  tx_pin: GPIO01
  baud_rate: 2400
  stop_bits: 1

modbus:
  uart_id: rs485_uart
  id: modbus_sdm

sensor:

  - platform: sdm_meter
    modbus_id: modbus_sdm
    phase_a:
      current:
        name: "SDM120M Current"
        accuracy_decimals: 0
        id: sdm120m_current
      voltage:
        name: "SDM120M Voltage"
        accuracy_decimals: 0
        id: sdm120m_voltage
      active_power:
        name: "SDM120M Active Power"
        accuracy_decimals: 0
        id: sdm120m_active_power
    update_interval: 13s

I hope this helps. 
I reverted to 11.5 for now with this particular IoT device

bigfloor, 2022-12-15:
Same problem here, running in HA, after upgrading to 2022:11.5 to 2022:12.0 it broke my ESP32-RS485 setup with:
Modbus CRC errors.
Have not been able to revert back to 2022:11.5 (as I'm running in HA)

bigfloor, 2022-12-16:
I used an older ESPHOME from the CLI (11.2) on my Growatt/ESP/MAX3485 combo, now it works again. Not touching that  ESP32 anytime soon !

wauswaus, 2022-12-18:
I don't think it is related to SDM meter component but somewhere in underlaying code. I have a growatt solar invertor where I also communicate via modbus and there I also see the CRC errors popping up.

kaareseras, 2022-12-18:
I just set Up an ESP32 to communicate with my Deye inverter. Work without any problems. That integration uses uat and modbus, so I think this issue is related to this meter.

can anyone describe how to roll back esphome? I’m using haasio

janneman001, 2022-12-18:
I use a separate stdalone 11.5 esphome instance for the sdm_meter device. I copy over the yaml, the secrets and generate/update the device from there.

wauswaus, 2022-12-18:
> I just set Up an ESP32 to communicate with my Deye inverter. Work without any problems. That integration uses uat and modbus, so I think this issue is related to this meter.
> 
> can anyone describe how to roll back esphome? I’m using haasio

Then i'll make a separate ticket for the Growatt component as well

janneman001, 2022-12-20:
Tried again; now with esphome 2022.12.2 on device with sdm_meter component. Same result with crc errors like above.

I also tried again to update a device with growatt_solar and also modbus component. This update gave no problems. Works as expected.

For me it still looks like it must be something in sdm_meter component.

janneman001, 2022-12-21:
Tried again; now with esphome 2022.12.3 on device with sdm_meter component. Same result with crc errors like above.

I also tried again to update a device with growatt_solar and also modbus component. This update gave no problems. However NB i use RTU protocol not RTU2. RTU works as expected.

For me it still looks like it must be something in sdm_meter component or underlying modbus.

Reverted back to 2022.11.5 for this particular device with sdm_meter only.

wauswaus, 2022-12-23:
I also tried growatt_solar with 2022.12.3 and that did not fix it, still CRC issues. I am running RTU2 though.

johanvanderkuijl, 2022-12-23:
Same here. Had to revert to 2022.11.5 to make it work. Using esp32, sdm_meter, 2x sdm120 and ttl->rs485 circuit. Update interval 2 seconds.

johiodo, 2022-12-26:
Same here.
ESPHome version: 2022.12.3
SDM220M  - Modbus CRC Check failed!
Board: ESP32 devkitc v4

mishomvp, 2022-12-27:
It's not working for me either.
SDM230M - Modbus CRC Check failed!
Board: ESP32 devkitc v4

kadamski, 2022-12-28:
I have the same problem. The are CRC errors on 2022.12, but no errors on 2022.11.5.  As soon as I cherry-pick c2e198311ca73112fa7c9c213101277200085a6f to 2022.11.5, the CRC error appears.

This most likely means that the problem is outside of esphome, as this patch intruduces platformio/espressif32 v5.2.0 instead of v3.5.0. Looking at the changelog (https://github.com/platformio/platform-espressif32/releases), we can see that the esp-idf was bumped from v4.3.2 into v4.4.2 so it is likely that the problem is in esp-idf itself.

kadamski, 2022-12-28:
Here's how it looks like on an oscilloscope. The OK is from 2022.11.5, the NOK case is from 2022.12.3.
[image]

kadamski, 2022-12-28:
That is weird. After changing the ESP_LOGV into ESP_LOGD in the first log in parse_modbus_byte_ function, everything starts working. So basically this is the code that "fixes" the problem for me:

```
diff --git a/esphome/components/modbus/modbus.cpp b/esphome/components/modbus/modbus.cpp
index 4d75675d..83d113df 100644
--- a/esphome/components/modbus/modbus.cpp
+++ b/esphome/components/modbus/modbus.cpp
@@ -39,7 +39,7 @@ bool Modbus::parse_modbus_byte_(uint8_t byte) {
   size_t at = this->rx_buffer_.size();
   this->rx_buffer_.push_back(byte);
   const uint8_t *raw = &this->rx_buffer_[0];
-  ESP_LOGV(TAG, "Modbus received Byte  %d (0X%x)", byte, byte);
+  ESP_LOGD(TAG, "Modbus received Byte  %d (0X%x)", byte, byte);
   // Byte 0: modbus address (match all)
   if (at == 0)
     return true;
```

I, of course, have DEBUG log_level set. So adding some delay into the parse_modbus_byte_() function, makes it working correctly. That would indicate some timing issue but so far I don't understand why.

kadamski, 2022-12-28:
Also adding
modbus:
  send_wait_time: 500ms
  
also "fixes" the problem. I'm not sure why that is but the actual response comes after ~128ms from the moment when the register read command is sent. It also takes over 350ms for the response to be fully sent.
[image]

janneman001, 2022-12-29:
Thnx @kadamski. Testing your last work-around on esphome 2022.12.3 was low hanging fruit for me.
However the (send_wait_time: 500ms) work around does not work for me. Maybe it should go together with DEBUG setup. That having said it must be something with the half-duplex nature of modbus.

Testing the LOGD workaround i have not enough mileage with esphome for yet.

kadamski, 2022-12-29:
@janneman001 there might be a lot of variariation between setups so I'm not that surprised that this didn't work for you. To be honest, I still don't understand why it works for me so it might just be a matter of luck.

You can also try just changing the whole logging level to VERBOSE, in that case a new message will be printed for every new character received on modbus which in my case also slows things down a little and makes the modbus working.
If you still get a CRC error with VERBOSE logging, sharing those lines "Modbus received byte ..." preceeding the CRC error might be interesting.

kadamski, 2022-12-29:
Another way to fix this for me is to update the arduino framework to 2.0.6, by using:

```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino
    version: 2.0.6
```

Again, could be a luck or some fix because this arduino-esp32 upgrades esp-idf to v4.4.3 and also has some UART fixes.

mishomvp, 2022-12-29:
Hello,

Last night I tested the way @kadamski  wrote and it really works. (ESP32-EVB and SDM230M)

wauswaus, 2022-12-30:
Using the 2.0.6 did the trick for me. No need to do anything else. Thanks for pointing this out!

mishomvp, 2022-12-30:
Thank you @kadamski ! Upgrading to 2.0.6 easily solves the problem!

kaareseras, 2022-12-30:
When specifing 2.0.6 i get this error when compiling: *** [.pioenvs/rs485mterreader3/bootloader.bin] Source `/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/bin/bootloader__40m.elf' not found, needed by target `.pioenvs/rs485mterreader3/bootloader.bin'. 

What am I doing wrong?

mishomvp, 2022-12-30:
Мore details? Your yaml configuration without sensitive data. What architecture are you compiling it on.

kaareseras, 2022-12-30:
I got it working now,  :-)

dxxxm, 2022-12-31:
Thank you @kadamski. Using arduino framework version 2.0.6 works ok for esp32 and both SDM120 & 630

bigfloor, 2023-01-02:
@kadamski  Worked for me as well !! (Growatt-Modbus)

saschaludwig, 2023-01-05:
Looks like a timing related issue. I also have the same problems.

Maybe related to this: https://github.com/esphome/issues/issues/3909

fozzyg, 2023-01-13:
I have the same CRC  errors .  I have 2 Modbus esp32 , both  are not working anymore,  
I can  not install the 2.0.6,  

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.6

I get this error :
Processing esp32-zolder (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Error: Could not find one of 'package.json' manifest files in the package

How do I get the Arduino  2.0.6

Geert10, 2023-01-14:
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino
    platform_version: platformio/espressif32 @ 3.5.0
    version: 1.0.6
 
works for me

fozzyg, 2023-01-14:
> 

Thank you @Geert10 
Everything  works perfect now

Flappiee, 2023-01-20:
I have a new Wemos D1 board (ESP8266).
I'm having problems with reading out an Eastron SDM120 and getting CRC errors.
I installed ESPEasy and then everything is fine.

Is that related to this issue?

Roos-AID, 2023-01-24:
> Thank you @kadamski ! Upgrading to 2.0.6 easily solves the problem!

SDM220 still has CRC errors with 2.0.6 and with 2.0.3 .  Still trying to find a version that works.

mishomvp, 2023-01-24:
@Roos-AID What controller are you using, because I use ESP32 and SDM220 and have no problem. Of course with 2.0.6.

Roos-AID, 2023-01-25:
I have 
esp32:
  board: lolin32
  framework:
    type: arduino

It is intermittent, when I boot multiple times, there are periods that it is ok.  As soon as it is OK it stays OK until next reboot.

I was running on 2.0.2  and problem seemed gone.  However, when I removed debug code, then it returned.  
I added debug code, and changed to 2.0.6 and voila it seemed ok again.  It was with 2.0.6 running ok for some time until reboot. But as soon as I rebooted it failed again.  

I will now first replace the HW and modbus wiring to make sure it is not causing this.

Flappiee, 2023-01-27:
When you have issues, try ESPEasy on your board. 
My board only had CRC errors with ESPHome and ESPEasy is working fine. 
This way you can check if there is something wrong with the board.

janneman001, 2023-02-02:
Tested today with esphome 2022.12.8. Still crc errors with sdm.

Reverting back for this particular instance to 2022.11.5

Geert10, 2023-02-13:
I dont use the SDM platform anymore
use the modbus controller
for sdm120 use baudrate 2400

uart:
  id: modbus_serial
  tx_pin: 17
  rx_pin: 16
  baud_rate: 9600
  stop_bits: 1

modbus:
  uart_id: modbus_serial
  id: sdm_modbus

modbus_controller:
  modbus_id: sdm_modbus
#  command_throttle: 200ms
  id: sdm220
  update_interval: 3s 
  address: 1
  
sensor:
  - id: volt
    platform: modbus_controller  
    name: "SDM220M Voltage"
    address: 0x0000
    register_count: 2
    unit_of_measurement: "V"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: voltage
    
  - id: amp
    platform: modbus_controller  
    name: "SDM220M Current"
    address: 0x0006
    register_count: 2
    unit_of_measurement: "A"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: current

  - id: power
    platform: modbus_controller  
    name: "SDM220M Power"
    address: 0x000C
    register_count: 2
    unit_of_measurement: "W"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: power

  - id: apppower
    platform: modbus_controller  
    name: "SDM220M Apparent Power"
    address: 0x0012
    register_count: 2
    unit_of_measurement: "VA"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: power

  - id: exportpower
    platform: modbus_controller  
    name: "SDM220M Export Active Energy"
    address: 0x004A
    register_count: 2
    unit_of_measurement: "kWh"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing

  - id: importpower
    platform: modbus_controller  
    name: "SDM220M Import Active Energy"
    address: 0x0048
    register_count: 2
    unit_of_measurement: "kWh"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing

.....

Geert10, 2023-02-13:
uart:
  id: modbus_serial
  tx_pin: 17
  rx_pin: 16
  baud_rate: 9600
  stop_bits: 1

modbus:
  uart_id: modbus_serial
  id: sdm_modbus

modbus_controller:
  modbus_id: sdm_modbus
  id: sdm220
  update_interval: 3s 
  address: 1
  
sensor:
  - id: volt
    platform: modbus_controller  
    name: "SDM220M Voltage"
    address: 0x0000
    register_count: 2
    unit_of_measurement: "V"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: voltage
    
  - id: amp
    platform: modbus_controller  
    name: "SDM220M Current"
    address: 0x0006
    register_count: 2
    unit_of_measurement: "A"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: current

  - id: power
    platform: modbus_controller  
    name: "SDM220M Power"
    address: 0x000C
    register_count: 2
    unit_of_measurement: "W"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: power

  - id: apppower
    platform: modbus_controller  
    name: "SDM220M Apparent Power"
    address: 0x0012
    register_count: 2
    unit_of_measurement: "VA"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: power

  - id: exportpower
    platform: modbus_controller  
    name: "SDM220M Export Active Energy"
    address: 0x004A
    register_count: 2
    unit_of_measurement: "kWh"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing

  - id: importpower
    platform: modbus_controller  
    name: "SDM220M Import Active Energy"
    address: 0x0048
    register_count: 2
    unit_of_measurement: "kWh"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing

Geert10, 2023-02-13:
I dont use the sdm platform anymore
use modbus controller
for SDM120 baudrate 2400

uart:
  id: modbus_serial
  tx_pin: 17
  rx_pin: 16
  baud_rate: 9600
  stop_bits: 1

modbus:
  uart_id: modbus_serial
  id: sdm_modbus

modbus_controller:
  modbus_id: sdm_modbus
  id: sdm220
  update_interval: 3s 
  address: 1
  
sensor:
  - id: volt
    platform: modbus_controller  
    name: "SDM220M Voltage"
    address: 0x0000
    register_count: 2
    unit_of_measurement: "V"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: voltage
    
  - id: amp
    platform: modbus_controller  
    name: "SDM220M Current"
    address: 0x0006
    register_count: 2
    unit_of_measurement: "A"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: current

  - id: power
    platform: modbus_controller  
    name: "SDM220M Power"
    address: 0x000C
    register_count: 2
    unit_of_measurement: "W"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: power

  - id: apppower
    platform: modbus_controller  
    name: "SDM220M Apparent Power"
    address: 0x0012
    register_count: 2
    unit_of_measurement: "VA"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: power

  - id: exportpower
    platform: modbus_controller  
    name: "SDM220M Export Active Energy"
    address: 0x004A
    register_count: 2
    unit_of_measurement: "kWh"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing

  - id: importpower
    platform: modbus_controller  
    name: "SDM220M Import Active Energy"
    address: 0x0048
    register_count: 2
    unit_of_measurement: "kWh"
    register_type: read
    value_type: FP32
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing

karllinder, 2023-02-17:
Looks like the SDM component is now broken, in 2023.2.x

Regardless setting of the new disable_crc check on Modbus.
With 2022.11.x no issues, upgrade to 2023.2.2 I get no reading on the modbus, regarding of the settings of disable_crc.

Config, only showing one sdm sensor.
```
uart:
  rx_pin: 16
  tx_pin: 17
  baud_rate: 9600
  stop_bits: 1

modbus:
  flow_control_pin: 5
  disable_crc: true

sensor:
  - platform: sdm_meter
    phase_a:
      current:
        name: "SDM630 Phase A Current"
```
Will result with 2023.2.2:
```
[11:54:49][V][modbus:199]: Modbus write: 01.04.00.00.00.50.F0.36 (8)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  66 (0X42)
[11:54:49][D][modbus:106]: Modbus CRC Check failed, but ignored! C071!=4200
[11:54:49][W][modbus:135]: Got Modbus frame from unknown address 0x00! 
[11:54:49][V][modbus:042]: Modbus received Byte  140 (0X8c)
[11:54:49][V][modbus:042]: Modbus received Byte  52 (0X34)
[11:54:49][V][modbus:042]: Modbus received Byte  24 (0X18)
[11:54:49][V][modbus:042]: Modbus received Byte  63 (0X3f)
[11:54:49][V][modbus:042]: Modbus received Byte  127 (0X7f)
[11:54:49][V][modbus:042]: Modbus received Byte  157 (0X9d)
[11:54:49][V][modbus:042]: Modbus received Byte  50 (0X32)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  64 (0X40)
[11:54:49][V][modbus:042]: Modbus received Byte  228 (0Xe4)
[11:54:49][V][modbus:042]: Modbus received Byte  51 (0X33)
[11:54:49][V][modbus:042]: Modbus received Byte  33 (0X21)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  66 (0X42)
[11:54:49][V][modbus:042]: Modbus received Byte  71 (0X47)
[11:54:49][V][modbus:042]: Modbus received Byte  252 (0Xfc)
[11:54:49][V][modbus:042]: Modbus received Byte  75 (0X4b)
[11:54:49][V][modbus:042]: Modbus received Byte  70 (0X46)
[11:54:49][V][modbus:042]: Modbus received Byte  7 (0X7)
[11:54:49][V][modbus:042]: Modbus received Byte  169 (0Xa9)
[11:54:49][V][modbus:042]: Modbus received Byte  174 (0Xae)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][D][modbus:106]: Modbus CRC Check failed, but ignored! 7C28!=00
[11:54:49][W][modbus:135]: Got Modbus frame from unknown address 0x8C! 
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:49][V][modbus:042]: Modbus received Byte  68 (0X44)
[11:54:49][V][modbus:042]: Modbus received Byte  20 (0X14)
[11:54:49][V][modbus:042]: Modbus received Byte  103 (0X67)
[11:54:49][V][modbus:042]: Modbus received Byte  76 (0X4c)
[... 107 lines cut ...]
[11:54:50][V][modbus:042]: Modbus received Byte  15 (0Xf)
[11:54:50][V][modbus:042]: Modbus received Byte  93 (0X5d)
[11:54:50][V][modbus:042]: Modbus received Byte  64 (0X40)
[11:54:50][V][modbus:042]: Modbus received Byte  198 (0Xc6)
[11:54:50][V][modbus:042]: Modbus received Byte  23 (0X17)
[11:54:50][V][modbus:042]: Modbus received Byte  12 (0Xc)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  68 (0X44)
[11:54:50][V][modbus:042]: Modbus received Byte  153 (0X99)
[11:54:50][V][modbus:042]: Modbus received Byte  21 (0X15)
[11:54:50][V][modbus:042]: Modbus received Byte  34 (0X22)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  0 (0X0)
[11:54:50][V][modbus:042]: Modbus received Byte  68 (0X44)
[11:54:50][V][modbus:042]: Modbus received Byte  154 (0X9a)
[11:54:50][V][modbus:042]: Modbus received Byte  223 (0Xdf)
[11:54:50][V][modbus:042]: Modbus received Byte  180 (0Xb4)
[11:54:50][V][component:200]: Component modbus took a long time for an operation (0.93 s).
[11:54:50][V][component:201]: Components should block for at most 20-30ms.
```

gkasprow, 2023-02-18:
I had issues with ESP8266 and MODBUS on Arduino natively. I traced it down to the wires using scope. The DE line of RS485 converter was deactivated while the UART was still sending data! The UART is buffered so some combination of baud rate, RC components (including parasitic capacitors) may cause it to work. The simple fix is either send one more dummy byte or wait with DE line deactivation.

gkasprow, 2023-02-18:
That problem probably does not appear with TTL-RS485 converters with auto-sensing.

bigfloor, 2023-02-21:
Fix is to change the Arduino version in your ESPhome Yaml file to use version 2.0.6:
[https://github.com/esphome/issues/issues/3937#issuecomment-1367844174]

majuss, 2023-02-21:
@gkasprow can you confirm this? For me still nothing is working.

janneman001, 2023-02-26:
Tested today with esphome 2023.2.4. Still crc errors with sdm.

Reverting back for this particular instance to 2022.11.5

janneman001, 2023-07-28:
Just to let you know:

Tested today with esphome 2023.7.0. Still crc errors with sdm.

Reverting back again for this particular instance to 2022.11.5

janneman001, 2023-07-31:
A temporary workaround to work with current version of esphome would be to read from the modbus with modbus_controller instead of sdm_meter.

example: 
sensor:
  - platform: modbus_controller
    name: "${master_name} AC Voltage"
    modbus_controller_id: $master_id
    id: "${master_id}_ac_volt"
    register_type: "read"
    value_type: FP32
    register_count: 2
    response_size: 4
    offset: 0
    address: 0x0
    unit_of_measurement: V
    device_class: VOLTAGE
    state_class: measurement
    entity_category: diagnostic
    icon: mdi:flash
    accuracy_decimals: 1

and so on.

gkasprow, 2023-08-22:
I'm using modbus_controller and CRC errors are still there
```

uart:
  id: mod_bus
  tx_pin: GPIO13
  rx_pin: GPIO39
  baud_rate: 9600
  stop_bits: 1

modbus:
  flow_control_pin: GPIO12
  #send_wait_time: 500ms
  id: modbus1

modbus_controller:
  - id: epever
    ## the Modbus device addr
    address: 0x1
    modbus_id: modbus1
    command_throttle: 100ms
    setup_priority: -10
    update_interval: 5.3s

sensor:

  - platform: modbus_controller
    modbus_controller_id: epever
    name: "ACC_CH1"
    unit_of_measurement: "V"
    register_type: holding
    address: 1
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
        - offset: 0
        - multiply: 1
        - calibrate_linear:
          - 0.0 -> 0.0
          - 1024.0 -> 5
```

gkasprow, 2023-09-03:
I finally made modbus working on ESP32 without CRC errors.
There were 2 independent issues:
- The UART-RS485 converter uses NE555 + 3k9+10n timing circuit to control DE line. It causes weird shape of the drive signal for 9600 baud. With much faster baud rates that works. I increased capacitor to 100nF and shape looks correct now even for 9600.
- Lack of biasing. That caused glitch at the begining of the response frame. This glitch was sometimes ignored by UART, that's why sometimes transmission was OK. I used 2x680R bias to make sure there is default state H on the bus.
[image]

ruimarinho, 2023-09-03:
Were those mods done exclusively on the UART-RS485 converter?

gkasprow, 2023-09-03:
Yes

Adminius, 2023-10-19:
I had same CRC errors with SDM72DM-V2, but not with SDM630MCT.
After some try&error on HW side, I switched to direct MODBUS communictaion without using SDM-component as mentioned before:
https://github.com/esphome/issues/issues/3912#issuecomment-1427387494

now it works as expected.
some useful code I used: https://gitlab.com/-/snippets/2483164

foorschtbar, 2023-11-21:
Same problem here. Updated to latest version and now the failed CRC messages :(

xxVDRxx, 2023-12-24:
I confirmed. This problem still occured.  I use Wemos D1 mini and actually since upgrade 2023.9.3 to 2023.11.6 have problem with sdm_meter platform and SDM630 :/

[22 more replies]

closed 2024-10-29 by jesserockz