esphome/issues #3916: MAX7219 Digit Display - Not working anymore

majoto85, 2022-12-14:
### The problem

Hello,

I updated to the latest ESPHome version (2022.12.0) and the MAX7219 digit display no longer works. I've reduced everything to the bare essentials to exclude other sources of error, but even a simple "Hello" is no longer displayed.

Thank you

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP32

### Board

NodeMCU

### Component causing the issue

MAX7219 Digit Display

### Example YAML snippet

```yaml
display:
  - platform: max7219digit
    cs_pin: 5
    num_chips: 4
    intensity: 15
    id: my_display
    lambda: |-
            it.print(0, 0, id(digit_font), "HELLO!");
```

### Anything in the logs that might be useful for us?

```txt
No
```

jtbnz, 2022-12-18:
I also have this issue after flashing via serial.  No errors in logs.  Last displayed info was left on the display until a power cycle.

jtbnz, 2022-12-20:
Trying to debug further, using log level very_verbose no errors are generated, the lamda is being executed.

display:
  - platform: max7219digit
    cs_pin: GPIO21
    num_chips: 4
    intensity: 15
    
    lambda: |-
      ESP_LOGD("custom","Before Print");
      it.print(0, 0, id(my_font), "123");
      ESP_LOGD("custom","After Print");

From Logfile:
[14:24:51][VV][scheduler:196]: Running interval 'update' with interval=500 last_execution=28994 (now=29495)
[14:24:51][D][custom:091]: Before Print
[14:24:51][D][custom:093]: After Print
[14:24:52][VV][scheduler:196]: Running interval 'update' with interval=500 last_execution=29494 (now=29997)
[14:24:52][D][custom:091]: Before Print
[14:24:52][D][custom:093]: After Print

Full log attached
[logs_garagetimer_logs.txt](https://github.com/esphome/issues/files/10263945/logs_garagetimer_logs.txt)

jtbnz, 2023-01-25:
Changing the cs_pin from GPIO21 to GPIO17 and clk_pin from GPIO19 to GPIO04 fixed this for me. 

3X ESP32 Dev kit C V2 has this issue and was fixed by moving pins