esphome/issues #3917: RP2040 with BSEC Software Library fails compiling

explo910, 2022-12-14:
### The problem

I am not sure if this problem is related to ESPHome or Platform.io. Please let me know so I can create the problem in the correct project.

When I try to compile a firmware for the Raspberry Pi Pico W with a bme680_bsec component, the compilation fails.
I tried the Home Assistant addon and the pip installation with the same error:

|-- BSEC Software Library @ 1.6.1480
|   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|-- MD5Builder @ 1.0.0
Linking /data/testesphome/.pioenvs/testesphome/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: cannot find -lalgobsec
collect2: error: ld returned 1 exit status
*** [/data/testesphome/.pioenvs/testesphome/firmware.elf] Error 1

With the pip version, I was able to isolate and fix the error by manually deleting libalgobsec.a from:
.esphome\build\testesphome\.piolibdeps\testesphome\BSEC Software Library\src\cortex-m0plus
to:
.esphome\build\testesphome\.piolibdeps\testesphome\BSEC Software Library\src\rp2040

After that, the compilation went through and worked after flashing to the Pico W.

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

bme680_bsec

### Example YAML snippet

```yaml
esphome:
  name: testesphome

rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  password: "xxx"

wifi:
  ssid: "xxx"
  password: "xxx"

  ap:
    ssid: "Testesphome Fallback Hotspot"
    password: "xxx"

i2c:
  sda: GPIO20
  scl: GPIO21
  
# Example configuration entry
bme680_bsec:
    address: 0x76
    temperature_offset: 0
    iaq_mode: static
    sample_rate: lp
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    gas_resistance:
      name: "BME680 Gas Resistance"
    iaq:
      name: "BME680 IAQ"
    iaq_accuracy:
      name: "BME680 Numeric IAQ Accuracy"
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"
```

### Anything in the logs that might be useful for us?

```txt
Building in release mode
arm-none-eabi-g++ -o .pioenvs\testesphome\firmware.elf -T C:\Users\myuser\Desktop\.esphome\build\testesphome\.pioenvs\testesphome\memmap_default.ld -march=armv6-m -mcpu=cortex-m0plus -mthumb @C:\Users\myuser\.platformio\packages\framework-arduinopico\lib\platform_wrap.txt -u_printf_float -u_scanf_float -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common .pioenvs\testesphome\FrameworkArduinoVariant\picow_digital.cpp.o .pioenvs\testesphome\FrameworkArduinoVariant\picow_init.cpp.o .pioenvs\testesphome\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o .pioenvs\testesphome\src\esphome\components\api\api_connection.cpp.o .pioenvs\testesphome\src\esphome\components\api\api_frame_helper.cpp.o .pioenvs\testesphome\src\esphome\components\api\api_pb2.cpp.o .pioenvs\testesphome\src\esphome\components\api\api_pb2_service.cpp.o .pioenvs\testesphome\src\esphome\components\api\api_server.cpp.o .pioenvs\testesphome\src\esphome\components\api\list_entities.cpp.o .pioenvs\testesphome\src\esphome\components\api\proto.cpp.o .pioenvs\testesphome\src\esphome\components\api\subscribe_state.cpp.o .pioenvs\testesphome\src\esphome\components\api\user_services.cpp.o .pioenvs\testesphome\src\esphome\components\bme680_bsec\bme680_bsec.cpp.o .pioenvs\testesphome\src\esphome\components\i2c\i2c.cpp.o .pioenvs\testesphome\src\esphome\components\i2c\i2c_bus_arduino.cpp.o .pioenvs\testesphome\src\esphome\components\i2c\i2c_bus_esp_idf.cpp.o .pioenvs\testesphome\src\esphome\components\logger\logger.cpp.o .pioenvs\testesphome\src\esphome\components\md5\md5.cpp.o .pioenvs\testesphome\src\esphome\components\mdns\mdns_component.cpp.o .pioenvs\testesphome\src\esphome\components\mdns\mdns_esp32_arduino.cpp.o .pioenvs\testesphome\src\esphome\components\mdns\mdns_esp8266.cpp.o .pioenvs\testesphome\src\esphome\components\mdns\mdns_esp_idf.cpp.o .pioenvs\testesphome\src\esphome\components\mdns\mdns_rp2040.cpp.o .pioenvs\testesphome\src\esphome\components\network\util.cpp.o .pioenvs\testesphome\src\esphome\components\ota\ota_backend_arduino_esp32.cpp.o .pioenvs\testesphome\src\esphome\components\ota\ota_backend_arduino_esp8266.cpp.o .pioenvs\testesphome\src\esphome\components\ota\ota_backend_arduino_rp2040.cpp.o .pioenvs\testesphome\src\esphome\components\ota\ota_backend_esp_idf.cpp.o .pioenvs\testesphome\src\esphome\components\ota\ota_component.cpp.o .pioenvs\testesphome\src\esphome\components\rp2040\core.cpp.o .pioenvs\testesphome\src\esphome\components\rp2040\gpio.cpp.o .pioenvs\testesphome\src\esphome\components\rp2040\preferences.cpp.o .pioenvs\testesphome\src\esphome\components\sensor\automation.cpp.o .pioenvs\testesphome\src\esphome\components\sensor\filter.cpp.o .pioenvs\testesphome\src\esphome\components\sensor\sensor.cpp.o .pioenvs\testesphome\src\esphome\components\socket\bsd_sockets_impl.cpp.o .pioenvs\testesphome\src\esphome\components\socket\lwip_raw_tcp_impl.cpp.o .pioenvs\testesphome\src\esphome\components\socket\socket.cpp.o .pioenvs\testesphome\src\esphome\components\text_sensor\filter.cpp.o .pioenvs\testesphome\src\esphome\components\text_sensor\text_sensor.cpp.o .pioenvs\testesphome\src\esphome\components\wifi\wifi_component.cpp.o .pioenvs\testesphome\src\esphome\components\wifi\wifi_component_esp32_arduino.cpp.o .pioenvs\testesphome\src\esphome\components\wifi\wifi_component_esp8266.cpp.o .pioenvs\testesphome\src\esphome\components\wifi\wifi_component_esp_idf.cpp.o .pioenvs\testesphome\src\esphome\components\wifi\wifi_component_pico_w.cpp.o .pioenvs\testesphome\src\esphome\core\application.cpp.o .pioenvs\testesphome\src\esphome\core\color.cpp.o .pioenvs\testesphome\src\esphome\core\component.cpp.o .pioenvs\testesphome\src\esphome\core\component_iterator.cpp.o .pioenvs\testesphome\src\esphome\core\controller.cpp.o .pioenvs\testesphome\src\esphome\core\entity_base.cpp.o .pioenvs\testesphome\src\esphome\core\helpers.cpp.o .pioenvs\testesphome\src\esphome\core\log.cpp.o .pioenvs\testesphome\src\esphome\core\scheduler.cpp.o .pioenvs\testesphome\src\esphome\core\util.cpp.o .pioenvs\testesphome\src\main.cpp.o "-L.piolibdeps\testesphome\BSEC Software Library\src\rp2040" -L.pioenvs\testesphome -Wl,--start-group -lalgobsec .pioenvs\testesphome\lib18d\libMD5Builder.a .pioenvs\testesphome\libe8f\libLittleFS.a .pioenvs\testesphome\lib50f\libPicoOTA.a .pioenvs\testesphome\lib450\libUpdater.a .pioenvs\testesphome\lib486\libSPI.a .pioenvs\testesphome\lib304\liblwIP_CYW43.a .pioenvs\testesphome\lib30c\liblwIP_Ethernet.a .pioenvs\testesphome\lib6a4\libWiFi.a .pioenvs\testesphome\lib52d\libLEAmDNS.a .pioenvs\testesphome\lib80a\libsodium.a .pioenvs\testesphome\lib8b1\libnoise-c.a .pioenvs\testesphome\lib602\libWire.a ".pioenvs\testesphome\liba3b\libBSEC Software Library.a" .pioenvs\testesphome\libFrameworkArduino.a C:\Users\myuser\.platformio\packages\framework-arduinopico\lib\ota.o C:\Users\myuser\.platformio\packages\framework-arduinopico\lib\libpico.a C:\Users\myuser\.platformio\packages\framework-arduinopico\lib\libbearssl.a -lm -lc -lstdc++ -lc -Wl,--end-group
c:/users/myuser/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: cannot find -lalgobsec
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\testesphome\firmware.elf] Error 1
```

### Additional information

I tried ESPHome 2022.11.4 to 2022.12.0

emefff, 2022-12-15:
Sorry to step in, but this is kind of related as the BME680 is an I2C device: does I2C work in your setup (referring to https://github.com/esphome/issues/issues/3838)?
What happens if you set

`platform: bme680`

(without BSEC)?

explo910, 2022-12-15:
I tried it before with the BME680 component and had no issues. 
I will try to find my yaml file and update you on my findings in #3838.

This issue here seems only to happen with the BSEC Library.

avaneerd, 2022-12-25:
On Windows I was able to compile it by adding a few build flags.
```
esphome:
  ...
  platformio_options:
    build_flags: 
      - -I .piolibdeps/<device-name>/BSEC\ Software\ Library/src/inc
      - -L .piolibdeps/<device-name>/BSEC\ Software\ Library/src/cortex-m0plus
      - -lalgobsec
```

I'm trying to make it work from the ESPHome dashboard in Home Assistant but I'm struggling to get that to work :x

rotinom, 2022-12-31:
Any luck getting this working?  Seeing the same thing.

swoga, 2023-01-12:
@avaneerd: Thank you for sharing your findings.
For it to work in the ESPHome Dashboard an absolute path must be specified:
```
esphome:
  name: <device-name>
  platformio_options:
    build_flags: 
      - -I /data/<device-name>/.piolibdeps/<device-name>/BSEC\ Software\ Library/src/inc
      - -L /data/<device-name>/.piolibdeps/<device-name>/BSEC\ Software\ Library/src/cortex-m0plus
      - -lalgobsec
```

neffs, 2023-03-18:
This is an issue with the BSEC Library. It's probably broken for all ARM chips. I just created a Pull Request for BSEC2: https://github.com/boschsensortec/Bosch-BSEC2-Library/pull/22

rotinom, 2023-03-18:
I concluded the same. Just not enough time to investigate further. Looks like the last drop of the bsec lib broke esp32 arch. The binaries are not present in the distro. --dw

pszober, 2023-04-11:
Hi,

I'm having the same issue but I'm unable to fix it using a method suggested by @swoga.

Setup:
HA Supervised 2023.4.2 running on ODROID-N2
ESP32-S3 WROOM1 chip with BME680 sensor
Compiling Esphome code using Esphome add-on in HA

It all worked fine when using bme680 platform without BSEC. After switching to BSEC library, I got the following error:
```
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find -lalgobsec
collect2: error: ld returned 1 exit status
*** [/data/multisensor-01/.pioenvs/multisensor-01/firmware.elf] Error 1
```

After adding platformio_options, cleaning files and compiling I was getting this:
```
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: skipping incompatible /data/multisensor-01/.piolibdeps/multisensor-01/BSEC Software Library/src/cortex-m0plus/libalgobsec.a when searching for -lalgobsec
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find -lalgobsec
collect2: error: ld returned 1 exit status
*** [/data/multisensor-01/.pioenvs/multisensor-01/firmware.elf] Error 1
```

My code:
```
esphome:
  name: multisensor-01
  friendly_name: Multi-Sensor 1
  platformio_options:
#    board_upload.flash_size: 2MB
#    board_upload.maximum_size: 4194304
#    board_build.partitions: "minimal.csv"
#    board_build.flash_mode: dio
    build_flags: 
      - -I /data/multisensor-01/.piolibdeps/multisensor-01/BSEC\ Software\ Library/src/inc
      - -L /data/multisensor-01/.piolibdeps/multisensor-01/BSEC\ Software\ Library/src/cortex-m0plus
      - -lalgobsec

esp32:
  board: adafruit_feather_esp32s3_tft
  variant: esp32s3
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.1.95

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Multi-Sensor 01 Fallback Hotspot"
    password: !secret ap_password

captive_portal:

i2c:
  - id: bus_a
    sda: 14
    scl: 13
    scan: false

#sensor:
#  - platform: bme680
#    temperature:
#      name: "BME680 Temperature"
#      oversampling: 16x
#    pressure:
#      name: "BME680 Pressure"
#    humidity:
#      name: "BME680 Humidity"
#    gas_resistance:
#      name: "BME680 Gas Resistance"
#    address: 0x77
#    update_interval: 60s

#switch:
#  - platform: gpio
#    pin: GPIO11
#    id: gpio11
#    restore_mode: ALWAYS_OFF

bme680_bsec:

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    iaq:
      name: "BME680 IAQ"
      id: iaq
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"

  - platform: template
    name: "BME680 IAQ Classification"
    icon: "mdi:checkbox-marked-circle-outline"
    lambda: |-
      if ( int(id(iaq).state) <= 50) {
        return {"Excellent"};
      }
      else if (int(id(iaq).state) >= 51 && int(id(iaq).state) <= 100) {
        return {"Good"};
      }
      else if (int(id(iaq).state) >= 101 && int(id(iaq).state) <= 150) {
        return {"Lightly polluted"};
      }
      else if (int(id(iaq).state) >= 151 && int(id(iaq).state) <= 200) {
        return {"Moderately polluted"};
      }
      else if (int(id(iaq).state) >= 201 && int(id(iaq).state) <= 250) {
        return {"Heavily polluted"};
      }
      else if (int(id(iaq).state) >= 251 && int(id(iaq).state) <= 350) {
        return {"Severely polluted"};
      }
      else if (int(id(iaq).state) >= 351) {
        return {"Extremely polluted"};
      }
      else {
        return {"error"};
      }
```

Any ideas?

pszober, 2023-04-11:
Amazing! Works like a charm. Thank you so much.

TamasKazsoki, 2023-04-11:
That fix is not working to me unfortunately:

Archiving /data/esp-ikea4/.pioenvs/esp-ikea4/libFrameworkArduino.a
Linking /data/esp-ikea4/.pioenvs/esp-ikea4/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: cannot find -lalgobsec
collect2: error: ld returned 1 exit status
*** [/data/esp-ikea4/.pioenvs/esp-ikea4/firmware.elf] Error 1
========================= [FAILED] Took 108.40 seconds =========================

Where did you put the lalgobsec.a file?

chaudi, 2023-10-06:
Started playing around with updating bsec/bsec2 libraries myself to see if I can get it working since it fails for me too with the above error.
But Im not sure where to place the libalgobsec.a file that is required, using esphome dashboard.

gretel, 2023-10-06:
> Started playing around with updating bsec/bsec2 libraries myself to see if I can get it working since it fails for me too with the above error. But Im not sure where to place the libalgobsec.a file that is required, using esphome dashboard.

not sure but i think you have to copy the file around manually for now: https://github.com/boschsensortec/BSEC-Arduino-library/pull/57

closed 2026-03-12 by bdraco