esphome/issues #3919: xtensa-esp32-elf-g++: not found on 2022.12.0

slydiman, 2022-12-15:
### The problem

After updating the ESPHome HA addon to 2022.12.0, I get the following error:
```
sh: 1: xtensa-esp32-elf-g++: not found
```

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP32

### Board

nodemcu

### Example YAML snippet

```yaml
substitutions:
  name: ble-k
  wifi_ap_password: ""

packages:
  esphome.ble-k-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

esphome:
  name: ble-k
  name_add_mac_suffix: false
  project:
    name: esphome.ble-k-proxy
    version: "1.0"  

esp32:
#  board: nodemcu-32s
  board: esp-wrover-kit
  framework:
    type: arduino
 
wifi:
  ssid: "xxx"
  password: "abc"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ble_k_fallback"
    password: "ABC"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:

improv_serial:

web_server:
  port: 80

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
  
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:

binary_sensor:
  - platform: ble_presence
    mac_address: '00:11:22:33:44:55'
    name: MiBand3_k

sensor:
  - platform: ble_rssi
    mac_address: '00:11:22:33:44:55'
    name: MiBand3_k_rssi

  - platform: uptime
    name: ble_k_uptime

  - platform: wifi_signal
    name:  ble_k_wifi
    update_interval: 60s

switch:
  - platform: restart
    name: ble_k_restart

button:
  - platform: safe_mode
    name: ble_k_safe_mode_boot
    entity_category: diagnostic
```

### Anything in the logs that might be useful for us?

```txt
Here is the full log of the clean install

INFO Reading configuration /config/esphome/ble-k.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ble-k (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.2.0 has been installed!
INFO espressif32@5.2.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40201.0
INFO Installing platformio/tool-esptoolpy @ ~1.40201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40201.0 has been installed!
INFO tool-esptoolpy@1.40201.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40400.0
INFO Installing platformio/tool-scons @ ~4.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40400.0 has been installed!
INFO tool-scons@4.40400.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
Compiling /data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/ble-k/.pioenvs/ble-k/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================= [FAILED] Took 193.73 seconds =========================
```
```

### Additional information

Note I see the file /usr/share/hassio/addons/data/5c53de3b_esphome/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ (size 778104 bytes). 
Probably something is wrong with the PATH env.

swifty99, 2022-12-15:
same here:
```
INFO Reading configuration /config/esphome/kwl.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing kwl (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
Compiling /data/kwl/.pioenvs/kwl/src/esphome/components/adc/adc_sensor.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/kwl/.pioenvs/kwl/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/kwl/.pioenvs/kwl/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/kwl/.pioenvs/kwl/src/esphome/components/api/api_pb2.cpp.o
*** [/data/kwl/.pioenvs/kwl/src/esphome/components/adc/adc_sensor.cpp.o] Error 127
*** [/data/kwl/.pioenvs/kwl/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [/data/kwl/.pioenvs/kwl/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/kwl/.pioenvs/kwl/src/esphome/components/api/api_pb2.cpp.o] Error 127
========================== [FAILED] Took 6.15 seconds ==========================
```
on a configuration that had no problems before:
```
esphome:
  name: ${plug_name}
  platform: ESP32
  board: nodemcu-32s
```

JackTalisker, 2022-12-15:
same as #3904

ssieb, 2022-12-15:
yes, duplicate of https://github.com/esphome/issues/issues/3904

closed 2022-12-15 by ssieb

nliaudat, 2023-01-20:
That's related to old libs on platformio cache.

To fix it on windows : 
`cd C:\Users\%USERNAME%\AppData\Local\Programs\Python\Python310\Lib\site-packages`
delete platformio & platformio-6.1.5.dist-info
`cd In C:\Users\%USERNAME%\`
delete .platformio

run : 
```
pip install --upgrade esphome
pip install --upgrade platformio
```

Samu84, 2023-01-25:
Hello! I found a way to fix this problem, when you install a docker on RPI with 32bit OS but kernel is modificated to 64bit (like me), when you install esphome docker using name image: esphome/esphome:2022.11, is going to download the image for 32bit, instead you need to force the install of the 64bit image using SHA256 for example :
image: esphome/esphome:2022.11@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d

peterfortuin, 2023-01-29:
Deleting the platformio cache solved my problem.

alpat59, 2023-02-16:
> Hello! I found a way to fix this problem, when you install a docker on RPI with 32bit OS but kernel is modificated to 64bit (like me), usually you install esphome docker using name image: esphome/esphome:2022.11, is going to download the image for 32bit, instead you need to force the install of the 64bit image using SHA256 for example :
> image: esphome/esphome:2022.11@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d

@Samu84 Pls could you tell me if this fix is applicable also when ESPHome is installed as HA Add-on ? Thank you

Samu84, 2023-02-16:
I am not using HA Add-on, if you mean add-on installed as a docker, probably yes, i always advice to make a backup of the config files YAML, to prevent risk to loose all the settings

Travis-HA, 2023-03-15:
To fix this error, I uninstalled the ESPhome Addon, reinstalled through Home Assistant Addon page and the issue disappeared.  None of the yaml configurations for any of my Spain devices were lost by doing this.

nogueirasa, 2023-04-08:
Hi, same problem here.

Trying to install Blackymas/NSPanel_HA_Blueprint

ESPHome:  v2023.3.2 (docker)
Home Assiatnt: 2023.4.1 (docker)
Raspberry Pi 4 OS 32bits

nogueirasa, 2023-04-09:
Tested vesion esphome/esphome:latest, but not compiling for me. Only on version esphome/esphome:2022.11 is working.

Roei639, 2023-05-05:
The problem still exists, installed using docker on the latest and latest version of esphome.

Log:
```
INFO Reading configuration /config/haier-room-parents.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing haieracparents (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/haieracparents/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/haieracparents/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/haieracparents/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/haieracparents/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/haieracparents/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/haieracparents/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/haieracparents/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/haieracparents/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 2.25 seconds ==========================
```

jb137, 2023-06-29:
You can pull the 64 bit arm image using:
`docker pull ghcr.io/esphome/esphome-hassio:2023.6.3  --platform aarch64`

After this you can stop/start the ESPhome addon and it should be able to compile again.

Roei639, 2023-08-12:
the problem is not solved

bramcloet, 2023-11-16:
Same problem here after update to ESPHome 2023.11.0

willrnsantana, 2023-11-17:
+1 with the error after upgrading to 2023.11.1

K0HAX, 2023-11-17:
I'm having this error after upgrading to 2023.11.1, running the ESPhome addon in Home Assistant on a Raspberry Pi 3. The Pi is running the Home Assistant Operating System, and Home Assistant is reporting that it is 32-bit.

I tried stopping and starting the ESPhome addon, and that did not fix the issue.

domotiquemaison07, 2023-11-17:
Same thing for me for 2023.11.1 upgrade.

ssieb, 2023-11-18:
Stop commenting on this issue.  This one is closed because it's a duplicate.