esphome/issues #3921: ESP32 boot loop - image size incorrect after upgrade (2022.11.5 -> 2022.12.0)

tubes41, 2022-12-15:
### The problem

Performed an ESPHome upgrade on a DFRobot FireBeetle ESP32-E, from v2022.11.5 to v2022.12.0.
Upgrade was performed over USB via the ESPHome web interface.

ESP is now stuck in a boot loop, with logs only accessible via USB. Looks like an image / partition size issue?

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

DFRobot FireBeetle ESP32-E

### Component causing the issue

Bluetooth Proxy

### Example YAML snippet

```yaml
esphome:
  name: esp-find-1

esp32:
  board: firebeetle32
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret ha_api

ota:
  password: !secret ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Find-1 Fallback Hotspot"
    password: "LvHRUFfY6BEA"

captive_portal:

esp32_ble_tracker:
  scan_parameters:
    active: true
bluetooth_proxy:   
  active: true
```

### Anything in the logs that might be useful for us?

```txt
ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (657) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x40083725:0x3ffe3aa00x400930e5:0x3ffe3ac0 0x400987ad:0x3ffe3ae0 0x400f7482:0x3ffe3c10 0x40082e26:0x3ffe3c40 0x400792ba:0x3ffe3c90  |<-CORRUPTED
```

goldmann, 2023-01-01:
Seeing similar behavior, adding a little bit more logs:

```
[22:48:55]ELF file SHA256: 0000000000000000
[22:48:55]
[22:48:55]Rebooting...
[22:48:55]ets Jul 29 2019 12:21:46
[22:48:55]
[22:48:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:48:55]configsip: 0, SPIWP:0xee
[22:48:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:48:55]mode:DIO, clock div:2
[22:48:55]load:0x3fff0030,len:1184
[22:48:55]load:0x40078000,len:13132
[22:48:55]load:0x40080400,len:3036
[22:48:55]entry 0x400805e4
[22:48:55]E (352) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.
[22:48:55]
[22:48:55]assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)
[22:48:55]
[22:48:55]
[22:48:55]Backtrace:0x400837a5:0x3ffe3aa00x4008b6b5:0x3ffe3ac0 0x40090bed:0x3ffe3ae0 0x400edd76:0x3ffe3c10 0x40082ea6:0x3ffe3c40 0x400792ba:0x3ffe3c90  |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400837a5: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
WARNING Decoded 0x4008b6b5: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
WARNING Decoded 0x40090bed: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
WARNING Decoded 0x400edd76: do_core_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:290
 (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:401
WARNING Decoded 0x40082ea6: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:648
```

goldmann, 2023-01-01:
I think the reason is that we specify the board as `firebeetle32`, which is this one: https://www.dfrobot.com/product-1590.html (which has 16MB of flash) but in fact we use FireBeetle 2: https://www.dfrobot.com/product-2195.html which is equipped with 4MB of flash. I don't think PlatformIO has added proper support for ESP32-­WROOM­-32E chipset yet.

Things are even worse: the PlatformIO entry for FireBeetle 2 (https://docs.platformio.org/en/latest/boards/espressif32/dfrobot_firebeetle2_esp32s3.html) is wrong - it specifies totally different chip.

Saying this I don't know why my other FireBeetle 2 devices work well with this esphome version, though.

etherfi, 2023-03-16:
This thing used to work fine, and now it's bricking my devices. This is a major bug! Why has this not been rolled back?

ssieb, 2023-04-23:
There's nothing esphome can do about this, it's a platformio problem.
The device is also not bricked.  The only way you could get this onto a device is with a serial flash, so it's easily fixable.

closed 2023-04-23 by ssieb