esphome/issues #3923: Compilation failed xtensa-esp32-elf-g++: not found

raidenlal, 2022-12-15:
### The problem

Setup as I have worked fine in previous days, but then some update (hass or esphome) broke it.

Today I wanted to update one ESP wirelessly and before actual compilation there was some update (looked like python?, didn't care too much). Now compilation always fails with: sh: 1: xtensa-esp32-elf-g++: not found

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

api/api_connection

### Example YAML snippet

```yaml
esphome:
  name: bt-bridge
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:
web_server:
api:
ota:
logger:
  level: VERBOSE

esp32_ble_tracker:

sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: "mac"
    moisture:
      name: "name"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/bt-bridge.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bt-bridge (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling /data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/bt-bridge/.pioenvs/bt-bridge/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 4.65 seconds ==========================
```

slydiman, 2022-12-15:
The same https://github.com/esphome/issues/issues/3904

raidenlal, 2022-12-15:
You are correct. I have googled several parts of the log, but it seems I didn't google enough. Sorry about it.

closed 2022-12-15 by raidenlal