esphome/issues #3924: esp01-c3 boot issue after update

evlo, 2022-12-15:
### The problem

This is serial log as the device does not become available on wifi
[image]

### Which version of ESPHome has the issue?

2022.12

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP32

### Board

esp01-c3

### Component causing the issue

bluetooth_proxy?

### Anything in the logs that might be useful for us?

```txt
Primary usage for the device is obis reading

external_components:
  - source: github://evlo/esphome_ZPA-ZE312
    components: [obis]
    refresh: 0s
```
```

### Additional information

I thought it might be related to bluetooth_proxy since I updated and enabled bt proxy at the same time, but even after removing bluetooth_proxy from config, it still does not boot properly.

RENOxDECEPTION, 2022-12-16:
I have this happening on my esp32-s3 as well. Going to have to try disabling UART for the time being.

evlo, 2022-12-17:
full config causing the issue with `2022.12.1`
I have tried deleting .esphome folder before compiling
```
# https://github.com/Xinyuan-LilyGO/T-01C3 LILYGO T-01C3

esphome:
  name: eh-elektromer-t01c3
esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework: 
    type: arduino
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

api:
logger:
  level: VERBOSE
ota:
web_server:
  version: 2
captive_portal:

uart:
  id: uart_bus
  tx_pin: GPIO21
  rx_pin: GPIO20
  rx_buffer_size: 2048  
  baud_rate: 300
  data_bits: 7
  stop_bits: 1
  parity: EVEN
```

evlo, 2022-12-17:
I have found my old post https://github.com/esphome/issues/issues/3419#issuecomment-1250396816 so that means using
```
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```
should work

deleted .esphome 
deleted c:\users\*\.platformio
tried esphome run ... 
no luck
deleted .esphome 
deleted c:\users\*\.platformio
`./pip3 install esphome==2022.8.0`

evlo, 2022-12-20:
I compiled on linux vm with latest `2022.12.3`, i could not flash in linux with usb passthrough, so i copied the firmware.bin
and flashed with `.\esptool.exe write_flash 0 .\firmware.bin` command and result is this boot loop, no warning about uart, etc.
I would have kinda hoped for at least same error independent on the os where are  build the yaml, but maybe i did something wrong with flashing?
[image]

evlo, 2022-12-21:
Today I remembered I have local home assistant PC and flashed on it using esphome dashboard and with esp-idf in yaml
bootloop
[image]

jysaloma, 2022-12-22:
I am having the same issue with HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
Which I am using as bluetooth proxy for HA.

Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-14110 with baud rate 115200
[16:35:39]0210207
[16:35:39]Build:Feb  7 2021
[16:35:39]rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[16:35:39]Saved PC:0x403d0f88
[16:35:39]SPIWP:0xee
[16:35:39]mode:DIO, clock div:1
[16:35:39]load:0x3fcd6100,len:0x420
[16:35:39]load:0x403ce000,len:0x90c
[16:35:39]load:0x403d0000,len:0x236c
[16:35:39]entry 0x403ce000
[16:35:39]ESP-ROM:esp32c3-api1-20210207
[16:35:39]Build:Feb  7 2021
[16:35:39]rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[16:35:39]Saved PC:0x403d0f88
[16:35:39]SPIWP:0xee
[16:35:39]mode:DIO, clock div:1
[16:35:39]load:0x3fcd6100,len:0x420
[16:35:39]load:0x403ce000,len:0x90c
[16:35:39]load:0x403d0000,len:0x236c
[16:35:39]entry 0x403ce000
[16:35:39]ESP-ROM:esp32c3-api1-20210207
[16:35:39]Build:Feb  7 2021
[16:35:39]rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[16:35:39]Saved PC:0x403d0f88
[16:35:39]SPIWP:0xee
[16:35:39]mode:DIO, clock div:1
[16:35:39]load:0x3fcd6100,len:0x420
[16:35:39]load:0x403ce000,len:0x90c
[16:35:39]load:0x403d0000,len:0x236c
[16:35:39]entry 0x403ce000
[16:35:39]ESP-ROM:esp32c3-api1-20210207
[16:35:39]Build:Feb  7 2021
[16:35:39]rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[16:35:39]Saved PC:0x403d0f88
[16:35:39]SPIWP:0xee
[16:35:39]mode:DIO, clock div:1
[16:35:39]load:0x3fcd6100,len:0x420
[16:35:39]load:0x403ce000,len:0x90c
[16:35:39]load:0x403d0000,len:0x236c
[16:35:39]entry 0x403ce000
[16:35:39]ESP-ROM:esp32c3-api1-20210207

jysaloma, 2022-12-22:
In my case I commented the version numbers and it compiled and works fine now

esphome:
  name: ${name}
  name_add_mac_suffix: false
  platformio_options:
    board_build.mcu: esp32c3
    #platform_packages:
    #  - platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: arduino
    #version: 2.0.2
    #platform_version: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip

[17:10:50][I][wifi:518]: WiFi Connected!
[17:10:50][I][app:062]: setup() finished successfully!
[17:10:50][I][app:102]: ESPHome version 2022.12.3 compiled on Dec 22 2022, 17:08:25

evlo, 2022-12-22:
But you do not have uart component in you working config, correct?

jysaloma, 2022-12-22:
@evlo sorry I don't. I tested on mine and seems to be the same issue as yours.

[17:28:12][I][logger:258]: Log initialized
[17:28:12][I][app:029]: Running through setup()...
[17:28:45]ESP-ROM:esp32c3-api1-20210207
[17:28:45]Build:Feb  7 2021
[17:28:45]rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[17:28:45]Saved PC:0x4203018e
WARNING Decoded 0x4203018e: uart_ll_get_txfifo_len at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32c3/include/hal/uart_ll.h:329 (discriminator 1)
 (inlined by) panic_print_char at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:79 (discriminator 1)
[17:28:45]SPIWP:0xee
[17:28:45]mode:DIO, clock div:1
[17:28:45]load:0x3fcd6100,len:0x438
[17:28:45]load:0x403ce000,len:0x918
[17:28:45]load:0x403d0000,len:0x24e4
[17:28:45]entry 0x403ce000
[17:28:45][I][logger:258]: Log initialized
[17:28:45][I][app:029]: Running through setup()...
[17:29:19]ESP-ROM:esp32c3-api1-20210207
[17:29:19]Build:Feb  7 2021
[17:29:19]rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[17:29:19]Saved PC:0x4203018c
WARNING Decoded 0x4203018c: uart_ll_get_txfifo_len at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32c3/include/hal/uart_ll.h:329 (discriminator 1)
 (inlined by) panic_print_char at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:79 (discriminator 1)
[17:29:19]SPIWP:0xee
[17:29:19]mode:DIO, clock div:1
[17:29:19]load:0x3fcd6100,len:0x438
[17:29:19]load:0x403ce000,len:0x918
[17:29:19]load:0x403d0000,len:0x24e4
[17:29:19]entry 0x403ce000
[17:29:19][I][logger:258]: Log initialized
[17:29:19][I][app:029]: Running through setup()...

evlo, 2022-12-23:
I tried on ESP32-CAM board and it goes into starting uart and then resets, no warning decoded etc. messages

evlo, 2022-12-24:
log from esp32-cam
```
[22:57:27][C][uart.arduino_esp32:077]: Setting up UART...
[22:57:33]ets Jun  8 2016 00:22:57
[22:57:33]
[22:57:33]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:57:33]configsip: 0, SPIWP:0xee
[22:57:33]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:57:33]mode:DIO, clock div:2
[22:57:33]load:0x3fff0030,len:1184
[22:57:33]load:0x40078000,len:13132
[22:57:33]load:0x40080400,len:3036
[22:57:33]entry 0x400805e4
[22:57:34][     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[22:57:34][   455][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[22:57:34][I][logger:258]: Log initialized
[22:57:34][C][ota:469]: There have been 4 suspected unsuccessful boot attempts.
```

evlo, 2022-12-24:
I tried on c3 with
```
logger:
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE
    web_server: INFO
    wifi: INFO
    app: INFO
    ota: INFO
    wifi_esp32: INFO
    scheduler: INFO
  baud_rate: 0
```
no luck, it does show on router evantually, i guess in the safemode, but
```
esphome logs .\eh-elektromer-cam.yaml --device 192.168.50.154
```
is stuck on
```
INFO Trying to reconnect to 192.168.50.154 in the background
```