esphome/issues #3925: ESP32 ledc not dimming or turning off display after upgrade to 2022.12.0

madmic1314, 2022-12-15:
### The problem

Before 2022.12.0 (2022.11.x) the turn off / dim option would work and the screen would dim / turn off as expected. After upgrading this is no longer the case. 

Have tried on two different brands of ESP32 with two screens (same make). Also tried using the webportal as well as in Home Assistant

Have set the restore mode to OFF. When the device boots, the screen is on, but the portal (and HA) report it's off. Turning it on + off does nothing

### Which version of ESPHome has the issue?

ESPHome 2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.6

### What platform are you using?

ESP32

### Board

ESP32 az-delivery-devkit-v4, ESP32 nodemcu-32s

### Component causing the issue

Display

### Example YAML snippet

```yaml
api:

binary_sensor:
  - platform: status
    name: ${friendly_name} Status

captive_portal:

esphome:
  name: ${devicename}
  comment: ${device_description}

http_request:
  useragent: esphome/$devicename
  timeout: 2s

ota:
  safe_mode: true
  password: !secret ota_password

sensor:
  - platform: uptime
    name: ${friendly_name} Uptime
    filters:
      - lambda: return x / 60.0;
    unit_of_measurement: minutes
  - platform: wifi_signal
    name: ${friendly_name} Wifi Signal
    update_interval: 60s

switch:
  - platform: restart
    name: ${friendly_name} Restart

web_server:
  port: 80
  version: 2
  include_internal: true

wifi:
  ap:
    ssid: "ESPDev"
    password: "HelpHelp"

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password

esphome:
  platform: ESP32
  board: nodemcu-32s  #have also tried: "az-delivery-devkit-v4" on the AZ Board

logger:
  level: debug

substitutions:
  devicename: mkinfoscreens2
  friendly_name: Mr K Info 2
  device_description: ESP32 device with colour screen

spi:
  clk_pin: GPIO21
  mosi_pin: GPIO22
  miso_pin: GPIO23

output:
  - platform: ledc
    pin: 32
    id: gpio_32_backlight_pwm

light:
  - platform: monochromatic
    output: gpio_32_backlight_pwm
    name: ${friendly_name} Backlight
    id: back_light
    restore_mode: ALWAYS_ON     # have also tried this OFF. The screen is on, but the portal (and HA) report it's off. Turning it on + off does nothing

font:
  - file: 'BebasNeue.ttf'
    id: f_36
    size: 36

display:
  - platform: ili9341
    model: TFT 2.4
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    id: disp1
    rotation: 270
    update_interval: 5s
    pages:
      - id: page1
        lambda: |-
          it.print(0, 30, id(f_36), TextAlign::BASELINE_LEFT, "Testing");
```

### Anything in the logs that might be useful for us?

```txt
Backlight was on:
23:33:58 | [D] | [light:035] | 'Mr K Info 2 Backlight' Setting:
-- | -- | -- | --
23:33:58 | [D] | [light:046] | State: OFF
23:33:58 | [D] | [light:084] | Transition length: 1.0s

No ch#ange to screen.

23:34:00 | [D] | [light:035] | 'Mr K Info 2 Backlight' Setting:
23:34:00 | [D] | [light:046] | State: ON
23:34:00 | [D] | [light:084] | Transition length: 1.0s
```

madmic1314, 2022-12-16:
Can confirm this is still the same in 2022.12.1

niahane, 2022-12-16:
> 

I confirm, I have the same symptom on an ILI9341 TFT LCD display. It stopped working on 2022.12.0. Version 2022.12.1 makes no difference, the part about the broken code is this:
'''
outputs:
  #Backlight pwm section
  - platform: ledc
    pins: 15
    id: backlight_pwm
    inverted: true
    internal: false
    
light:
  - platform: monochromatic
    output: backlight_pwm
    name: "Backlight"
    id: back_light
    restore_mode: ALWAYS_ON
'''

Downgrading to 2022.11.5 start working again.

madmic1314, 2022-12-16:
> > 
> 
> I confirm, I have the same symptom on an ILI9341 TFT LCD display. It stopped working on 2022.12.0. Version 2022.12.1 makes no difference, the part about the broken code is this:
> 
> ```
> outputs:
>   #Backlight pwm section
>   - platform: ledc
>     pins: 15
>     id: backlight_pwm
>     inverted: true
>    
>     
> light:
>   - platform: monochromatic
>     output: backlight_pwm
>     name: "Backlight"
>     id: back_light
>     restore_mode: ALWAYS_ON
>     internal: false
> ```
> 
> Downgrading to 2022.11.5 start working again.

How did you downgrade ESPHome? I've tried looking and can't see how to do this. There's nothing in 2022.12 that I currently need.

niahane, 2022-12-16:
> How did you downgrade ESPHome? I've tried looking and can't see how to do this. There's nothing in 2022.12 that I currently need.

I've had trouble with this in the past so whenever there is an esphome update I leave the "Create backup before update" checkbox. So I always find the backup of previous versions easily restored. I don't think there is any other way to go back one version, you can specify which version of esphome to use in the advanced configuration but it never worked for me.

madmic1314, 2022-12-16:
Good idea! Have restored 2022.11.5 and can confirm it's working again - same code for the device, recompiled and uploaded again.

NdK73, 2022-12-17:
I can confirm the same behaviour on ESP32-2432S028 board.
The only PWM not working is the one related to backlight:
```
output:
  - platform: ledc
    pin: GPIO21
    id: backlight_pwm
  - platform: ledc
    pin: GPIO4
    id: light_pwm_r
    inverted: true
    frequency: 19531Hz
  - platform: ledc
    pin: GPIO16
    id: light_pwm_g
    inverted: true
    frequency: 19531Hz
  - platform: ledc
    pin: GPIO17
    id: light_pwm_b
    inverted: true
    frequency: 19531Hz
  - platform: ledc
    pin: GPIO26
    id: rtttl_out
    frequency: 19531Hz
light:
  - platform: monochromatic
    output: backlight_pwm
    name: "${upper_devicename} Backlight"
    id: disp_backlight
    restore_mode: ALWAYS_ON
  - platform: rgb
    name: "${upper_devicename} RGB LED"
    red: light_pwm_r
    green: light_pwm_g
    blue: light_pwm_b
    restore_mode: ALWAYS_OFF
```
RGB controls and RTTTL work as expected.

NdK73, 2022-12-20:
Another update: still not working in 2022.12.3
```
esp32:
  board: esp32dev
  framework:
    type: arduino # REQUIRED for integrated webserver
    #version: 1.0.6 # Does not compile in latest update
    #version: 2.0.0 # Broken ledc on GPIO21
```
I also tried `version: dev` but that **also** broke RTTTL output (**really** noisy when not playing).
Seems something stopped working with major version upgrade.

For completeness, I also tried using framework `type: esp-idf`, but:
  - had to remove webserver
  - the issue with backlight remains
  - the board is nearly unresponsive (~120 sec to complete 'boot', 6-7 sec between action in HA and visible change in RGB LED, display redraw taking more than 2 sec)...

drts-sv, 2022-12-21:
Can confirm this is still the same in 2022.12.3

Electronlibre2012, 2022-12-24:
Hello,

i search a lot after all my esp32 with a TFT or LCD display doesnt dimmed anymore...and found this post...i am not crazy lol : dimm a led doesnt work anymore...

Hope a dev will do something...but we are at the 2022.12.8 and still not working...

madmic1314, 2022-12-24:
2012.12.8 is the Home Assistant version, not the esphome version.

The current workaround is to restore a backup is esphome of version 2022.11.x, recompile and reflash the esp.

Electronlibre2012, 2022-12-24:
yes, sure, my bad...lol

hope it will be fixed next year 😉

i wait, no problem

NdK73, 2023-01-05:
Waiting for a new ESPHome version, I also tried changing platform from monochromatic to rgb (since RGB is working for the onboard LED, it MIGHT have worked for backlight...) and changing the order of 'output' definitions (possible off-by-one?)...
As you might expect both tests were negative.
The last test confirmed it's a problem driving GPIO21 in the new release of Arduino framework: I just swapped the pin definitions between backlight and R channel (GPIO4): from the monochromatic channel I could control the R LED, but nothing happened to backlight when changing the color of the RGB light.

SLboat, 2023-01-19:
[image]
[image]
https://github.com/esphome/esphome/commit/33b1a853b9afa55c4f26cacbda1133f3bc36fdb6
https://github.com/esphome/esphome/pull/3820

seems this bug had been fix, but not yet release at current version(2022.12.3)

SLboat, 2023-01-19:
> seems

replace the new version ledc component should work well.

NdK73, 2023-01-19:
Nope.
At least not with my board (esp32dev). I also tried wrapping in quotes the pull request URI.

In the log I see:
```[17:38:50][C][ledc.output:114]: LEDC Output:
[17:38:50][C][ledc.output:115]:   Pin GPIO26
[17:38:50][C][ledc.output:116]:   LEDC Channel: 0
[17:38:50][C][ledc.output:117]:   Frequency: 19531.0 Hz
[17:38:50][C][ledc.output:114]: LEDC Output:
[17:38:50][C][ledc.output:115]:   Pin GPIO4
[17:38:50][C][ledc.output:116]:   LEDC Channel: 1
[17:38:50][C][ledc.output:117]:   Frequency: 1000.0 Hz
[17:38:50][C][ledc.output:114]: LEDC Output:
[17:38:50][C][ledc.output:115]:   Pin GPIO16
[17:38:50][C][ledc.output:116]:   LEDC Channel: 2
[17:38:50][C][ledc.output:117]:   Frequency: 1000.0 Hz
[17:38:50][C][ledc.output:114]: LEDC Output:
[17:38:50][C][ledc.output:115]:   Pin GPIO17
[17:38:50][C][ledc.output:116]:   LEDC Channel: 3
[17:38:50][C][ledc.output:117]:   Frequency: 1000.0 Hz
[17:38:50][C][ledc.output:114]: LEDC Output:
[17:38:50][C][ledc.output:115]:   Pin GPIO21
[17:38:50][C][ledc.output:116]:   LEDC Channel: 4
[17:38:50][C][ledc.output:117]:   Frequency: 1000.0 Hz
[...]
[17:38:50][C][light:104]: Light 'Display 01 Backlight'
[17:38:50][C][light:106]:   Default Transition Length: 1.0s
[17:38:50][C][light:107]:   Gamma Correct: 2.80
[17:38:50][C][light:104]: Light 'Display 01 RGB LED'
[17:38:50][C][light:106]:   Default Transition Length: 1.0s
[17:38:50][C][light:107]:   Gamma Correct: 2.80
```
If I create another monochromatic control on a different GPIO (say GPIO22) I can see the PWM wave on the oscilloscope. So it seems JUST GPIO21 is not working anymore :(

Gopten, 2023-01-22:
Suspect this is related to a problem I have running an RTTTL passive buzzer by LEDC on ESP32. Once I updated the ESP to 2022.12.03 the buzzer continues to bleep at a low level and heats up.
`output:
  - platform: ledc
    pin: GPIO19
    id: rtttl_out
    inverted: True`
Prior to the update this would only  have happened if inverted was set to false.

NdK73, 2023-01-23:
Latest update () left me with BOTH backlight and RTTTL not working... I tried commenting out the 'useless' channels and noticed the RGB LED on the back changed color when I tried to change the backlight setting. Really strange, since I commented out their definitions!
Note that the pin definitions were already tested with old releases and by schematic!
Seems ledc is assuming a fixed binding between channels and GPIOs, ignoring yaml contents.
Relevant sections of my yaml file are:
```
esp32:
  board: esp32dev
  framework:
    type: arduino

output:
  - id: light_pwm_r
    platform: ledc
    pin: GPIO4
    inverted: true
    #frequency: 19531Hz
  - id: light_pwm_g
    platform: ledc
    pin: GPIO16
    inverted: true
    #frequency: 19531Hz
  - id: light_pwm_b
    platform: ledc
    pin: GPIO17
    inverted: true
    #frequency: 19531Hz
  - id: bl_pwm
    platform: ledc
    pin: GPIO21 # backlight e su CN3
    frequency: 1000Hz
    #frequency: 19531Hz
  - id: rtttl_out
    platform: ledc
    pin: GPIO26
    frequency: 5000Hz

rtttl:
  output: rtttl_out

light:
  - id: disp_backlight
    platform: monochromatic
    output: bl_pwm
    name: "${upper_devicename} Backlight"
    restore_mode: ALWAYS_ON
  - platform: rgb
    name: "${upper_devicename} RGB LED"
    red: light_pwm_r
    green: light_pwm_g
    blue: light_pwm_b
    restore_mode: ALWAYS_OFF
```

The parts I commented out are the "platform: rgb" section and the light_pwm_{r,g,b} ones.
And disp_backlight ended up using GPIO4 instead of GPIO21.

Electronlibre2012, 2023-01-24:
thanks @NdK73 to understand what happened, i am in the same boat since about 1 month and nobody seems concerned by this problem...which is a real bad one because all my display stay on all the time what is not nice for their lifetime...

I dont understand why we are only a few to complain that it doesnt work anymore???

I thought a lot of user use the LedC component...seems not...what can we do?

If you can read this post @OttoWinter that will be great, thanks for your work, i understand this is perhaps not a crucial bug but it would be nice if this can be solved in future releases...my dev skill is badly not enought advance to debug or understand what happened with LedC component...

majdzik84, 2023-01-24:
I have the same problem ! :(

misiek42, 2023-01-24:
Hi
You should uncomment as I have indicated:
display:
  - platform: ili9341
    model: TFT 2.4
    cs_pin: 14
    dc_pin: 27
   # led_pin: 32 uncomment  this entry
    reset_pin: 33
    id: disp1
    rotation: 270
    update_interval: 5s
    pages:

Gopten, 2023-01-24:
`
output:
  - platform: ledc
    pin: GPIO19
    id: rtttl_out
    inverted: True
    zero_means_zero: True
`

I added Zero means zero to my code and when I restarted it was still clicking, but once I played a note on the buzzer the clicking stopped and so far hasn't returned and no heating up. I added a single note to be played on_boot.

NdK73, 2023-01-25:
@misiek42 thanks for pointing out `led_pin` definition: seems with 2.0.0 of Arduino framework priorities got swapped between outputs and display. Finally it works correctly and dims.

@Gopten in my case zero_means_zero seems not to change anything, but it's now in config and stays there :)

segfly, 2023-01-26:
> I dont understand why we are only a few to complain that it doesnt work anymore???

I'm also don't understand given this is pretty basic/essential functionality. Frustratingly, there have been a few releases this week but they are only cherry-picking from dev branch.

This month has been my first time using esphome, but I'm not new to arduino/esp32/platformio and I thought I was losing my mind until I came across this issue/PR. Unfortunately, on the `seeed_xiao_esp32c3` just applying PR #3820 with the recommended framework/platform versions did nothing. I kept getting boot loops as it failed to set the ledc hw pwm.

I had to also apply PR #3564. The following worked for me:
```
esp32:
  board: seeed_xiao_esp32c3
  variant: esp32c3

external_components:
  - source: github://pr#3564
    components: 
      - esp32
      - esp32_camera
      - ethernet_info
      - ethernet
      - shelly_dimmer
      - wifi
      - dallas
  - source: github://pr#3820
    components: [ ledc ]
```

I'm still running into issues setting a specific frequency besides 1KHz as during boot ledc fails initialize. I think the recommended frequencies in the docs may be calculated incorrectly https://esphome.io/components/output/ledc.html#recommended-frequencies - not 100% sure yet as I am just now starting to dig further into how the frequency and duty resolution are being calculated.

segfly, 2023-01-27:
An update for anyone with the same issue: For the `seeed_xiao_esp32c3`, and therefore I assume all C3 boards, the bit depth does seem to be incorrect as listed the LEDC documentation page, as well it also seems to be calculated incorrectly in `ledc_output.cpp`.

At 5KHz, `ledc_output.cpp` will calculate a duty resolution of 13 bits. This will cause ledc init to fail with the error `requested frequency and duty resolution cannot be achieved, try reducing freq_hz or duty_resolution`. Now, I'm unsure why that is, because according to the [espressif docs example](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#ledc-api-supported-range-frequency-duty-resolution), 5Khz should work at 13 bits.

It is only 1 bit off however. If a resolution of 12 bits is used, ledc inits just fine with a frequency of 5Khz. I found this 1 bit off error consistent for higher frequencies as well (I haven't tried lower frequencies).

My board is a 160Mhz board, which seems to be the same clock freq assumed in the espressif docs, so I don't quite understand why the calculated resolution fails to init unless it is reduced by 1 bit. I did try setting `board_build.f_cpu: 160000000L` for good measure, but it made no difference, and still requires reducing the bit resolution to work.

Nevertheless, `ledc_output.cpp` can be patched to decrement the bit depth by 1 as follows and it completely fixes the problems I encountered:

```
diff --git a/esphome/components/ledc/ledc_output.cpp b/esphome/components/ledc/ledc_output.cpp
index 34cf4cd1..36868de7 100644
--- a/esphome/components/ledc/ledc_output.cpp
+++ b/esphome/components/ledc/ledc_output.cpp
@@ -39,6 +39,7 @@ optional<uint8_t> ledc_bit_depth_for_frequency(float frequency) {
     const float min_frequency = ledc_min_frequency_for_bit_depth(i, (frequency < 100));
     const float max_frequency = ledc_max_frequency_for_bit_depth(i);
     if (min_frequency <= frequency && frequency <= max_frequency) {
+      i--;
       ESP_LOGD(TAG, "Resolution calculated as %d", i);
       return i;
     }
```

I may turn this into a pull request, but I just purchased a few more C3 variant boards (and an S2) to try and see if this issue exists on on other boards or if there is something odd about this seeed xiao esp32c3.