esphome/issues #3927: Cannot comile with v2022.12.1

RaceAap, 2022-12-16:
### The problem

Compile fails with: "assert.h:20:10: fatal error: sdkconfig.h: No such file or directory"

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.6

### What platform are you using?

ESP32

### Board

esp32dev, mhetesp32devkit

### Component causing the issue

sdkconfig.h ??

### Example YAML snippet

```yaml
substitutions:
  esphome_name: minitemp01

esphome:
  name: minitemp01
  platform: ESP32
  board: mhetesp32devkit

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_pass
  domain: XXXXXXXXXXXXXX
  manual_ip:
    static_ip: XXXXXXXXXXXXXX
    gateway: XXXXXXXXXXXXXX
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Minitemp01 Fallback Hotspot"
#    password: !secret esphome_wifi_ap_pass
#captive_portal:

# Enable logging
logger:
  level: DEBUG
  #level: INFO

# Enable Home Assistant API
api:
  password: !secret esphome_api_pass

ota:
  password: !secret esphome_ota_pass

time:
  - platform: homeassistant

# Enable bluetooth  
esp32_ble_tracker:

switch:

  - platform: gpio
    name: "${esphome_name}-Onboard-LED"
    pin: 2
    inverted: False

  - platform: restart
    name: ${esphome_name}-restart
    id: restart_switch

text_sensor:
  - platform: template
    name: "${esphome_name} Uptime"
    id: uptime_human
    icon: mdi:clock-start

sensor:
  - platform: wifi_signal
    name: "${esphome_name} WiFi Signal"
    update_interval: 120s
```

### Anything in the logs that might be useful for us?

```txt
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing esptest01 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Compiling /data/esptest01/.pioenvs/esptest01/src/esphome/components/api/api_connection.cpp.o
Compiling /data/esptest01/.pioenvs/esptest01/src/esphome/components/api/api_frame_helper.cpp.o
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/wchar.h:6,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:40,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/wchar.h:6,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:40,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/esptest01/.pioenvs/esptest01/src/esphome/components/api/api_connection.cpp.o] Error 1
compilation terminated.
*** [/data/esptest01/.pioenvs/esptest01/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 7.49 seconds ==========================
```

### Additional information

Home Assistant runs on a VM on a Synology with 64bits Hassos.
Issue looks similar as  #3904 but is not limited to 32bits arm architecture.

glandau, 2022-12-16:
same here :( Rasberry 3B 32bit stuck at prepairing Installation

RaceAap, 2022-12-19:
Did some extra testing.
When I use a standalone docker image: esphome/esphome:2022.12.1 all works good, but when i use the Home Assistant Add-on from within Home Assistant I cannot compile.
The issue is specific for the HomeAssistant add-on.

lambu76, 2023-02-04:
I have same issue on ESP 2022.12.8.

Any news about resolution ?

lambu76, 2023-03-02:
problem still present in ESP 2023.2.4

nogueirasa, 2023-04-09:
Tested vesion esphome/esphome:latest, but not compiling for me. Only on version esphome/esphome:2022.11 is working.

nogueirasa, 2023-04-09:
Hi, same problem here.

Trying to install Blackymas/NSPanel_HA_Blueprint

ESPHome: v2023.3.2 (docker)
Home Assiatnt: 2023.4.1 (docker)
Raspberry Pi 4 OS 32bits

Tested vesion esphome/esphome:latest, but not compiling for me. Only on version esphome/esphome:2022.11 is working.