esphome/issues #3928: Warning while compiling for m5stack atom speaker kit

radek-kafka, 2022-12-16:
### The problem

Not a problem, just during compilation for m5stack atom speaker kit there are some warnings about using of depreciated functions. 

In file included from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms/esp/32/clockless_rmt_esp32.h:95,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms.h:36,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/FastLED.h:52,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/FastLED.cpp:2:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms/esp/32/clockless_rmt_esp32.h:95,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms.h:36,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/FastLED.h:52,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/bitswap.cpp:2:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms/esp/32/clockless_rmt_esp32.h:95,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/platforms.h:36,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/FastLED.h:52,
                 from /data/m5stack-atom-speaker-kit/.piolibdeps/m5stack-atom-speaker-kit/FastLED/colorutils.cpp:7:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~

### Which version of ESPHome has the issue?

2022.12

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.6

### What platform are you using?

ESP8266

### Board

m5AtomLite

### Anything in the logs that might be useful for us?

```txt
INFO Compiling app...
Processing m5stack-atom-speaker-kit (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
[... 494 lines cut ...]
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib915/ESPmDNS/ESPmDNS.cpp.o
Archiving /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib4e6/libESPAsyncWebServer-esphome.a
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_core/ed25519/core_ed25519.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_core/ed25519/core_ristretto255.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_hash/crypto_hash.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_hash/sha256/hash_sha256.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_onetimeauth/crypto_onetimeauth.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_onetimeauth/poly1305/donna/poly1305_donna.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_onetimeauth/poly1305/onetimeauth_poly1305.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_scalarmult/crypto_scalarmult.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_scalarmult/curve25519/ref10/x25519_ref10.c.o
Archiving /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib915/libESPmDNS.a
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_scalarmult/curve25519/scalarmult_curve25519.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_stream/chacha20/ref/chacha20_ref.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_stream/chacha20/stream_chacha20.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_stream/crypto_stream.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/crypto_verify/sodium/verify.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/sodium/codecs.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/sodium/core.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-kit/lib97b/libsodium/sodium/runtime.c.o
Compiling /data/m5stack-atom-speaker-kit/.pioenvs/m5stack-atom-speaker-
```

ssieb, 2022-12-16:
This is a problem with the fastled library, there's nothing esphome can do about it, unless there's a newer version of the library that has it fixed.

radek-kafka, 2022-12-16:
Ok .. many thanks .. Just thought that something at my side is not up to date or broken ..

closed 2022-12-16 by radek-kafka