esphome/issues #3929: Not possible to run docker esphome/esphome container - problem with platformio

jfdzar, 2022-12-16:
### The problem

Running the docker container of esphome is not working
`docker run --rm -v "${PWD}":/config -it esphome/esphome run XXXX.yaml`

It looks like there is a problem when downloading PlatformIO it enters a loop and cannot compile the file

Output from Terminal

INFO Reading configuration espcam-home.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing espcam-home (board: esp32cam; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
-------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Downloading...
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Platform Manager: Warning! Package Mirror: [Errno 1] Operation not permitted
WARNING Warning! Package Mirror: [Errno 1] Operation not permitted

### Which version of ESPHome has the issue?

linux/arm/v7 latest (10f4fce46442)

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

'--

### What platform are you using?

ESP32

### Board

ESP32-Cam

### Component causing the issue

docker image

### Example YAML snippet

```yaml
It does not compile any YAML
```

### Anything in the logs that might be useful for us?

```txt
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing espcam-home (board: esp32cam; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
-------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Downloading...
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Platform Manager: Warning! Package Mirror: [Errno 1] Operation not permitted
WARNING Warning! Package Mirror: [Errno 1] Operation not permitted
Platform Manager: Looking for another mirror...
WARNING Looking for another mirror...
Downloading  [####################################]  100%
Downloading...
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Platform Manager: Warning! Package Mirror: [Errno 1] Operation not permitted
WARNING Warning! Package Mirror: [Errno 1] Operation not permitted
Platform Manager: Looking for another mirror...
WARNING Looking for another mirror...
Downloading  [####################################]  100%
Downloading...
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Platform Manager: Warning! Package Mirror: [Errno 1] Operation not permitted
WARNING Warning! Package Mirror: [Errno 1] Operation not permitted
Platform Manager: Looking for another mirror...
WARNING Looking for another mirror...
Downloading  [####################################]  100%
Downloading...
```

### Additional information

I am running it on a raspberry pi running Raspbian 5.10.103-v7+ headless

jfdzar, 2022-12-17:
Looks like this is solving the issue

I guess the parameter --privilege was missing on the docker command

[https://github.com/esphome/issues/issues/3700#issuecomment-1352425681](https://github.com/esphome/issues/issues/3700#issuecomment-1352425681)

closed 2022-12-17 by jfdzar

steamfoundry, 2022-12-20:
> 

`--privileged`