esphome/issues #3932: Can't merge

sbrocket, 2022-12-16:
### The problem

https://esphome.io/guides/configuration-types.html#packages states:

> All definitions from packages will be merged with your main config in non-destructive way so you could always override some bits and pieces of package configuration.

> Dictionaries are merged key-by-key. **Lists of components are merged by component ID if specified.** Other lists are merged by concatenation. All other config values are replaced with the later value.

However, the examples under this don't include any examples of merging lists of components by component ID, and my attempt at doing so results in config validation failing because of a redefined component ID.

```
# Snippet from pkg/sonoff-s31-plug.yaml
switch:
  - platform: gpio
    name: ${device_name_upper} Relay
    pin: GPIO12
    id: relay
    restore_mode: RESTORE_DEFAULT_ON
```
```
# s31-fan.yaml
substitutions:
  device_name: "s31-fan"
  device_name_upper: S31 Fan

packages:
  common: !include pkg/common.yaml
  s31-plug: !include pkg/sonoff-s31-plug.yaml
  override-relay:
    switch:
    - platform: gpio
      id: relay
      pin: GPIO12
      restore_mode: RESTORE_DEFAULT_OFF
```

This results in the following config validation error:

```
INFO Reading configuration /config/s31-fan.yaml...
INFO Detected timezone 'Etc/UTC'
Failed config

switch.gpio: [source /config/pkg/sonoff-s31-plug.yaml:39]
  platform: gpio
  name: S31 Fan Relay
  pin: 
    number: 12
    mode: 
      output: True
      analog: False
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
  
  ID relay redefined! Check switch->0->id.
  id: relay
  restore_mode: RESTORE_DEFAULT_ON
  disabled_by_default: False
  interlock_wait_time: 0ms
```

Putting the override `switch:` dictionary at the top level of s31-fan.yaml results in a slightly different but still "ID relay redefined" error. Am I misunderstanding `Lists of components are merged by component ID if specified.` from the documentation or doing so incorrectly, or does this not actually work?

My esphome is a few months out of date, 2022.9.4, so I'll try updating to see if there was a recent change in behavior. I didn't see anything related in the changelog though.

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

sonoff s31

### Component causing the issue

gpio

sbrocket, 2022-12-16:
It looks like this language was added to the documentation fairly recently, in https://github.com/esphome/esphome-docs/commit/ef898d5203763e835e5707d63b0745b9805f9c16, but that the change it was (incompletely) documenting has not yet landed - https://github.com/esphome/esphome/pull/3555. So yes, the documentation appears to be incorrect currently.

SebTM, 2023-09-03:
Seems like still an issue for me trying to redefine light1 e.g. does not work as it throws an error "ID light1 redefined! Check light->1->id." with Failed config

ssieb, 2023-09-03:
The mentioned PR has been merged, so this issue is not relevant any more.  Please come ask on the esphome discord server about your config.

closed 2023-09-03 by ssieb