esphome/issues #3933: Water usage from slimmelezer

peno64, 2022-12-17:
### The problem

It is not possible to retrieve the water usage from slimmemeter. In the code I see in fields.h for example the definition of water_delivered but in the yaml configuration file it is not possible to specify it. It is also not documented in https://esphome.io/components/sensor/dsmr.html 
Belgium readers do send this data and I see the definition  in fields.h is correct so why is it tnot possible to specify it in the configation file?

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"

esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "2.0"

uart:
  baud_rate: 115200
  rx_pin: D7

dsmr:

sensor:
  - platform: dsmr
    water_delivered:
      name: "Water Consumed"
```

### Anything in the logs that might be useful for us?

```txt
Compiling this with: esphome compile water.yaml
gives:

Failed config

sensor.dsmr: [source water.yaml:23]
  platform: dsmr

  [water_delivered] is an invalid option for [sensor.dsmr]. Did you mean [power_delivered], [power_delivered_l3], [power_delivered_l2]?
  water_delivered:
    name: Water Consumed
```

Calypoter, 2023-05-06:
Is there a way to add this option? I also would like to see it with esphome

peno64, 2023-05-06:
See this: https://github.com/peno64/slimmemeter/wiki
Works perfect for me

peturdainn, 2023-11-21:
And now it works out of the box so this issue can be closed I think