esphome/issues #3936: Gfonts usage fail in 2022.12.1

wwj1978, 2022-12-18:
### The problem

`   Only truetype (.ttf) files are supported. Please make sure you're using the correct format or rename the extension to .ttf.`
This is the error I got when trying to run the same yaml after updating to `2022.12.1`. I'm using the gfonts like this:

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

not relevant

### What platform are you using?

ESP32

### Board

Wroom32

### Component causing the issue

font

### Example YAML snippet

```yaml
font:
  - file: "file: gfonts://Roboto"
    id: "font2"
    size: 48
  ```
```

### Anything in the logs that might be useful for us?

```txt
Failed config

font: [source file-name.yaml:82]
  -
    Only truetype (.ttf) files are supported. Please make sure you're using the correct format or rename the extension to .ttf.
    file: file: gfonts://Roboto
    id: font2
    size: 48
```
```

ssieb, 2022-12-19:
Why do you have an extra `file:` in the font reference?

closed 2022-12-19 by ssieb

wwj1978, 2022-12-19:
Sorry, this Was a mistake in pasting the code.

This is the yaml font declaration:

```
font:
  - file: "gtonts://Roboto"
    id: "nerd20"
    size: 20
  - file: "gfonts://Roboto"
    id: "font2"
    size: 48
```

And this is the esphome run log:

```
Failed config

font: [source sv-testing.yaml:101]
  -
    Only truetype (.ttf) files are supported. Please make sure you're using the correct format or rename the extension to .ttf.
    file: gtonts://Roboto
    id: nerd20
    size: 20
  - file:
      family: Roboto
      weight: 400
      italic: False
      type: gfonts
    id: font2
    size: 48
    glyphs:
      -
      - !
      - "
      - %
      - (
      - )
      - +
      - ,
      - -
      - .
      - /
      - 0
      - 1
      - 2
      - 3
      - 4
      - 5
      - 6
      - 7
      - 8
      - 9
      - :
      - =
      - A
      - B
      - C
      - D
      - E
      - F
      - G
      - H
      - I
      - J
      - K
      - L
      - M
      - N
      - O
      - P
      - Q
      - R
      - S
      - T
      - U
      - V
      - W
      - X
      - Y
      - Z
      - _
      - a
      - b
      - c
      - d
      - e
      - f
      - g
      - h
      - i
      - j
      - k
      - l
      - m
      - n
      - o
      - p
      - q
      - r
      - s
      - t
      - u
      - v
      - w
      - x
      - y
      - z
      - °
  ```

wwj1978, 2022-12-19:
Extremally embarrassed.

koenvervloesem, 2022-12-19:
These things often need an extra pair of eyes ;-)