esphome/issues #3937: Growatt solar component started giving 'modbus CRC errors'

wauswaus, 2022-12-18:
### The problem

After updating from ESPHome 2022.11.5 tot 2022.12.1 the Growatt solar component log constantly "Modbus CRC Check failed" errors and no values are being reported to Home Assistant anymore.

There is also a ticket for the [SDM meter component](https://github.com/esphome/issues/issues/3912) where a simular problem is described which could be related.

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

Growatt solar

### Example YAML snippet

```yaml
uart:
  - id: uart2
    baud_rate: 9600
    tx_pin: GPIO17
    rx_pin: GPIO16

modbus:
  uart_id: uart2

sensor:
  - platform: growatt_solar
    update_interval: 30s
    protocol_version: RTU2

    inverter_status:
      name: "Growatt Status Code"

    energy_production_day:
      name: "Growatt Today's Generation"
      accuracy_decimals: 2
      icon: mdi:solar-power
      state_class: total_increasing
```

nagyrobi, 2022-12-21:
Try latest update

wauswaus, 2022-12-23:
Tried  2022.12.3

No luck...

bigfloor, 2022-12-27:
Same problem here, latest 2022.12.3 did not fix it, downgrade to 2022.11.2 (via the CLI on my Macbook) everything works again. 
ESP32 with MAX3485 module and growatt code (growatt_solar) using RTU2

wauswaus, 2022-12-30:
The issue is not in ESPHome, but in the underlaying arduino libraries.
What fixed it for me is to use latest Arduino framework:

```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino
    version: 2.0.6
```

closed 2022-12-30 by wauswaus

bigfloor, 2022-12-31:
Confirmed ! Fixed it for me as well

nagyrobi, 2022-12-31:
How about `esp-idf` framework?

wroadd, 2023-01-07:
> How about `esp-idf` framework?

Same error with esp-idf.
With the protocol version "RTU", there is no CRC errors (but some of received results are false, of course), RTU2 version gives errors (I've checked with other Modbus server, and the received messages from the inverter are still the same)

gkasprow, 2023-02-18:
https://github.com/esphome/issues/issues/3912#issuecomment-1435782516