esphome/issues #3938: ArduinoInternalGPIOPin missing in esphome 2022.12, do I need renumbering of pins?

halecivo, 2022-12-18:
### The problem

Hello, I know ArduinoInernalGPIOPin support has been dropped as part of this commit https://github.com/esphome/esphome/pull/3564/commits/3f9deb12638671dbd300e60ad8ac15de899af6fd, however the commit message is very vague for me and I don't know why this was changed and what is correct way to update the old code. Old code I used referenced pins by Arduino numbering and I am not sure if this is still possible in some way, or renumbering is needed (which is unnecessary risk of possible mistake for me). 

I am sorry if the information was available and I just missed it.

FYI @jesserockz 

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
climate:
- platform: custom
  lambda: |-
    auto thermostat_in = new esp32::ArduinoInternalGPIOPin();
    thermostat_in->set_pin(14);
    thermostat_in->pin_mode(gpio::FLAG_INPUT);
    auto thermostat_out = new esp32::ArduinoInternalGPIOPin();
    thermostat_out->set_pin(17);
    thermostat_out->pin_mode(gpio::FLAG_OUTPUT);
    auto boiler_in = new esp32::ArduinoInternalGPIOPin();
    boiler_in->set_pin(25);
    boiler_in->pin_mode(gpio::FLAG_INPUT);
    auto boiler_out = new esp32::ArduinoInternalGPIOPin();
    boiler_out->set_pin(16);
    boiler_out->pin_mode(gpio::FLAG_OUTPUT);
    auto ot = new esphome::opentherm::OpenThermGWClimate(thermostat_in, thermostat_out, boiler_in, boiler_out);
    App.register_component(ot);
```

### Anything in the logs that might be useful for us?

```txt
error: expected type-specifier
     auto thermostat_in = new esp32::ArduinoInternalGPIOPin();
```

ssieb, 2022-12-18:
GitHub issues are not for support requests.  Ask on the esphome discord or one of the forums.
If you make this a proper component instead of using the "custom" method, you wouldn't have this problem.  You can configure the pins properly in the yaml config.

closed 2022-12-18 by ssieb

halecivo, 2022-12-18:
I’m sorry, but I really think there is an issue with documentation of the change. Good engineers describe interface changes. I could not find any detailed information nor in release notes, nor in doxygen (like first deprecation of old interface with migration note), nor in the commit message. That is why I raised the issue here.

If missing proper change description in generally well documented project is not an issue for you, okay then. 

Thanks for the direction about being able to configure pin still in the same way from yaml.

jesserockz, 2022-12-18:
Considering you found the commit that dropped it, you should be able to see the pull request number too, which has a description for the change.

These are internal classes used by esphome code generation, and not meant to be used like this.

As ssieb said, and as the [documentation states](https://esphome.io/components/sensor/custom.html#custom-sensor-component) internal APIs can and will break if needed and should not be relied on.