esphome/issues #3939: SDM120M - CRC Errors on 22.12.1

Ralphy140, 2022-12-18:
### The problem

Connecting to SDM120M using RS485 to UART adaptor is generating CRC errors and not populating sensors. single Modbus quey works fine so know i can comminicate with the modbus on the device correctly. Device works IF logger verbose is enabled. It seems that the code is processing the modbus data stream before all data has been received. Attached debug logs attached 
[logs_modbus-power_upload.txt](https://github.com/esphome/issues/files/10254817/logs_modbus-power_upload.txt)

[message (1).txt](https://github.com/esphome/issues/files/10254819/message.1.txt)

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO016
  tx_pin: GPIO017
  baud_rate: 9600 #if your energy meter is SDM230M or SDM120M than change the baud_rate: 2400
  stop_bits: 1
  debug:

  
sensor:
  - platform: sdm_meter
    phase_a:
      current:
        name: "SDM120M Current"
      voltage:
        name: "SDM120M Voltage"
      active_power:
        name: "SDM120M Power"
      power_factor:
        name: "SDM120M Power Factor"
      apparent_power:
        name: "SDM120M Apparent Power"
      reactive_power:
        name: "SDM120M Reactive Power"
      phase_angle:
        name: "SDM120M Phase Angle"
    frequency:
      name: "SDM120M Frequency"
    total_power:
      name: "SDM120M Total Power"
    import_active_energy:
      name: "SDM120M Import Active Energy"
    export_active_energy:
      name: "SDM120M Export Active Energy"
    import_reactive_energy:
      name: "SDM120M Import Reactive Energy"
    export_reactive_energy:
      name: "SDM120M Export Reactive Energy"
    update_interval: 10s
```

karllinder, 2022-12-23:
Same issue with SDM630, downgrade to 22.10.2 solved the issue temporarily.

janneman001, 2022-12-25:
Possibly duplicate of: https://github.com/esphome/issues/issues/3912

janneman001, 2022-12-25:
btw about the yaml snippet above. The sdm120m has to operate with 2400 baud.

saschaludwig, 2023-01-05:
Looks like a timing related issue. I also have the same problems.

Maybe related to this: https://github.com/esphome/issues/issues/3909

Ralphy140, 2023-01-05:
> btw about the yaml snippet above. The sdm120m has to operate with 2400 baud.

I have set the SDM120 to 9600 as well and it works just fine with a workaround of running verbose on the logger so its 100% no that casing this issue

karllinder, 2023-02-17:
see my post at https://github.com/esphome/issues/issues/3912#issuecomment-1434480862 for the latest test with 2023.2

karllinder, 2023-04-29:
@jesserockz Sorry to ping you.
Still the same issue with 2023.4.2
14:32:36][W][modbus:108]: Modbus CRC Check failed! C071!=C100
[14:32:36][W][modbus:108]: Modbus CRC Check failed! D213!=433B

closed 2024-10-29 by jesserockz