esphome/issues #3940: Docs for publish_state() refers to MQTT

scriptengine, 2022-12-19:
### The problem

In the sensor component docs for **publish_state()** is says
_"Manually cause the sensor to push out a value. It will then be processed by the sensor filters, and once done be **published to MQTT**."_
I don't believe publish state uses MQTT, but I don't have the ESPHome experience to be confident to change the manual page myself.
https://esphome.io/components/sensor/index.html
https://github.com/esphome/esphome-docs/blob/current/components/sensor/index.rst

### Which version of ESPHome has the issue?

 v2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.1

### What platform are you using?

ESP32

### Board

M5Stick - not relevant

### Component causing the issue

Manual page for 

### Example YAML snippet

```yaml
n/a
```

### Anything in the logs that might be useful for us?

```txt
n/a
```

bkaufx, 2022-12-19:
I think it would publish to MQTT.  Did you try it out?  If anything, it is a little weird to call out MQTT specifically here when it will be published via the API and logs as well, and in all 3 cases only if configured.  And potentially there are more than just those 3 places it might publish if configured to do so.  The statement in the documentation could probably be made a little more generic.

scriptengine, 2022-12-22:
Brian:

>> If anything, it is a little weird to call out MQTT specifically here when it will be published via the API
That is my point.
The update will be published via the API.

I guess it would also be published by MQTT if that was the way it was configured, but by default I don’t think MQTT is relevant here (am I right)
So that is my point.
This document should or could be reworded to remove references to MQTT and talk about the API.

If my facts are correct I’m willing to have a go at submitting the change but if it would be easier/ faster / quicker or more correct for somebody with experience to update the docs that’s no problem.

>> I think it would publish to MQTT. Did you try it out?
No. I’m not using MQTT, the native API back to Home Assistant is doing everything I want and more. MQTT is not relevant for my use case (and I don’t have it coded up.)

THANKYOU for everything you do and have done to make ESPHome what it is, it’s a wonderful piece of “kit” – thankyou

Jeremy Pack

bkaufx, 2022-12-22:
I'd probably just do a PR for the documentation if I were you.  You could either just remove "to mqtt" or you could add some description to the effect of it will publish in whatever ways you have configured.  Once you create the PR someone who manages the documentation might chime in and let you know how they want it.  Not sure how that works.

I definitely don't deserve much thanks compared to Otto, Jesse, and everyone else who actually works their butts off for the ESPHome project.

scriptengine, 2022-12-22:
Closed as moved to documentation 2565

closed 2022-12-22 by scriptengine