esphome/issues #3942: NaN on e-ink display while HA sensor has corect state

remyderuysscher, 2022-12-19:
### The problem

NaN is shown on display while logging output shows correct state of HA sensor. 

I suspect the display component is broken in combination with waveshare e-ink display.  
Time, text and glyphs on display are working fine.

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.5

### What platform are you using?

ESP32

### Board

 LILYGO-H208 

### Component causing the issue

display 

### Example YAML snippet

```yaml
sensor:
  - platform: homeassistant
    entity_id: luxtronik.temperatuur_tapwater
    id: tapwater_temp
    internal: true
	
display:
...
  it.printf(0, 0, id(roboto16), "Tapwater: %.1f", id(tapwater_temp).state);
```

### Anything in the logs that might be useful for us?

```txt
19:39:31    [D] [homeassistant.sensor:024]  

'luxtronik.temperatuur_tapwater': Got state 41.60

19:39:31    [D] [sensor:127]    

'tapwater_temp': Sending state 41.60000  with 1 decimals of accuracy
```

### Additional information

d

ssieb, 2022-12-19:
Since you're writing something on the display, it's clearly not broken.  Are you updating the display before the sensor gets a value?

ssieb, 2022-12-19:
Also, GitHub issues are not for support.  Come to the esphome discord or one of the forums.

closed 2022-12-19 by ssieb

remyderuysscher, 2022-12-19:
Which forum are you referring to? I used the Tweakers forum. I  believe this is an bug with the display component with HA sensors since the value is correctly received but the float is not displayed.  See "Got state 41.60".