esphome/issues #3943: esp-idf: bootloop after upgrading from 2022.9.2 to 2022.12.1

marrobHD, 2022-12-19:
### The problem

After upgrading firmware from 2022.9.2 to 2022.12.1 the device wont fully boot. It crashes at esp-idf wifi init.

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32doit-devkit-v1

### Component causing the issue

wifi

### Example YAML snippet

```yaml
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    # Advanced tweaking options
    advanced:
      ignore_efuse_mac_crc: true
```

### Anything in the logs that might be useful for us?

```txt
[16:48:58]Rebooting...
[16:48:58]ets Jun  8 2016 00:22:57
[16:48:58]
[16:48:58]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:48:58]configsip: 0, SPIWP:0xee
[16:48:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:48:58]mode:DIO, clock div:2
[16:48:58]load:0x3fff0018,len:4
[16:48:58]load:0x3fff001c,len:1044
[16:48:58]load:0x40078000,len:8896
[16:48:58]load:0x40080400,len:5828
[16:48:58]entry 0x400806ac
[16:48:58]I (333) cpu_start: Pro cpu up.
[16:48:58]I (333) cpu_start: Single core mode
[16:48:58]I (341) cpu_start: Pro cpu start user code
[16:48:58]I (341) cpu_start: cpu freq: 160000000
[16:48:58]I (341) cpu_start: Application information:
[16:48:58]I (346) cpu_start: Project name:     midesklamppro
[16:48:58]I (351) cpu_start: App version:      2022.12.1
[16:48:58]I (356) cpu_start: Compile time:     Dec 19 2022 15:28:56
[16:48:58]I (363) cpu_start: ELF file SHA256:  d1f9df240af3f1ed...
[16:48:58]I (369) cpu_start: ESP-IDF:          4.4.2
[16:48:58]I (374) heap_init: Initializing. RAM available for dynamic allocation:
[16:48:58]I (381) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[16:48:58]I (387) heap_init: At 3FFB7390 len 00028C70 (163 KiB): DRAM
[16:48:58]I (393) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[16:48:58]I (399) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[16:48:58]I (405) heap_init: At 400938C0 len 0000C740 (49 KiB): IRAM
[16:48:58]I (412) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[16:48:58]I (419) spi_flash: detected chip: generic
[16:48:58]I (423) spi_flash: flash io: dio
[16:48:58]I[I][logger:258]: Log initialized
[16:48:58][C][ota:469]: There have been 9 suspected unsuccessful boot attempts.
[16:48:58][D][esp32.preferences:113]: Saving 1 preferences to flash...
[16:48:58][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[16:48:58][I][app:029]: Running through setup()...
[16:48:58][C][power_supply:010]: Setting up Power Supply...
[16:48:58][D][esp-idf:000]: I (462) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[16:48:58]
[16:48:58][D][ledc.output:041]: Calculating resolution bit-depth for frequency 40000.000000
[16:48:58][D][ledc.output:046]: Resolution calculated as 10
[16:48:58][D][ledc.output:041]: Calculating resolution bit-depth for frequency 40000.000000
[16:48:58][D][ledc.output:046]: Resolution calculated as 10
[16:48:58][D][esp-idf:000]: I (509) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
[16:48:58]
[16:48:58][D][binary_sensor:034]: 'button': Sending initial state OFF
[16:48:58][C][light:036]: Setting up light 'Mi Desk Lamp Pro'...
[16:48:58][D][light:348]: 'Mi Desk Lamp Pro' - Setting cold/warm white channels using white/color temperature values.
[16:48:58][D][light:035]: 'Mi Desk Lamp Pro' Setting:
[16:48:58][D][light:040]:   Color mode: Cold/warm white
[16:48:58][D][light:046]:   State: ON
[16:48:58][D][light:050]:   Brightness: 20%
[16:48:58][D][light:065]:   Color temperature: 370.0 mireds
[16:48:58][D][light:070]:   Cold white: 19%, warm white: 100%
[16:48:58][C][rotary_encoder:128]: Setting up Rotary Encoder 'rotation'...
[16:48:58][D][esp-idf:000]: I (571) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[16:48:58]
[16:48:58][D][esp-idf:000]: I (580) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[16:48:58]
[16:48:58][C][wifi:037]: Setting up WiFi...
[16:48:58][C][wifi:038]:   Local MAC: 04:CF:8C:1B:0C:30
[16:48:59][D][esp-idf:000]: I (605) wifi:
[16:48:59][D][esp-idf:000]: wifi driver task: 3ffc193c, prio:23, stack:6656, core=0
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (624) wifi:
[16:48:59][D][esp-idf:000]: wifi firmware version: eeaa27d
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (631) wifi:
[16:48:59][D][esp-idf:000]: wifi certification version: v7.0
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (642) wifi:
[16:48:59][D][esp-idf:000]: config NVS flash: enabled
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (662) wifi:
[16:48:59][D][esp-idf:000]: config nano formating: disabled
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (663) wifi:
[16:48:59][D][esp-idf:000]: Init data frame dynamic rx buffer num: 32
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (682) wifi:
[16:48:59][D][esp-idf:000]: Init management frame dynamic rx buffer num: 32
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (693) wifi:
[16:48:59][D][esp-idf:000]: Init management short buffer num: 32
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (703) wifi:
[16:48:59][D][esp-idf:000]: Init dynamic tx buffer num: 32
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (724) wifi:
[16:48:59][D][esp-idf:000]: Init static rx buffer size: 1600
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (734) wifi:
[16:48:59][D][esp-idf:000]: Init static rx buffer num: 10
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (744) wifi:
[16:48:59][D][esp-idf:000]: Init dynamic rx buffer num: 32
[16:48:59][D][esp-idf:000]: 
[16:48:59]
[16:48:59][D][esp-idf:000]: I (755) wifi_init: rx ba win: 6
[16:48:59]
[16:48:59][D][esp-idf:000]: I (755) wifi_init: tcpip mbox: 32
[16:48:59]
[16:48:59][D][esp-idf:000]: I (764) wifi_init: udp mbox: 6
[16:48:59]
[16:48:59][D][esp-idf:000]: I (765) wifi_init: tcp mbox: 6
[16:48:59]
[16:48:59][D][esp-idf:000]: I (775) wifi_init: tcp tx win: 5744
[16:48:59]
[16:48:59][D][esp-idf:000]: I (785) wifi_init: tcp rx win: 5744
[16:48:59]
[16:48:59][D][esp-idf:000]: I (785) wifi_init: tcp mss: 1440
[16:48:59]
[16:48:59][D][esp-idf:000]: I (795) wifi_init: WiFi IRAM OP enabled
[16:48:59]
[16:48:59][D][esp-idf:000]: I (795) wifi_init: WiFi RX IRAM OP enabled
[16:48:59]
[16:48:59][D][esp-idf:000]: I (806) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
[16:48:59]
[16:48:59][D][esp-idf:000]: W (816) phy_init: fail
[16:48:59]abort() was called at PC 0x400f5a97 on core 0
[16:48:59]
[16:48:59]
[16:48:59]Backtrace:0x40082192:0x3ffc17000x40088505:0x3ffc1720 0x4008e552:0x3ffc1740 0x400f5a97:0x3ffc17b0 0x4014a854:0x3ffc17f0 0x4014a919:0x3ffc1820 0x40129872:0x3ffc1840 0x40129fe5:0x3ffc1860 0x401288c8:0x3ffc1880 0x4008fecd:0x3ffc18a0 
[16:48:59]
[16:48:59]
[16:48:59]
[16:48:59]
[16:48:59]ELF file SHA256: d1f9df240af3f1ed
[16:48:59]
[16:48:59]Rebooting...
[16:48:59]ets Jun  8 2016 00:22:57
```

Tybeevdroovaart, 2022-12-19:
Same issue over here with de D1 Mini. It also gets into a bootloop.

lawriege, 2022-12-23:
I've had some success following the CONFIG_FREERTOS_UNICORE: n then CONFIG_FREERTOS_UNICORE: y as per
https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/104#issuecomment-1356182034

mmakaay, 2023-01-15:
> Same issue over here with the D1 Mini. It also gets into a bootloop.

The issue for D1 mini is very likely not the same as this one. So you might want to open a separate issue for that one if it still is an issue. The issue concerned here is for ESP32 and specifically ESP32 with an invalid CRC checksum for the MAC address that is written on the ESP32 efuse.

**We're a step closer for fixing and preventing this issue in a future-proof way.**

I [filed a feature request](https://github.com/espressif/esp-idf) a little while ago for a build flag that we can use for not rebooting the device when the MAC CRC is invalid. The implementation for this is completed, and it was confirmed that this change will be backported into ESP-IDF 4.4. As soon as we can make use of this new build flag, I will update my Xiaomi bslamp 2 project to make use of this.