esphome/issues #3946: Device won't enter deep sleep after update

matthew73210, 2022-12-20:
### The problem

Hello,
After updating to the latest version of esphome on the 17/12/2022. The epaper has stopped updating, through the logs I can see the info being picked up from HA. 
Normally after 20s a script is called which updates the battery voltage, screen then goes to sleep. Then wake up after an hour and repeat. While running a blue LED is lit.
The latter two things don't happen, it just loops then disconnects without running the update / sleep part of the script. The blue LED stays on the whole time while never going off till the battery dies. I don't get the "entering sleep for xx" in the logs anymore.

Before the update it had worked for the past year.

### Which version of ESPHome has the issue?

2022.12.XX

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.7

### What platform are you using?

ESP32

### Board

Lilygo T5-4.7 inch e-paper

### Component causing the issue

scipt 

### Example YAML snippet

```yaml
substitutions:
  esp_name: ESP Display #Device Name
  esp_hostname: esp-display 
  run_time: 2min #can be as long as needed to get data 
  sleep_time: 1h # normal sleep time
  #night_sleep_time: 3h # 1st sleep time after midnight

esphome:
  name: e-ink-screen

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

web_server:
  port: 80

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: xxxxxxxxxxxx
  password: xxxxxxxxxxx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "E-Ink-Screen Fallback Hotspot"
    password: "lechanton"
[... 576 lines cut ...]
[14:38:19][D][homeassistant.sensor:024]: 'sensor.outside_pm_1um_weight_concentration': Got state 2.11
[14:38:19][D][sensor:127]: 'ext_1um': Sending state 2.11000  with 1 decimals of accuracy
[14:38:19][D][homeassistant.sensor:024]: 'sensor.outside_pm_2_5um_weight_concentration': Got state 2.24
[14:38:19][D][sensor:127]: 'ext_25um': Sending state 2.24000  with 1 decimals of accuracy
[14:38:19][D][homeassistant.sensor:024]: 'sensor.outside_pm_4um_weight_concentration': Got state 2.25
[14:38:19][D][sensor:127]: 'ext_4um': Sending state 2.25000  with 1 decimals of accuracy
[14:38:19][D][homeassistant.sensor:024]: 'sensor.outside_pm_10um_weight_concentration': Got state 2.25
[14:38:19][D][sensor:127]: 'ext_10um': Sending state 2.25000  with 1 decimals of accuracy
[14:38:20][D][homeassistant.sensor:024]: 'sensor.garage_co2_equivalent': Got state 1526.50
[14:38:20][D][sensor:127]: 'gar_voc': Sending state 1526.50000  with 1 decimals of accuracy
[14:38:20][D][main:375]: 20s has passed
[14:38:20][D][sensor:127]: 'ESP Display Battery Voltage': Sending state 3.47000 V with 2 decimals of accuracy
[14:38:20][D][sensor:127]: 'ESP Display Battery': Sending state 21.00000 % with 0 decimals of accuracy
INFO e-ink-screen.local: Ping timed out!
INFO Disconnected from ESPHome API for e-ink-screen.local
WARNING Disconnected from API
```

### Additional information

Usually, after this part, I'd get, "Entering sleep for xx"

[14:38:20][D][main:375]: 20s has passed
[14:38:20][D][sensor:127]: 'ESP Display Battery Voltage': Sending state 3.47000 V with 2 decimals of accuracy
[14:38:20][D][sensor:127]: 'ESP Display Battery': Sending state 21.00000 % with 0 decimals of accuracy

matthew73210, 2022-12-22:
Managed to pick up this, doesn't appear over wifi log

[D][main:375]: 20s has passed
[D][sensor:127]: 'ESP Display Battery Voltage': Sending state 4.70000 V with 2 decimals of accuracy
[D][sensor:127]: 'ESP Display Battery': Sending state 100.00000 % with 0 decimals of accuracy
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400da98f  PS      : 0x00060d30  A0      : 0x800daaf6  A1      : 0x3ffb24d0  
A2      : 0xbffb8a88  A3      : 0x3f40208f  A4      : 0x00000001  A5      : 0x3ffb2540  
A6      : 0x00000021  A7      : 0x00000020  A8      : 0x80088e6e  A9      : 0x3ffb24e0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x0ffd114c  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbffb8ab4  LBEG    : 0x4008a470  LEND    : 0x4008a486  LCOUNT  : 0x00000000  

Backtrace:0x400da98c:0x3ffb24d00x400daaf3:0x3ffb24f0 0x400e3b8d:0x3ffb2510 0x400e3bba:0x3ffb2540 0x400da81c:0x3ffb25a0 0x400da864:0x3ffb25c0 0x400e43a1:0x3ffb25e0 0x401959af:0x3ffb2600 0x401959c3:0x3ffb2620 0x401959c3:0x3ffb2640 0x401959e3:0x3ffb2660 0x400e473d:0x3ffb2680 0x40195a15:0x3ffb26a0 0x401959af:0x3ffb26c0 0x401959c3:0x3ffb26e0 0x40195295:0x3ffb2700 0x400dcfa1:0x3ffb2720 0x401956e5:0x3ffb2790 0x401957a5:0x3ffb27b0 0x400e2318:0x3ffb27d0 0x400e4a66:0x3ffb2800 0x400f1b49:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:258]: Log initialized
[C][ota:469]: There have been 2 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][spi:023]: Setting up SPI bus...
[D][binary_sensor:034]: 'ESP Display Button 3': Sending initial state OFF
[D][binary_sensor:034]: 'ESP Display Button 1': Sending initial state OFF
[D][binary_sensor:034]: 'ESP Display Button 2': Sending initial state OFF
[C][adc:044]: Setting up ADC 'ESP Display Battery Voltage'...
[C][adc:083]: ADC 'ESP Display Battery Voltage' setup finished!
[D][text_sensor:067]: 'ESP Mac Wifi Address': Sending state 'C4:DD:57:92:9E:58'

ssieb, 2023-04-22:
Your `on_time:` doesn't do what you think it does.  That will occur at the 20th second of the minute, not in 20 secs.
It's hard to say what's crashing, but does it work without the deep sleep?

AStoker, 2025-04-18:
I'm having similar behavior. I'll get the debug log saying deep sleep is about to start, but then it just keeps spitting out my sensor data logs every 3 seconds
```
[22:13:26][C][deep_sleep:026]: Setting up Deep Sleep...
[22:13:26][C][deep_sleep:029]:   Sleep Duration: 3600000 ms
[22:13:29][D][sensor:094]: 'eInk-WeatherBoard Recorded Display Refresh': Sending state 97.00000 Refreshes with 0 decimals of accuracy
[22:13:32][D][sensor:094]: 'eInk-WeatherBoard Recorded Display Refresh': Sending state 97.00000 Refreshes with 0 decimals of accuracy
....
```

Not sure if the `update_interval` on the sensor would be preventing the deep sleep?

ssieb, 2025-04-18:
That's setting it up, not starting.  Note that the log lines are `[C]` for config.

AStoker, 2025-04-21:
Gotcha, thanks for clarifying. I'm still new to all this. I discovered my reason was due to a faulty delay logic calculation, causing the whole execution to perpetually wait.