esphome/issues #3948: ESPHome Watchdog triggered with more than 100 gpios

djtef, 2022-12-20:
### The problem

I report this issue following the thread created on Home Assistant forum [here](https://community.home-assistant.io/t/esphome-watchdog-triggered-with-more-than-100-gpios/444641)

Summary : 
I made a board with 12 MCP23017 to add theoretically 12x16=192 gpio on the 2 I2C buses to my esp32.
But the esp32 is stuck in a boot loop when I add 50 outputs (switch) and 52 inputs (binary_sensor) so 102 additional gpio (and 12 covers).
It seems the watchdog timer limit is reached :
```
[11:31:42][I][app:062]: setup() finished successfully!
[11:31:42][D][esp32.preferences:114]: Saving preferences to flash...
[11:31:43]E (9958) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[11:31:43]E (9958) task_wdt:  - loopTask (CPU 1)
[11:31:43]E (9958) task_wdt: Tasks currently running:
[11:31:43]E (9958) task_wdt: CPU 0: ipc0
[11:31:43]E (9958) task_wdt: CPU 1: IDLE
[11:31:43]E (9958) task_wdt: Aborting.
[11:31:43]
[11:31:43]abort() was called at PC 0x400fd3f8 on core 0
```
I tried to increase the i2c frequency from 100 to 200KHz but nothing changed.
I tried to unplung the i2c plug from my board to have a standalone esp32 : same problem.
I can declare more gpio without reaching this limit by shrinking the id length.

After some discussion on the ESPHome Discord server @ssieb found a workaround by disabling the restore_mode of the output gpios. 
Indeed, I don't have the problem anymore if I add this line on all outputs : 
`restore_mode: ALWAYS_OFF`

@ssieb thinks that the the flash saving is taking too long. Probably because there’s so much data to save

However, I can have sometimes this error even with the workaround: 
`[18:15:58][E][esp32.preferences:145]: Error saving 1 preferences to flash. Last error=ESP_ERR_NVS_NOT_ENOUGH_SPACE for key=233825507`

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4 and poe

### Example YAML snippet

```yaml
substitutions:
  name: test

esphome:
  name: ${name}

    
esp32:
  board: esp32dev
 
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: !secret ap_ssid
    password: !secret ap_password
[... 1400 lines cut ...]
      mode: INPUT_PULLUP
      inverted: True
  - platform: gpio #in3 40
    id: in120
    pin:
      mcp23xxx: mcp23017_hub_b4
      number: 8
      mode: INPUT_PULLUP
      inverted: True
```

### Anything in the logs that might be useful for us?

```txt
[11:31:42][I][app:062]: setup() finished successfully!
[11:31:42][D][esp32.preferences:114]: Saving preferences to flash...
[11:31:43]E (9958) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[11:31:43]E (9958) task_wdt:  - loopTask (CPU 1)
[11:31:43]E (9958) task_wdt: Tasks currently running:
[11:31:43]E (9958) task_wdt: CPU 0: ipc0
[11:31:43]E (9958) task_wdt: CPU 1: IDLE
[11:31:43]E (9958) task_wdt: Aborting.
[11:31:43]
[11:31:43]abort() was called at PC 0x400fd3f8 on core 0
```

ssieb, 2022-12-20:
Try https://esphome.io/components/button/factory_reset.html, it might fix that preferences error.

djtef, 2022-12-21:
> Try https://esphome.io/components/button/factory_reset.html, it might fix that preferences error.

The factory reset seems to resolve the preferences error. Thanks @ssieb