esphome/issues #3949: wont compile get errors now with the last 2 updates of esphome

comet424, 2022-12-20:
### The problem

i getting this errors  trying to upload to esp32    last 2 updates have disabled me to update to latest
```Updating /config/esphome/vehicle1-temp-switchs.yaml
------------------------------------------------------------

INFO Reading configuration /config/esphome/vehicle1-temp-switchs.yaml...
INFO Detected timezone 'America/Toronto'
INFO Detected timezone 'America/Toronto'
INFO Generating C++ source...
INFO Compiling app...
Processing vehicle1-temp-switchs (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- AsyncMqttClient-esphome @ 0.8.6
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_connection.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_server.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/esp32/gpio.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/mqtt/mqtt_component.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/core/application.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/core/util.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/main.cpp.o
Archiving /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib01c/libFS.a
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/libbc6/Update/Updater.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib5b4/ESPAsyncWebServer-esphome/AsyncEventSource.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib5b4/ESPAsyncWebServer-esphome/AsyncWebSocket.cpp.o
Archiving /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/libbc6/libUpdate.a
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib5b4/ESPAsyncWebServer-esphome/WebAuthentication.cpp.o
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h: In instantiation of 'void esphome::globals::RestoringGlobalsComponent<T>::setup() [with T = std::__cxx11::basic_string<char>]':
src/esphome/components/globals/globals_component.h:39:8:   required from here
src/esphome/components/globals/globals_component.h:40:16: error: no matching function for call to 'esphome::ESPPreferences::make_preference<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(unsigned int)'
     this->rtc_ = global_preferences->make_preference<T>(1944399030U ^ this->name_hash_);
In file included from src/esphome/components/switch/switch.h:5,
                 from src/esphome/core/controller.h:23,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/preferences.h:58:23: note: candidate: 'template<class T, typename std::enable_if<std::is_trivially_copyable<_Tp>::value, bool>::type <anonymous> > esphome::ESPPreferenceObject esphome::ESPPreferences::make_preference(uint32_t, bool)'
   ESPPreferenceObject make_preference(uint32_t type, bool in_flash) {
                       ^~~~~~~~~~~~~~~
src/esphome/core/preferences.h:58:23: note:   template argument deduction/substitution failed:
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h:40:16: note:   candidate expects 2 arguments, 1 provided
     this->rtc_ = global_preferences->make_preference<T>(1944399030U ^ this->name_hash_);
In file included from src/esphome/components/switch/switch.h:5,
                 from src/esphome/core/controller.h:23,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/preferences.h:63:23: note: candidate: 'template<class T, typename std::enable_if<std::is_trivially_copyable<_Tp>::value, bool>::type <anonymous> > esphome::ESPPreferenceObject esphome::ESPPreferences::make_preference(uint32_t)'
   ESPPreferenceObject make_preference(uint32_t type) {
                       ^~~~~~~~~~~~~~~
src/esphome/core/preferences.h:63:23: note:   template argument deduction/substitution failed:
src/esphome/core/preferences.h:62:77: error: no type named 'type' in 'struct std::enable_if<false, bool>'
   template<typename T, enable_if_t<is_trivially_copyable<T>::value, bool> = true>
                                                                             ^~~~
*** [/data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/main.cpp.o] Error 1
========================== [FAILED] Took 8.20 seconds ==========================
==== [ERROR] /config/esphome/vehicle1-temp-switchs.yaml ====```

### Which version of ESPHome has the issue?

 2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.7

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
i dont know where in the code this is coming from.. id have to post my entire program
```

### Anything in the logs that might be useful for us?

```txt
logs dunno  just the error above
```

ssieb, 2022-12-20:
Try doing a clean build.

comet424, 2022-12-20:
didnt work
```INFO Reading configuration /config/esphome/vehicle1-temp-switchs.yaml...
INFO Detected timezone 'America/Toronto'
INFO Detected timezone 'America/Toronto'
INFO Generating C++ source...
INFO Compiling app...
Processing vehicle1-temp-switchs (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing ottowinter/AsyncMqttClient-esphome @ 0.8.6
INFO Installing ottowinter/AsyncMqttClient-esphome @ 0.8.6
Unpacking  [####################################]  100%
Library Manager: AsyncMqttClient-esphome@0.8.6 has been installed!
INFO AsyncMqttClient-esphome@0.8.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- AsyncMqttClient-esphome @ 0.8.6
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_connection.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/api_server.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/list_entities.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/esphome/components/api/proto.cpp.o
[... 112 lines cut ...]
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/preferences.h:58:23: note: candidate: 'template<class T, typename std::enable_if<std::is_trivially_copyable<_Tp>::value, bool>::type <anonymous> > esphome::ESPPreferenceObject esphome::ESPPreferences::make_preference(uint32_t, bool)'
   ESPPreferenceObject make_preference(uint32_t type, bool in_flash) {
                       ^~~~~~~~~~~~~~~
src/esphome/core/preferences.h:58:23: note:   template argument deduction/substitution failed:
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h:40:16: note:   candidate expects 2 arguments, 1 provided
     this->rtc_ = global_preferences->make_preference<T>(1944399030U ^ this->name_hash_);
In file included from src/esphome/components/switch/switch.h:5,
                 from src/esphome/core/controller.h:23,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/preferences.h:63:23: note: candidate: 'template<class T, typename std::enable_if<std::is_trivially_copyable<_Tp>::value, bool>::type <anonymous> > esphome::ESPPreferenceObject esphome::ESPPreferences::make_preference(uint32_t)'
   ESPPreferenceObject make_preference(uint32_t type) {
                       ^~~~~~~~~~~~~~~
src/esphome/core/preferences.h:63:23: note:   template argument deduction/substitution failed:
src/esphome/core/preferences.h:62:77: error: no type named 'type' in 'struct std::enable_if<false, bool>'
   template<typename T, enable_if_t<is_trivially_copyable<T>::value, bool> = true>
                                                                             ^~~~
*** [/data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/main.cpp.o] Error 1
========================= [FAILED] Took 41.94 seconds =========================```

ssieb, 2022-12-20:
Try re-installing the addon then.  It's something specific to your system, not a general problem.

comet424, 2022-12-20:
kk  just a sec

comet424, 2022-12-20:
didnt work  i still get  errors...  it worked last week cuz i working on that sleep issue that causes it to constantly wake up issue
but now i cant upload  was hoping one of these 2 esphome updates would fixed  my issue ,,

```INFO Reading configuration /config/esphome/vehicle1-temp-switchs.yaml...
INFO Detected timezone 'America/Toronto'
INFO Detected timezone 'America/Toronto'
INFO Generating C++ source...
INFO Compiling app...
Processing vehicle1-temp-switchs (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- AsyncMqttClient-esphome @ 0.8.6
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/main.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib64d/WiFi/WiFiScan.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib64d/WiFi/WiFiServer.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib64d/WiFi/WiFiUdp.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib01c/FS/FS.cpp.o
Archiving /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib64d/libWiFi.a
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib01c/FS/vfs_api.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/libbc6/Update/HttpsOTAUpdate.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/libbc6/Update/Updater.cpp.o
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib5b4/ESPAsyncWebServer-esphome/AsyncEventSource.cpp.o
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h: In instantiation of 'void esphome::globals::RestoringGlobalsComponent<T>::setup() [with T = std::__cxx11::basic_string<char>]':
src/esphome/components/globals/globals_component.h:39:8:   required from here
src/esphome/components/globals/globals_component.h:40:16: error: no matching function for call to 'esphome::ESPPreferences::make_preference<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(unsigned int)'
     this->rtc_ = global_preferences->make_preference<T>(1944399030U ^ this->name_hash_);
In file included from src/esphome/components/switch/switch.h:5,
                 from src/esphome/core/controller.h:23,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/preferences.h:58:23: note: candidate: 'template<class T, typename std::enable_if<std::is_trivially_copyable<_Tp>::value, bool>::type <anonymous> > esphome::ESPPreferenceObject esphome::ESPPreferences::make_preference(uint32_t, bool)'
   ESPPreferenceObject make_preference(uint32_t type, bool in_flash) {
                       ^~~~~~~~~~~~~~~
src/esphome/core/preferences.h:58:23: note:   template argument deduction/substitution failed:
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h:40:16: note:   candidate expects 2 arguments, 1 provided
     this->rtc_ = global_preferences->make_preference<T>(1944399030U ^ this->name_hash_);
In file included from src/esphome/components/switch/switch.h:5,
                 from src/esphome/core/controller.h:23,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/preferences.h:63:23: note: candidate: 'template<class T, typename std::enable_if<std::is_trivially_copyable<_Tp>::value, bool>::type <anonymous> > esphome::ESPPreferenceObject esphome::ESPPreferences::make_preference(uint32_t)'
   ESPPreferenceObject make_preference(uint32_t type) {
                       ^~~~~~~~~~~~~~~
src/esphome/core/preferences.h:63:23: note:   template argument deduction/substitution failed:
src/esphome/core/preferences.h:62:77: error: no type named 'type' in 'struct std::enable_if<false, bool>'
   template<typename T, enable_if_t<is_trivially_copyable<T>::value, bool> = true>
                                                                             ^~~~
Archiving /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib01c/libFS.a
Compiling /data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/lib5b4/ESPAsyncWebServer-esphome/AsyncWebSocket.cpp.o
*** [/data/vehicle1-temp-switchs/.pioenvs/vehicle1-temp-switchs/src/main.cpp.o] Error 1
========================== [FAILED] Took 9.24 seconds ==========================```

comet424, 2022-12-20:
if its specific to home assistant how do i downgrade  esphome 2 updates and go back to a previous version that did work

comet424, 2022-12-20:
all my other esp 8826 and esp32s compile  just wont complile in 3 of my codes of my car temperature display.. so far i just use 1 of them so something speicifc in the that yaml   is not liked anymore  all my others will compile

ssieb, 2022-12-20:
If you allowed HA to backup the addon when you upgraded, then you can restore the previous versions.  Can you paste your config that isn't working?

comet424, 2022-12-20:
i deleted the backups  logs keep filling up or so  so HA is always full..  ill past my code.. this is the code where i have the problem  where if ESP32 goes to sleep when there is no wifi  it constantly wakes up as it goes to sleep then wakes up..  where i posted it in my other topic  but here is the code
and this is where i find the one esp32s  crashs when first plugged in i get the blue and red led on the board and have to press reboot button to get it to boot up

```
esphome:
  name: vehicle1-temp-switchs

##############################################
### On Shut Down, Sleep The Nextion Screen ###
##############################################
  on_shutdown:
    then:
        - lambda: |-
           id(nextion1).sleep(true);

esp32:
  board: nodemcu-32s #esp-wrover-kit
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "nfcIMjlaXqJMuwLJV2r8o+IT4pGZHoWS6oqv1w0WTyU="
  services:
    - service: update_nextion
      then:
        - lambda: 'id(nextion1)->upload_tft();'
ota:
  password: "d999903d78ca32e5a1be30d01e4aead2"

wifi:
  networks:
# House Wifi
  - ssid: "test"#!secret wifi_ssid
    password: !secret wifi_password
# Gate Wifi
#  - ssid: !secret gate_wifi_ssid
#    password: !secret gate_wifi_password

#  manual_ip:
#    static_ip: 192.168.0.246
#    gateway: 192.168.0.1
#    subnet: 255.255.255.0  
  ap:
    ssid: "vehicle1-temp-switchs"
    password: "kevintt123"

  
web_server:
captive_portal:
    

#####################
### Global Values ###
#####################
globals:
  - id: id_broker
    type: std::string
    restore_value: yes
    initial_value: '"192.168.0.12"'
  - id: current_brightness       # Display Value   Current
    type: float
    restore_value: no
    initial_value: '100'
  - id: oldvalue                 # Display Value  Old Value
[... 415 lines cut ...]
       ///   id(current_brightness)= oldvalue;    //set the brightness to saved value when key is turned on
        ////////////////////////////////////
        /// Displays Temperature Outside ///
        ///////////////////////////////////
          //it.set_component_text_printf("dtemp", "%.1f c", id(outtemp).state);
          it.set_component_text_printf("dtemp", "%.1f", id(outtemp).state);
        //////////////////////////////////////
        /// Syncs Time With Home Assistant ///
        //////////////////////////////////////
          it.set_nextion_rtc_time(id(date_time_iso).now());
        ///}
        ///else
        ///{
        ///  br=0;  // Set brightness to 0 To Shut off Screen
        ///}
      ///////////////////////////////////////////////////
      /// Change brightness if it needs to be changed ///
      ///////////////////////////////////////////////////
      ///if (id(current_brightness)!=br) {  
      ///  oldvalue = id(current_brightness);  // Save The Current Screen Brightness
      ///  it.set_backlight_brightness(br);
      ///  id(current_brightness)=br;
      ///  }

```

RaceAap, 2022-12-23:
I have a similar issue like this: #3927
At the moment I'm using a standalone docker version of ESPHome (outside of HomeAssistant) so I could go on with my project.

jesserockz, 2022-12-23:
The HA backup for ESPHome is `10kb` each version. Deleting them is not giving you any space back =)

jesserockz, 2022-12-23:
```yaml
globals:
  - id: id_broker
    type: std::string
    restore_value: yes
    initial_value: '"192.168.0.12"'
```

You cannot have restoring string globals. It just doesnt work

comet424, 2022-12-23:
welll not 100% sure what restoring value doesi i was googling looking how to work it.. as i wanted my projected to connect to 2 different mqtt broker servers.. if its on this wifi network  you use this ip address for mqtt server...   when you use this other network  then you use this mqtt server.....    and when i mentioned about deleting backups  i talking about deleting them all.. so its blank  and logs etc.. yet still fills up