esphome/issues #3951: Bluetooth-proxy doesn't work for HA anymore

txitxo0, 2022-12-21:
### The problem

Hi, 

After the last updates the bluetooth-proxy device I had working from 2 months ago just didn't proxy packets to HA anymore. 
I was following [this other issue](https://github.com/esphome/issues/issues/3913) related but it didn't fix mine.

I have been working in update the framework to use esp-idf as they ask me for and it is properly commented in docs ([in the warning section](https://esphome.io/components/bluetooth_proxy.html)) but nothing appears to fix this up. 

I am using an ESP32-wroom in an Esphome addon 2022.12.3 in a Home Assistant OS 2022.12.7

I am trying to proxying two different [xiaomi ble thermometers](https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc) who don't have very good coverage and they are both working before with the same atc firmware. One of them to be able to look at it is the one wich the MAC=A4:C1:38:BE:CE:9B

Apparently is working and the device log shows the sentence: "Proxying packet from ATC_BECE9B - A4:C1:38:BE:CE:9B. RSSI: -49 dB". But nothing wrong appears in logs from both sides, HA (aioesphome) or ESPHome device log.

The rest of the behaviour of this devices who has two more sensors (PIR and illuminance) is working fine.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.7

### What platform are you using?

ESP32-IDF

### Board

esp32-wroom

### Component causing the issue

bluetooth proxy

### Example YAML snippet

```yaml
substitutions:
  name: "btproxy-estudio"
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

logger:
  level: VERBOSE

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.3.50
    gateway: 192.168.1.1
    subnet: 255.255.255.0

binary_sensor:
  - platform: gpio
    pin: GPIO26
    name: "${name}-motion"
    device_class: motion
    filters:
      - delayed_on: 250ms

sensor:
  - platform: adc
    pin: GPIO32
    name: "${name}-illuminance"
    device_class: illuminance
    unit_of_measurement: lx
    update_interval: 60s
    attenuation: auto
    filters:
      - lambda: |-
          return (x / 10000.0) * 2000000.0 * 1.67;
    
    # Uptime sensor
  - platform: template
    name: btproxy Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL) / 1024;
    unit_of_measurement: 'kB'
    state_class: measurement
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-estudio.yaml...
INFO Starting log output from 192.168.3.50 using esphome API
INFO Successfully connected to 192.168.3.50
[12:51:23][I][app:102]: ESPHome version 2022.12.3 compiled on Dec 21 2022, 08:33:54
[12:51:23][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[12:51:23][C][wifi:504]: WiFi:
[12:51:23][C][wifi:362]:   Local MAC: E0:5A:1B:75:8A:90
[12:51:23][C][wifi:363]:   SSID: [redacted]
[12:51:23][C][wifi:364]:   IP Address: 192.168.3.50
[12:51:23][C][wifi:366]:   BSSID: [redacted]
[12:51:23][C][wifi:367]:   Hostname: 'btproxy-estudio'
[12:51:23][C][wifi:369]:   Signal strength: -53 dB ▂▄▆█
[12:51:23][V][wifi:371]:   Priority: 0.0
[12:51:23][C][wifi:373]:   Channel: 6
[12:51:23][C][wifi:374]:   Subnet: 255.255.255.0
[12:51:23][C][wifi:375]:   Gateway: 192.168.1.1
[12:51:23][C][wifi:376]:   DNS1: 0.0.0.0
[12:51:23][C][wifi:377]:   DNS2: 0.0.0.0
[12:51:23][V][bluetooth_proxy:033]: Proxying packet from  - 46:B0:00:1B:87:C5. RSSI: -81 dB
[12:51:23][C][logger:293]: Logger:
[12:51:23][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'btproxy-estudio-motion'
[12:51:23][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[12:51:23][C][gpio.binary_sensor:016]:   Pin: GPIO26
[12:51:23][C][template.sensor:023]: Template Sensor 'btproxy Free Memory'
[12:51:23][C][template.sensor:023]:   State Class: 'measurement'
[12:51:23][C][template.sensor:023]:   Unit of Measurement: 'kB'
[12:51:23][C][template.sensor:023]:   Accuracy Decimals: 1
[12:51:23][C][template.sensor:024]:   Update Interval: 60.0s
[12:51:23][C][bluetooth_proxy:065]: Bluetooth Proxy:
[12:51:23][C][bluetooth_proxy:066]:   Active: YES
[12:51:23][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode Boot'
[12:51:23][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[12:51:23][C][adc:087]: ADC Sensor 'btproxy-estudio-illuminance'
[12:51:23][C][adc:087]:   Device Class: 'illuminance'
[12:51:23][C][adc:087]:   State Class: 'measurement'
[12:51:23][C][adc:087]:   Unit of Measurement: 'lx'
[12:51:23][C][adc:087]:   Accuracy Decimals: 2
[12:51:23][C][adc:097]:   Pin: GPIO32
[12:51:23][C][adc:099]:  Attenuation: auto
[12:51:23][C][adc:126]:   Update Interval: 60.0s
[12:51:23][C][esp32_ble_tracker:870]: BLE Tracker:
[12:51:23][C][esp32_ble_tracker:871]:   Scan Duration: 300 s
[12:51:23][C][esp32_ble_tracker:872]:   Scan Interval: 1100.0 ms
[12:51:23][C][esp32_ble_tracker:873]:   Scan Window: 1100.0 ms
[12:51:23][C][esp32_ble_tracker:874]:   Scan Type: ACTIVE
[12:51:23][C][esp32_ble_tracker:875]:   Continuous Scanning: True
[12:51:23][C][mdns:103]: mDNS:
[12:51:23][C][mdns:104]:   Hostname: btproxy-estudio
[12:51:23][V][mdns:105]:   Services:
[12:51:23][V][mdns:107]:   - _esphomelib, _tcp, 6053
[12:51:23][V][mdns:109]:     TXT: version = 2022.12.3
[12:51:23][V][mdns:109]:     TXT: mac = e05a1b758a90
[12:51:23][V][mdns:109]:     TXT: platform = ESP32
[12:51:23][V][mdns:109]:     TXT: board = esp32dev
[12:51:23][V][mdns:109]:     TXT: network = wifi
[12:51:23][V][mdns:109]:     TXT: project_name = esphome.bluetooth-proxy
[12:51:23][V][mdns:109]:     TXT: project_version = 1.0
[12:51:23][V][mdns:109]:     TXT: package_import_url = github://esphome/bluetooth-proxies/esp32-generic.yaml@main
[12:51:23][C][ota:093]: Over-The-Air Updates:
[12:51:23][C][ota:094]:   Address: 192.168.3.50:3232
[12:51:23][C][api:138]: API Server:
[12:51:23][C][api:139]:   Address: 192.168.3.50:6053
[12:51:23][C][api:143]:   Using noise encryption: NO
[12:51:23][C][improv_serial:032]: Improv Serial:
[12:51:24][V][bluetooth_proxy:033]: Proxying packet from ATC_162894 - A4:C1:38:16:28:94. RSSI: -32 dB
[12:51:26][V][bluetooth_proxy:033]: Proxying packet from ATC_162894 - A4:C1:38:16:28:94. RSSI: -33 dB
[12:51:26][V][bluetooth_proxy:033]: Proxying packet from ATC_BECE9B - A4:C1:38:BE:CE:9B. RSSI: -49 dB
[12:51:27][V][bluetooth_proxy:033]: Proxying packet from  - 46:B0:00:1B:87:C5. RSSI: -82 dB
[12:51:28][V][bluetooth_proxy:033]: Proxying packet from  - 46:B0:00:1B:87:C5. RSSI: -83 dB
```

syphernl, 2022-12-21:
The BT Proxy in ESPHome 2022.12.x is also pretty unstable for me. 
I run it on two ESP32's currently.

For one ESP all devices in its vicinity showed as unavailable in HASS after a few hours. 
After a reboot of the ESP all devices are reachable again.

I still had `esphome logs` open to another ESP32 (which seems to work fine as all devices in its vicinity were showing as available) but this one showed:
```
[12:49:26][D][esp32_ble_tracker:327]: Starting scan...
[13:19:26][D][esp32_ble_tracker:327]: Starting scan...
[13:29:26][E][esp32_ble_tracker:344]: ESP-IDF BLE scan never terminated, rebooting to restore BLE stack...
```

nagyrobi, 2022-12-21:
Works fine here, Olimex board. Passively proxying 11 Xiaomi sensors with pvvx firmware in BTHome mode.

Ulrar, 2022-12-22:
The other issue's resolution did fix my Switchbot's, but I noticed yesterday that I had another device (Airthings Wave+) that was still misbehaving.
I thought it might be a range issue so I moved one of my bluetooth proxies close to it but no improvement.

Rolling back to 2022.11 solves the issue, so it is a problem present in all 2022.12.x versions. You can clearly tell when I rolled back : 
[image]

The[ other issue](https://github.com/esphome/issues/issues/3913) was a problem with the ServiceData not being parsed correctly, so the integration's diagnostics still showed the devices but without the ServiceData.
In this case I've looked at it and it's a different issue, the diagnostics do not always show this device, it's completely missing from the list sometimes. And sometimes it's there.
The issue from HA's side is also different as it does log an error saying it failed to get the 'battery' attribute, but I suspect that's just weird error handling in the integration.

I've noticed that restarting HA always gets you one good reading on init, and moving proxies around also works once, so it looks to me like you can get one good reading per proxy per HA restart.
I'd say it's an ESPHome integration problem but reloading the proxy's integration doesn't seem to make a difference.

Is there a way to get something like a CPU Load sensor for the board ? I saw a few 'Too many BLE events' messages on one of my proxies so I wonder if they might be getting overloaded on 2022.12 somehow, but it's hard to tell, in VERY_VERBOSE there's just so much log it's hard to follow. That might be a red herring, I just saw that once while looking at something else, it might be unrelated.

Anyway I'm assuming this is related to OP's issue but maybe not, just wanted to chime in with some details in case it is.

mutlucan96, 2022-12-22:
I also have a similar problem. Everything was working fine on 2022.11. I've updated it to 2022.12.3 via serial port. The proxy is working fine for somewhere between 1 hour to 24 hours. It's still connected, I can access logs via Wi-Fi, but there aren't any BLE logs anymore:
`INFO Reading configuration /config/esphome/salon2.yaml...
INFO Starting log output from 192.168.1.189 using esphome API
INFO Successfully connected to 192.168.1.189
[10:20:30][I][app:102]: ESPHome version 2022.12.2 compiled on Dec 20 2022, 09:41:52
[10:20:30][C][wifi:504]: WiFi:
[10:20:30][C][wifi:362]:   Local MAC: 94:B9:7E:FA:9F:AC
[10:20:30][C][wifi:363]:   SSID: 'Mutlu'[redacted]
[10:20:30][C][wifi:364]:   IP Address: 192.168.1.189
[10:20:30][C][wifi:366]:   BSSID: 3C:7C:3F:E0:8D:50[redacted]
[10:20:30][C][wifi:367]:   Hostname: 'salon2'
[10:20:30][C][wifi:369]:   Signal strength: -59 dB ▂▄▆█
[10:20:30][C][wifi:373]:   Channel: 1
[10:20:30][C][wifi:374]:   Subnet: 255.255.255.0
[10:20:30][C][wifi:375]:   Gateway: 192.168.1.1
[10:20:30][C][wifi:376]:   DNS1: 192.168.1.1
[10:20:30][C][wifi:377]:   DNS2: 8.8.8.8
[10:20:30][C][logger:293]: Logger:
[10:20:30][C][logger:294]:   Level: VERBOSE
[10:20:30][C][logger:295]:   Log Baud Rate: 115200
[10:20:30][C][logger:296]:   Hardware UART: UART0
[10:20:30][C][bluetooth_proxy:065]: Bluetooth Proxy:
[10:20:30][C][bluetooth_proxy:066]:   Active: YES
[10:20:30][C][esp32_ble_tracker:870]: BLE Tracker:
[10:20:30][C][esp32_ble_tracker:871]:   Scan Duration: 300 s
[10:20:30][C][esp32_ble_tracker:872]:   Scan Interval: 1100.0 ms
[10:20:30][C][esp32_ble_tracker:873]:   Scan Window: 1100.0 ms
[10:20:31][C][esp32_ble_tracker:874]:   Scan Type: PASSIVE
[10:20:31][C][esp32_ble_tracker:875]:   Continuous Scanning: True
[10:20:31][C][mdns:103]: mDNS:
[10:20:31][C][mdns:104]:   Hostname: salon2
[10:20:31][V][mdns:105]:   Services:
[10:20:31][V][mdns:107]:   - _esphomelib, _tcp, 6053
[10:20:31][V][mdns:109]:     TXT: version = 2022.12.2
[10:20:31][V][mdns:109]:     TXT: mac = 94b97efa9fac
[10:20:31][V][mdns:109]:     TXT: platform = ESP32
[10:20:31][V][mdns:109]:     TXT: board = nodemcu-32s
[10:20:31][V][mdns:109]:     TXT: network = wifi
[10:20:31][C][ota:093]: Over-The-Air Updates:
[10:20:31][C][ota:094]:   Address: 192.168.1.189:3232
[10:20:31][C][ota:097]:   Using Password.
[10:20:31][C][api:138]: API Server:
[10:20:31][C][api:139]:   Address: 192.168.1.189:6053
[10:20:31][C][api:141]:   Using noise encryption: YES
[10:20:38][V][esp32_ble_client:111]: [0] [52:C7:F9:B3:08:37] gattc_event_handler: event=41 gattc_if=4
[10:20:38][V][esp32_ble_client:111]: [1] [] gattc_event_handler: event=41 gattc_if=5
[10:20:38][V][esp32_ble_client:111]: [2] [70:F0:90:EA:7C:34] gattc_event_handler: event=41 gattc_if=6`

After I reboot the ESP, everything is working fine. I can see BLE signals are detected on the logs until I can't. I couldn't catch the logs when it stopped working because the browser crashed when I keep the logs open for a long time on VERBOSE mode.

This is my config:
`esphome:
  name: salon2
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "Mutlu"
  password: "***"
  fast_connect: true
  manual_ip:
    static_ip: 192.168.1.189
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 8.8.8.8
# Enable logging
#logger:
  #level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:
  password: "***"

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: false

bluetooth_proxy:
  active: true`

I don't know if it's a coincidence but I feel like Bluetooth Proxy works longer when the log level is lower. When I set the log level to VERBOSE, it stopped working within an hour. When I set it to DEBUG it worked for hours. When I don't enable the logger at all, it works for more than a day.

flacjacket, 2022-12-23:
Echoing the issues seen here with 2022.12, both with a serial update and an OTA update from 2022.11. I also have an Airthings Wave+ and it will connect successfully as an active bluetooth proxy device when HA starts up, but will go unavailable after 5 minutes. Rolling back ESPHome to 2022.11 OTA fixes the problem.

grobius, 2022-12-23:
Sorry a newbie question, but how to you roll back to 2022.12 -> 2022.11? Remove ESPHome from HA and reinstall, the recompile devices?

Ulrar, 2022-12-23:
@grobius You just need to use the container tagged `esphome/esphome:2022.11` and then update your devices as usual.
That said if you're using the Hass OS addon, I have no idea on how you'd do that sorry

mutlucan96, 2022-12-23:
@grobius for the Home Assistant OS add-on, if you have a backup, only restore the ESPHome add-on then update your devices.

grobius, 2022-12-23:
> @grobius You just need to use the container tagged `esphome/esphome:2022.11` and then update your devices as usual. That said if you're using the Hass OS addon, I have no idea on how you'd do that sorry

Thanks a lot!

txitxo0, 2022-12-28:
Not for me. I have 5 LYWSD03MMC just two of them experiment problems because of coverge so I started to use the esphome bt proxy and no problems anymore until this release. After that they both doesn't work in that place with a poor coverage unless I move then close to the HA server itself. So for me it is still not working with (at least) this xiaomi ble thermometer.

I was trying a new configuration thinking it will be better, but apparently not and the problem persists. My current config is this below:

```
substitutions:
  name: "btproxy-estudio"

esphome:
  name: ${name}
  name_add_mac_suffix: false

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

logger:
api:
ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

binary_sensor:
  - platform: gpio
    pin: GPIO26
    name: "${name}-motion"
    device_class: motion
    filters:
      - delayed_on: 250ms

sensor:
  - platform: adc
    pin: GPIO32
    name: "${name}-illuminance"
    device_class: illuminance
    unit_of_measurement: lx
    update_interval: 60s
    attenuation: auto
    filters:
      - lambda: |-
          return (x / 10000.0) * 2000000.0 * 1.67;
    
    # Uptime sensor
  - platform: template
    name: btproxy Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL) / 1024;
    unit_of_measurement: 'kB'
    state_class: measurement
```

grobius, 2022-12-28:
> sdkconfig_options:
>       CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

Ok, out of curiosity what does the CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y enable?

Did you reflash the proxy over serial (not OTA), mine was not working at all until I reflashed via serial

Ulrar, 2022-12-28:
Can't talk to that other issue but in my case yes, it's flashed over serial.
I do have other bluetooth devices working fine with the new firmware so it's something to do with the Airthings, or more likely with a type of announces

txitxo0, 2022-12-28:
> > sdkconfig_options:
> >       CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
> 
> Ok, out of curiosity what does the CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y enable?
> 
> Did you reflash the proxy over serial (not OTA), mine was not working at all until I reflashed via serial

I read somewhere, for some boards the Bluetooth device is not enabled by default, so I just tried, but it didn't work better than before.

Yes, I burnt it directly via USB, not OTA to let the build format partitions and this stuff

lio0909, 2022-12-31:
Same issue here. 
The log:
```[10:13:41][E][esp32_ble_tracker:344]: ESP-IDF BLE scan never terminated, rebooting to restore BLE stack...
[10:13:41][I][app:127]: Forcing a reboot...```
And after 5 minutes I get this:
```[10:18:43][I][ota:113]: Boot seems successful, resetting boot loop counter.
[10:18:43][D][esp32.preferences:113]: Saving 1 preferences to flash...
[10:18:43][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:18:44][D][esp32_ble_tracker:327]: Starting scan...
```
until it reboots completely, between 1 and 5-6 hours...

Ulrar, 2022-12-31:
I haven't noticed any reboots, that might be a different problem.
I'll add an uptime sensor to confirm

Ulrar, 2023-01-02:
My uptime sensor is at 58 hours so clearly it's not rebooting, that must be a different issue again

lio0909, 2023-01-02:
ok, seems 2022.12.3 brings a lot of challenges... let's hope it gets fixed in a 2023.x.x version

nagyrobi, 2023-01-02:
I've got 286 hours uptime on my Olimex board which also runs RF Bridge with 22 timed covers, and a relay, and web interface...

wormiedk, 2023-01-07:
I think this issue: https://github.com/home-assistant/core/issues/82453#issuecomment-1374605386 Is related. After I downgraded to 2022.11.5 my airthings devices started working again

mutlucan96, 2023-01-08:
2022.11.1 is also disconnecting and needs reboot sometimes but not as frequently.  I've made automation, when my Bluetooth entity goes unavailable, it reboots ESP.

grobius, 2023-01-08:
> I think this issue: [home-assistant/core#82453 (comment)](https://github.com/home-assistant/core/issues/82453#issuecomment-1374605386) Is related. After I downgraded to 2022.11.5 my airthings devices started working again

thanks, I will test that. I asume that you reverted back to using the arduino framework? (  framework:
    type: arduino)

Ulrar, 2023-01-08:
> I asume that you reverted back to using the arduino framework? ( framework: type: arduino)

That's right.
That said the issue in 2022.12 is present regardless of framework, from what I've seen

wormiedk, 2023-01-08:
> > I asume that you reverted back to using the arduino framework? ( framework: type: arduino)
> 
> That's right. That said the issue in 2022.12 is present regardless of framework, from what I've seen

@Ulrar I just realised that my ESP crashes after 2-3 readouts from my airthings sensors. Could you share the complete config for ESPHOME 2023.11 that you are using?

mkbrown, 2023-01-08:
Adding to the list of AirThings Wave+ issues… brand new wave+ device, with Olimex ESP32-EVB flashed with Bluetooth BLE proxy.

errors in Home Assistant integration page:

Retrying setup: Unable to fetch data: BT-MAC - BT-MAC: Failed to connect: No backend with an available connection slot that can reach address BT-MAC was …

and

Retrying setup: Could not find Airthings device with address BT-MAC

Other log messages:

Error fetching airthings_ble data: Unable to fetch data: 'battery'

Config entry 'Airthings Wave+ (serial#)' for airthings_ble integration not ready yet: Unable to fetch data: 'battery'; Retrying in background

BT-MAC Bluetooth device connection timed out but already unsubscribed

Ulrar, 2023-01-08:
@wormiedk : I added an uptime sensor to one of mine to confirm it wasn't crashing a while ago, it's up to 200 hours now so clearly that's not my issue.
Config is just regular old bluetooth proxy : 

```
esphome:
  name: esp32-relay-livingroom

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  password: !secret esp32-relay-livingroom-password
  encryption:
    key: !secret esp32-relay-livingroom-key

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_pass
  fast_connect: !secret wifi_iot_fast
  domain: .mydomain.tld

mdns:
  disabled: true

ota:
  password: !secret esp32-relay-livingroom-ota

bluetooth_proxy:
  active: true

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

button:
  - platform: restart
    name: Restart

switch:
  - platform: safe_mode
    name: use_safe_mode

sensor:
  - platform: uptime
    name: Uptime Sensor
```

I'm seeing the same behavior as described by @mkbrown, on startup of either HA or the board I get one reading from the AirThings Wave+ and the integration setups fine.
Then every time I get `Error fetching airthings_ble data: Unable to fetch data: 'battery'`, including if I try to reload the integration, until I restart either the board of HA itself.

mkbrown, 2023-01-08:
Adding some debug logging from home assistant. It does look like the ESP BTLE proxy reboots/restarts, because logs are lost there.

hass logs

`

2023-01-08 09:49:29.007 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hubitat which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-08 09:49:29.013 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ical which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-08 09:49:31.415 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Airthings Wave Mini (SERIAL#)' for airthings_ble integration not ready yet: Could not find Airthings device with address MINI-MAC; Retrying in background

2023-01-08 09:49:31.716 WARNING (MainThread) [homeassistant.components.bluetooth_le_tracker.device_tracker] No Bluetooth LE devices to track!
2023-01-08 09:49:31.717 ERROR (MainThread) [homeassistant.components.device_tracker] Error setting up platform legacy bluetooth_le_tracker

2023-01-08 09:49:41.272 WARNING (MainThread) [homeassistant.bootstrap] Support for the running Python version 3.9.2 is deprecated and will be removed in Home Assistant 2023.2; Please upgrade Python to 3.10

2023-01-08 09:49:43.280 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.airthings_wave_mini_SERIAL#_voc (<class 'homeassistant.components.airthings_ble.sensor.AirthingsSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+airthings_ble%22

2023-01-08 09:56:03.563 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.airthings_wave_mini_SERIAL#_voc (<class 'homeassistant.components.airthings_ble.sensor.AirthingsSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+airthings_ble%22
2023-01-08 10:13:52.665 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.airthings_wave_SERIAL#_voc (<class 'homeassistant.components.airthings_ble.sensor.AirthingsSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+airthings_ble%22

2023-01-08 10:18:53.449 ERROR (MainThread) [homeassistant.components.airthings_ble] Error fetching airthings_ble data: Unable to fetch data: 'battery'
2023-01-08 10:34:52.511 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Airthings Wave+ (SERIAL#)' for airthings_ble integration not ready yet: Unable to fetch data: 'battery'; Retrying in background
2023-01-08 10:35:22.438 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Airthings Wave+ (SERIAL#)' for airthings_ble integration not ready yet: Unable to fetch data: 'battery'; Retrying in background

2023-01-08 11:31:21.748 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 1.582 seconds (success: True)
2023-01-08 11:32:11.554 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esp32-btle-proxy-042950 @ IPADDR: Error while reading data: [Errno 104] Connection reset by peer
2023-01-08 11:32:23.100 WARNING (MainThread) [aioesphomeapi.client] F4:60:77:75:48:CB: Bluetooth device connection timed out but already unsubscribed
2023-01-08 11:32:24.815 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 26.717 seconds (success: False)
2023-01-08 11:33:45.778 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.546 seconds (success: False)
2023-01-08 11:35:03.220 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.572 seconds (success: False)
2023-01-08 11:35:03.221 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Airthings Wave+ (SERIAL#)' for airthings_ble integration not ready yet: Unable to fetch data: 'battery'; Retrying in background
2023-01-08 11:35:08.981 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.543 seconds (success: False)
2023-01-08 11:35:19.981 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.576 seconds (success: False)
2023-01-08 11:35:40.824 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.539 seconds (success: False)
2023-01-08 11:36:22.038 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.873 seconds (success: True)
[... cut ...]
2023-01-08 12:11:04.676 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.598 seconds (success: False)
2023-01-08 12:11:36.241 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 2.076 seconds (success: True)
2023-01-08 12:12:25.483 DEBUG (MainThread) [homeassistant.components.airthings_ble] Finished fetching airthings_ble data in 0.596 seconds (success: False)
`

The logs from the ESP at the same times are gone.  Would have to set up multiple sessions to capture everything simultaneously

Ulrar, 2023-01-08:
@mkbrown Are you sure it's restarting ? That's so strange that I'm not seeing that, can you add an uptime sensor to confirm ?

chrisbertsch, 2023-01-08:
I opened an issue yesterday in home-assistant/core#85413 as I don't have any issues with ESPHome version 2022.12.3 for my other Bluetooth devices via the Bluetooth proxy such as SwitchBot and iBeacons. The only integration that is problematic for me is just Airthings BLE. I have been running the Bluetooth proxy on an Olimex ESP32-POE-ISO with no unexpected restarts of EPSHome per my uptime sensor for a few weeks now and ESPHome 2022.12.3 works fine with Home Assistant 2022.11.5. From iterating through the releases of Home Assistant it seems like the issue was introduced between Home Assistant versions 2022.12.0b1 and 2022.12.0b2.

mkbrown, 2023-01-09:
Confirming that the patched file @chrisbertsch references  above fixes the issue I had reported here along with the OP.

[image]

Luminance doesn’t exist on the Wave+.  Battery shows as unknown, but there is no battery shown as exposed in the Wave Mini I also have, so I presume that it isn’t yet decoded in the AirThings_BLE library.

@Ulrar I kind of “assumed” it was restarting, because every time I visited the web console logs, they showed the network Config like a fresh start. I “assumed” that the logger operated ring buffer in RAM to keep the most recent messages, but both assumptions were obviously wrong. Really wish ESPHOME had a syslog target like Tasmota does…

Added an uptime sensor and a free memory sensor on my btproxy and will monitor the logs for a bit…

hope that helps!

Ulrar, 2023-01-09:
@mkbrown yeah logs always start when you connect, that's expected unfortunately. 

There's definitely a bug here somewhere, I believe battery should be reported with every read. Or at least it was until the last month updates

grobius, 2023-01-10:
Rolled back to 2022.11.5. Reading now works again. However I get "wavey" reading on temp (same on 2 units) which I didn't have before. Anyone else having the same?

[image]

Ulrar, 2023-01-10:
I haven't noticed that on mine no, that's weird

Ulrar, 2023-01-10:
Just saw this, which I may have missed before I suppose : 
```
esp32-hotwater-sensor [10:97:bd:36:4f:00]: - <wave+ mac>: ESPHomeClient bleak client was not properly disconnected before destruction
esp32-hallway-sensor [58:bf:25:9e:b5:80]: - <wave+ mac>: ESPHomeClient bleak client was not properly disconnected before destruction
```

Could that be the issue, the connection is left in a bad state so the next re-connect is messed up ?
It's also doing it from multiple proxy at the same time for some reason, although that may be expected I don't know.

vqvu, 2023-01-12:
I have an Airthings Wave+ as well and ran into the same problem as others here.

From the esphome logs, I noticed that connections that work (the one that happens immediately after HA restart) connect "without cache"

```
[03:35:48][I][bluetooth_proxy:254]: [0] [<mac>] Connecting v3 without cache
[03:35:48][D][esp32_ble_tracker:216]: Pausing scan to make connection...
[03:35:48][I][esp32_ble_client:064]: [0] [<mac>] 0x00 Attempting BLE connection
[03:35:48][D][esp-idf:000]: W (29978) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[03:35:49][D][esp32_ble_tracker:327]: Starting scan...
[03:35:51][I][esp32_ble_client:078]: [0] [<mac>] Disconnecting.
[03:35:51][D][esp-idf:000]: W (32381) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[03:35:51][D][esp-idf:000]: W (32385) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[03:35:51][D][esp-idf:000]: W (32387) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[03:35:51][D][esp-idf:000]: W (32389) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[03:35:51][D][esp-idf:000]: W (32401) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
```

while ones that don't work (missing battery) connect with cache

```
[03:40:51][I][bluetooth_proxy:250]: [0] [<mac>] Connecting v3 with cache
[03:40:51][D][esp32_ble_tracker:216]: Pausing scan to make connection...
[03:40:51][I][esp32_ble_client:064]: [0] [<mac>] 0x00 Attempting BLE connection
```

This seems to refer to [this part of the btproxy code](https://github.com/esphome/esphome/blob/027284c29c4d2c119935f5d343bdb7ae6492502e/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp#L248). So I tried changing it so that a cached connection request is treated the same as a non-cached request.

```diff
248,249c248,249
<         connection->set_connection_type(espbt::ConnectionType::V3_WITH_CACHE);
<         ESP_LOGI(TAG, "[%d] [%s] Connecting v3 with cache", connection->get_connection_index(),
---
>         connection->set_connection_type(espbt::ConnectionType::V3_WITHOUT_CACHE);
>         ESP_LOGI(TAG, "[%d] [%s] Connecting v3 without cache", connection->get_connection_index(),
```

After rebuilding and updating my esp32 device, the integration works again. 

The fact that the connection cache seems to be involved neatly explains why reverting either esphome or HA seems to fix things, since I assume the cache flow only works if both are at version 2022.12 or later.

Unfortunately, I'm stuck at this point. I don't know enough about Bluetooth or the bt-proxy protocol or the Airthings BLE protocol to understand if the problem is:
1. esphome is incorrectly maintaining the connection cache, or
2. HA is incorrectly requesting a cached connection when it shouldn't, or
3. The `airthings_ble` integration specifically is improperly requesting a cached connection when it shouldn't

---

As a side note, simply disabling the `cache_services` option (without applying the patch above) doesn't seem to help.

```
bluetooth_proxy:
  active: true
  cache_services: false
```

Not sure if this means there are two caches or if there's a bug with this configuration option as well.

vqvu, 2023-01-13:
I spent a bit more time on this, and it seems to be an MTU problem. I added some extra logging to the underlying [airthings-ble](https://github.com/vincegio/airthings-ble) library and noticed that the data that was being received looked truncated.

Adding some logging to [bluetooth_connection.cpp#L143](https://github.com/esphome/esphome/blob/351ea045175c7f9fbd1afc3ff51cf1b61098a416/esphome/components/bluetooth_proxy/bluetooth_connection.cpp#L143) confirms that the esp32 only receives 20 bytes in the "bad" case and 30 bytes in the "good" case. After some searching, I found [an article](https://punchthrough.com/maximizing-ble-throughput-part-2-use-larger-att-mtu-2/), that says that the ATT MTU is 23 bytes, 20 of which is available to the payload---clearly not a coincidence.

Looking more into the esphome BLE client, I saw that [esp_ble_gattc_send_mtu_req](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gattc.html#_CPPv426esp_ble_gattc_send_mtu_req13esp_gatt_if_t8uint16_t) is [not called with the V3_WITH_CACHE connection type](https://github.com/esphome/esphome/blob/351ea045175c7f9fbd1afc3ff51cf1b61098a416/esphome/components/esp32_ble_client/ble_client_base.cpp#L140). That means the ATT MTU in use for the connection is the default of 23 bytes, which explains why only 20 bytes of data was sent.

Interestingly enough, it seems the Wave+ that will actually send the full 30 bytes across two packets. The `airthings-ble` library just can't handle that right now, because it [gives up after receiving the first packet](https://github.com/vincegio/airthings-ble/blob/543e804995b7e0e36ce15697c5f7dab1432caa57/airthings_ble/parser.py#L338).

So I think there is definitely a bug with the bluetooth client, since it doesn't attempt to negotiate an MTU at all in "cached" mode. There's also a bug in the `airthings-ble` library, since it should probably be able to handle notifications sent using the default MTU.

I confirmed the hypothesis above locally. Changing either `ble_client_base.cpp` to call `esp_ble_gattc_send_mtu_req` in cached mode or `airthings_ble` to wait until it has received all 30 bytes both fix the problem for me.

I'll try to find the time to send out a PR for `airthings-ble` soon. That one seems simple enough. But the fix to `ble_client_base` seems fairly complicated at first glance (I don't understand the state machine well enough to make changes), so I'll let @jesserockz take a look at that.

vqvu, 2023-01-13:
Oh, and the error messages that @Ulrar noticed in https://github.com/esphome/issues/issues/3951#issuecomment-1377060065 is because the `airthings-ble` library leaks its BLE connection when it crashes. That needs to be fixed as well.

bdraco, 2023-01-15:
Suggested fix in this issue https://github.com/esphome/issues/issues/4041

Can someone with the device try manually patching in that API call when the ble address matches?

Ulrar, 2023-01-16:
@bdraco : 

Unfortunately, I don't think it's working : 

```
2023-01-16 09:47:31.511 ERROR (MainThread) [homeassistant.components.airthings_ble] Error fetching airthings_ble data: Unable to fetch data: F4:60:77:74:E1:ED - F4:60:77:74:E1:ED: Failed to connect: 'APIClient' object has no attribute 'bluetooth_device_set_mtu'
```

bdraco, 2023-01-16:
> @bdraco :
>
>
>
> Unfortunately, I don't think it's working :
>
>
>
> ```
>
> 2023-01-16 09:47:31.511 ERROR (MainThread) [homeassistant.components.airthings_ble] Error fetching airthings_ble data: Unable to [...]

I was thinking about patching the esphome code for the specific device in the connection path

Ulrar, 2023-01-16:
Ah. I'm happy to try any patch but I don't know the esp internals, I wouldn't know where to start to make it myself

bdraco, 2023-01-16:
No worries. I ordered the devices listed in this thread.  They should be here in a week or two and I'll see if I can get it sorted out

mutlucan96, 2023-01-17:
Let me kindly remind everyone, this issue is originally not specific to airthings.

vqvu, 2023-01-17:
You're right. Now that we know more about the cause of the Airthings issue, I'm happy to move discussion about that to https://github.com/esphome/issues/issues/4041, which is more targeted to the MTU problem.

closed 2023-01-20 by jesserockz

mutlucan96, 2023-01-27:
The update didn't solve my issue. Again, the original problem and mine are nothing to do with Airthings.

jeroenhubgit, 2023-03-24:
I gave up hope for ESPHome.

Last year I only had problems with it. Not only with bluetooth. After certain updates most sensors are offline again or showing different results. Now I only use ESPhome for experiments, but not for long term reliable solutions. Maybe it will change in the future :).

Switching to OpenMqttgateway was the best solution for Bluetooth. You should try it. Once uploaded it always works with stable and good connection. For non-bluetooth solutions, Tasmota seems also more stable for me. Mostly it's more difficult, but at least it works.

Ulrar, 2023-03-24:
It's been rock solid for me since 2023.2, everything works pretty much instantly, very happy with it

jeroenhubgit, 2023-03-25:
> It's been rock solid for me since 2023.2, everything works pretty much instantly, very happy with it

Thats good to hear :). So I will still give it a try. But one month is not much, it mean stability over multiple months.

mutlucan96, 2023-03-25:
2023.2.4 and it's still dropping connection about every hour.

spiderman07, 2023-03-26:
mine is dropping every 24h(+/- 1h). power off and then power on fixes the issue for the next period.
Home Assistant 2023.3.6

Ulrar, 2023-03-26:
That used to be a problem for me pre esphome 2023.1.
I have 5 of them around the house now and haven't noticed any drops, I'm using doit-devkit-v1 boards in case that might be the difference maker