esphome/issues #3953: Magiquest Remote Receiver Protocol Change - Returning Incorrect Results

ChuckMac, 2022-12-21:
### The problem

When using a newer Magiquest wand the wand_id and magnitude values returned are random / not correct as the protocol has been updated to include a checksum.  wand_id should be static for each wand but returns a different value each time.

The upstream Magiquest IR remote receiver library I believe this was originally ported from was updated as the newer wands behave differently than previous iterations:

https://github.com/Arduino-IRremote/Arduino-IRremote/discussions/1027

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

remote_receiver/magiquest_protocol

### Example YAML snippet

```yaml
remote_receiver:
  id: rcvr
  pin:
    number: GPIO5
    inverted: true
    mode:
      input: true
  tolerance: 55%
  dump: 
    - raw
    - magiquest
```

joshuariney, 2023-11-28:
Same issue

jaypoc, 2024-04-30:
Recently visited Great Wolf Lodge and picked up the wand. I've managed to build an IR receiving unit with blinking lights and sound that reacts to the wand. Encased in a 3d printed crystal structure.. .then realized the Wand ID being decoded is the same as what I get from my TV Remote, and I also noticed at one point it was not 100% consistent.... The one wand usually returns ID 19667 (as does my TV remote) but sometimes varies. Would love to know if anyone is still working on this