esphome/issues #3955: Unable to compile .yaml

xilock, 2022-12-22:
### The problem

ESPHOME FOR Synology NAS's docker.Can't compile.
[image]
Nothing display
[image]
When click retry,display WARNING in terminal.

### Which version of ESPHome has the issue?

lastest

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

NodeMCU-Series

unhold, 2022-12-26:
What's the content of your YAML?

Does it compile on a desktop installation?

special-k-uk, 2023-03-10:
I'm having the exact same issue when trying to compile or view the device logs.

I'm trying to run ESPHome2022.11.5 in a docker container, as newer versions have an issue for some of my devices
https://github.com/esphome/issues/issues/3993

So the idea being a run a docker container for the older version just to support these devices, and then run the latest version on my HA installation for all other devices.

I've copied the entire contents of the "esphome" folder from HA instance into the container. All configs re visible on the front end. The problem arises when I try to view any device logs or to compile any of the yaml, I get the same 400 errors as @xilock 
[image]

and also get a blank screen when trying to compile
[image]

@unhold all my yaml files compile successfully when done from my HA installation (RPI HAOS)

I'm assuming it's a permissions issue but I'm not sure where to start looking

xilock, 2023-03-13:
> What's the content of your YAML?
> 
> Does it compile on a desktop installation?

Thank you for your reply
My YAML compiled and used in a virtual machine without any error.
Now I use ESPHOME in Oracle VM VirtualBox.