esphome/issues #3956: Web Server not working on AP mode when a WiFi network is configured

paviro, 2022-12-22:
### The problem

The web server does not seem to be started on AP mode when there is a WiFi network configured but out of reach. When not configuring a WiFi network and only AP mode the server is normally started.

My use case is a portable heater that can be used with home assistant when in reach of the network but can still be used with the build in web server if not. This does not seem possible right now and I would have to choose to either only allow home assistant access or only access via the build in AP.

As the web server does indeed work over the AP when no network is configured this seems like a bug and not like an intended limitation?

### Which version of ESPHome has the issue?

 v2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32

### Component causing the issue

Web Server

Conclusio, 2022-12-31:
Currently, I try similar things (with an ESP8266) and want to have access to the web server while in AP mode.

The main issue here is the setup priority of the web server.

According to `web_server.cpp` we have `float WebServer::get_setup_priority() const {return setup_priority::WIFI - 1.0f; }`. This means the web server is initialized **after** the WiFi connection is established.

In contrast to this, the `captive_portal` is configured as `setup_priority::WIFI + 1.0f` (i.e. before WiFi), since it needs to be available in AP mode as well. This might be a design choice or nobody thought about the web server being available in AP mode.

Anyway, changing the setup priority does resolve the issue, BUT there are other issues as well:

While in AP mode, the ESP tries to constantly scan for WiFi networks (if `fast_connect` is `false`), or just tries to connect directly (if `fast_connect` is `true`). Both variants introduce a delay which (for some reason) can interrupt the HTTP connection, meaning server-sent events are stopped from being received at the client (browser). In case of sensor readings, the browser would need to be refreshed constantly. For your portable heater setup this might not be a problem.

Related issue #3253

paviro, 2022-12-31:
Awesome thank you for the insights! I was looking through the code but couldn't quite piece that part together.

paviro, 2023-01-03:
@Conclusio this does not seem to work for me btw. I only get boot loops when changing the Webserver to ?`setup_priority::WIFI + 1.0f`
```
[C][web_server:099]: Setting up web server...

assert failed: tcpip_api_call IDF/components/lwip/lwip/src/api/tcpip.c:497 (Invalid mbox)

Backtrace:0x40083e45:0x3ffb23d00x4008ca29:0x3ffb23f0 0x4009237d:0x3ffb2410 0x40115221:0x3ffb2540 0x4016f798:0x3ffb2570 0x400f3efa:0x3ffb25c0 0x400e33b1:0x3ffb25e0 0x400e340a:0x3ffb2630 0x400e349e:0x3ffb2670 0x40181935:0x3ffb2690 0x401819e5:0x3ffb26b0 0x400e9c54:0x3ffb26d0 0x400ed526:0x3ffb2700 0x400fa56e:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:258]: Log initialized
[C][ota:469]: There have been 2 suspected unsuccessful boot attempts.
```

Conclusio, 2023-01-09:
This could be ESP32-specific, but honestly I don't know. The wifi component has platform-specific implementations (ESP32, ESP8266, ...).

If you don't configure a WiFi network at all does the same error occur?

brgaulin, 2023-01-10:
> If you don't configure a WiFi network at all does the same error occur?

In my case, thats exactly what it was. Once I removed the wifi config, it started working ok. This still sounds like a bug, but if it is a hardware limitation we should note that in the WebServer docs. Maybe near the `local` flag. Or in the Wifi docs if it is more global.

I was using a D1 mini.

Brausmith, 2023-01-23:
I don't have anything technical to add, but I have been experiencing the same today.
As soon as I removed the wifi credentials, the web UI via the AP started to work again.

Hiekkaharju, 2024-04-23:
> > With these I seem to have an (nearly) allways on webserver and HA updated with the sensors when the RV is in home wifi range. Sorry, I'm not making a pull request: 1) I don't know how. 2) my code is REALLY quick and dirty. 3) I don't want to publish code in specifically github.
> 
> How did you do that? I really need this too. I had to hang a router where there is no Internet and HA, just so that the device works normally at home and in the country. Tell me where and what is possible to fix. Unable to use because of these new (

Sorry, but I believe I kind of already did that in the above post. A good C++ coder should be able turn that list of changes to working code in rather short time, utilizing the references I made to Captive portal. I changed only three files, web_server.h, web_server.cpp and wifi_component.cpp.
On the other hand I believe that chances that someone who has not made C-something programs would be able to make those changes, and get code that compiles, with any reasonable amount of guidance, are low.
Rest are somewhere in between those.
And then there are the things related to, and multiple ways of, using non-release code when building an Esphome FW.
And the maintenance as Esphome is updated.
Also note that I do not know if the same would work with ESP8266.
Would someone be doing a PR for this, I'd be happy to answer questions about specific details, should a such person want e.g. confirmation on something.

ssieb, 2024-04-24:
Why do you have the captive portal configured if you don't want to use it?  Remove it from the config and the web server will work fine.

udavst, 2024-04-25:
My captive portal is turned off, but the web-server starts working only after connecting to the SSID, or by removing SSID from the firmware (

Hiekkaharju, 2024-04-25:
> Why do you have the captive portal configured if you don't want to use it? Remove it from the config and the web server will work fine.

That is not true. Read the explanations.

ssieb, 2024-04-25:
> My captive portal is turned off, but the web-server starts working only after connecting to the SSID, or by removing SSID from the firmware

I missed that part.

ssieb, 2024-04-25:
Try that PR.

closed 2024-05-07 by ssieb