esphome/issues #3957: Fix compile error for uptime sensor with esp-idf - still there

sanderlv, 2022-12-23:
### The problem

This issue is still there now:
https://github.com/esphome/esphome-docs/pull/1534

```
Processing shelly-plus-2pm-01 (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/shelly-plus-2pm-01/.pioenvs/shelly-plus-2pm-01/src/main.o
/config/esphome/shelly-plus-2pm-01.yaml: In lambda function:
/config/esphome/shelly-plus-2pm-01.yaml:176:17: error: 'String' was not declared in this scope
                 (days ? String(days) + "d " : "") +
                 ^~~~~~
/config/esphome/shelly-plus-2pm-01.yaml:176:17: note: suggested alternative: 'trunc'
                 (days ? String(days) + "d " : "") +
                 ^~~~~~
                 trunc
*** [/data/shelly-plus-2pm-01/.pioenvs/shelly-plus-2pm-01/src/main.o] Error 1
========================= [FAILED] Took 12.29 seconds =========================
```

### Which version of ESPHome has the issue?

 2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32-IDF

### Board

Shelly plus 2pm single core

### Example YAML snippet

```yaml
Compiling /data/shelly-plus-2pm-01/.pioenvs/shelly-plus-2pm-01/src/main.o
/config/esphome/shelly-plus-2pm-01.yaml: In lambda function:
/config/esphome/shelly-plus-2pm-01.yaml:176:17: error: 'String' was not declared in this scope
                 (days ? String(days) + "d " : "") +
                 ^~~~~~
/config/esphome/shelly-plus-2pm-01.yaml:176:17: note: suggested alternative: 'trunc'
                 (days ? String(days) + "d " : "") +
                 ^~~~~~
                 trunc
```

sanderlv, 2022-12-23:
An other problem is I cannot add this:

```
text_sensor:
  - platform: template
    name: Uptime
    id: uptime_human
    icon: mdi:clock-start
  - platform: wifi_info
    ip_address:
      name: "${esp_name} IP Address"
    ssid:
      name: "${esp_name} Connected SSID"
    bssid:
      name: "${esp_name} Connected BSSID"
    mac_address:
      name: "${esp_name} Mac Wifi Address"
  - platform: version
    name: "${esp_name} - ESPHome Version"
    
switch:
  - platform: restart
    name: ${esp_name} Restart
```

Gives me a huge load of errors in compiling

jesserockz, 2022-12-23:
`String` is an Arduino class, you are using ESP-IDF so `String` does not exist. 
You linked to a merged PR, but you didnt even change you local YAML to match.

closed 2022-12-23 by jesserockz

sanderlv, 2022-12-23:
Jesse, thank you. Sorry I am technical but maybe not enough. What does this answer mean? It is just not possible or do I need to do something else/change to get things working?

What do you mean with: You linked to a merged PR, but you didnt even change you local YAML to match.

sanderlv, 2022-12-23:
Think I needed to change this (I think):
Fixed by switching to to_string(...) instead of using String(...).

But now before even getting there I suddenly have this:
[image]