esphome/issues #3958: SSL Error during compile due to corporate proxy cannot be fixed due to lack of information

fkkjohn, 2022-12-23:
### The problem

This bug is a lack of any identifying information on where these files are being downloaded from (server), and what transport system is being used to download them (pip, git, whatever). I get a useless path on who knows whose computer, and the name of the package. What certificate am I supposed to install? Which server am I supposed to trust? I'm beyond frustrated with this.

Here's what I get from the command line:

`Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40201.0
INFO Installing platformio/tool-esptoolpy @ ~1.40201.0
WARNING Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
WARNING Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:992)'))': /v3/packages/platformio/tool/tool-esptoolpy
Error:`

This is an unacceptable level of detail. There are zero results from web searches regarding this, and nothing I can do. Anyone, please point out how to get around this. I hope it's staring me in the face and I look like an idiot so I can fix it and get on with my life.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

DAVe3283, 2023-02-03:
You probably just need to add your corporate root CA to the appropriate docker containers. Here is a script I use for that purpose, though we don't proxy all traffic, only some stuff, so I can't guarantee it will fix your issue.

Place your corporate root CA in Home Assistant's /ssl/ directory, call it whatever you want, just update the script below to the right name.

```bash
#!/usr/bin/env bash

containers=(addon_5c53de3b_esphome addon_a0d7b954_vscode)
cert_src=/ssl/root.crt
cert_dst=/usr/local/share/ca-certificates/CorporateDomainNameHere.crt

for container in "${containers[@]}"; do
	echo Adding ${cert_src} to ${container}:${cert_dst}
	docker exec -it ${container} bash -c "ln -fs ${cert_src} ${cert_dst} && update-ca-certificates"
	echo
done
```

You will have to run this script from the SSH & Web Terminal add-on, and you must disable `Protection mode` for that add-on.

ssieb, 2023-02-03:
Any proxy that requires you to use their CA cert is inherently broken.
But also, for the initial poster, that's all from platformio and nothing that esphome can do anything about.

closed 2023-02-03 by ssieb