esphome/issues #3959: ESPHome wont compile or update esp8266

Jarled, 2022-12-23:
### The problem

ESPHome wont compile or update my WeMos D1 Mini ESP8266
ESPHome shows blank black validate, install, and log window for device
 i get the below error in the logs.

```
2022-12-23 14:54:45,065 ERROR Uncaught exception GET /info?configuration=esphome-web-5509cb.yaml (0.0.0.0)
HTTPServerRequest(protocol='http', host='192.168.0.5:8123', method='GET', uri='/info?configuration=esphome-web-5509cb.yaml', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1711, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 133, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 158, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 807, in get
    self.write(DashboardEntry(yaml_path).storage.to_json())
AttributeError: 'NoneType' object has no attribute 'to_json'
2022-12-23 14:54:45,073 ERROR 500 GET /info?configuration=esphome-web-5509cb.yaml (0.0.0.0) 9.65ms
```
I have reimaged my machine with fresh install of HA OS and same error message straight away, please help.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP8266

### Board

WeMos D1 Mini ESP8266

### Anything in the logs that might be useful for us?

```txt
Home Assistant 2022.12.8
Supervisor 2022.11.2
Operating System 9.4
Frontend 20221213.1 - latest
## System Information

version | core-2022.12.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.80
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.29.0
Stage | running
Available Repositories | 1198
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 28.6 GB
disk_used | 3.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2022.12.3), Samba share (10.0.0), SSH & Web Terminal (13.0.0), File editor (5.4.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 3 December 2022 at 20:09
-- | --
current_recorder_run | 23 December 2022 at 14:10
estimated_db_size | 14.38 MiB
database_engine | sqlite
database_version | 3.38.5

</details>
```

ChrisItIs0, 2023-03-31:
I am having the same issue. Did you ever find a solution to this problem?

Jarled, 2023-03-31:
> I am having the same issue. Did you ever find a solution to this problem?

I ended up having to install Docker, install ESPhome on docker then was able to flash it using CLI.

ChrisItIs0, 2023-03-31:
> 
Interesting, I have ESPhome running in Docker, sometimes it works on the web dashboard and sometimes it doesn't.

ChrisItIs0, 2023-05-02:
> 
For me, the problem was that I didn't enable Websocket support in my reverse proxy. If you're using one too, you should defenetly check that. Since enabling it, it works with no problems