esphome/issues #3961: esp-idf does not build on windows

evlo, 2022-12-24:
### The problem

I have tried on multiple win10 and win11 computers, python 3.11.1, wheels installed, 
result is always what is in screenshot bellow
I think it is not permissions, but filename issue

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

t01-c3,esp32-cam

### Component causing the issue

pyYaml?

### Example YAML snippet

```yaml
esphome:
  name: ehidfdemo
esp32:
  board: esp32cam
  framework: 
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
```

### Anything in the logs that might be useful for us?

```txt
![image](https://user-images.githubusercontent.com/8813196/209432055-956e8402-02ea-4951-92a1-fae12f90648e.png)
```

### Additional information

maybe not exactly issue with esphome, but something in the build stuff it is using python, wheels, pip, some library, framework, etc.

djanknecht, 2023-01-06:
Hi @all, same issue here.

I'm running Windows 1122H2. 
I set up a new project from the scratch witout any sensors.

What i tried so far:
- Reinstalled Python, wheel, esphome (no change if using Py 3.10 nor 3.11)
- Tried different installation directories (installed for all users or directly in c:\Phyton3.X)
- Deleted all fragments of cached Phyton libraries

Compile using Arduino Framework works without any issues.

Verbose Log:
 
[verbose-log.txt](https://github.com/esphome/issues/files/10358099/verbose-log.txt)

Other Logs, project directory and the project file:
[CMakeOutput.log](https://github.com/esphome/issues/files/10358038/CMakeOutput.log)
[dot.esphome.zip](https://github.com/esphome/issues/files/10358057/dot.esphome.zip)
[test.yml.txt](https://github.com/esphome/issues/files/10358060/test.yml.txt)

Hope someone is able to find the issue.

Kind regards,

Dirk

djanknecht, 2023-01-06:
I think i found a workaround.
After calling 
"pip install -U --force-reinstall "cryptography>=2.1.4,<35.0.0" "future>=0.15.2" "idf-component-manager~=1.0"
manually once it works!

In my eyes, we have something like a circular reference here - during compile, pyyaml is needed, but the script tries to uninstall it.

eerorossi, 2023-01-13:
> I think i found a workaround. After calling "pip install -U --force-reinstall "cryptography>=2.1.4,<35.0.0" "future>=0.15.2" "idf-component-manager~=1.0" manually once it works!
>
> In my eyes, we have [...]

This works for me too.

marshn, 2023-01-17:
> I think i found a workaround. After calling "pip install -U --force-reinstall "cryptography>=2.1.4,<35.0.0" "future>=0.15.2" "idf-component-manager~=1.0" manually once it works!
>
> In my eyes, we have [...]

Using v2023.12.3, Windows 10, PIP and ESP32-EDF I had a very similar problem (I think something to do with the '~' characters in the path) during a reinstall of the charset_normalizer package which was happening with every build. In my case just running this was enough to stop the reinstall loop:

pip install -U --force-reinstall "idf-component-manager~=1.0" 

Problem didn't happen with 2022.11.5, but did when I tried 2022.12.1 and 2022.12.2.

insurgent, 2023-03-15:
> 
> pip install -U --force-reinstall "idf-component-manager~=1.0"
> 

This worked for me - thanks!

Sorjak, 2023-05-24:
Took me too long to find this thread while attempting to build firmware for my ESP32-S3-BOX, but @djanknecht's answer fixed it for me. I'm going to copy down the error text here in case it makes this easier to find:

ERROR: Could not install packages due to an OSError: [WinError 5] Access is denied: 'C:\\Users\\uli\\projects\\esphome\\voice_assistant\\.venv\\Lib\\site-packages\\~aml\\_yaml.cp311-win_amd64.pyd'