esphome/issues #3962: Locally pressing a button does not update its state

chatziko, 2022-12-24:
### The problem

The [state of a button](https://www.home-assistant.io/integrations/button/#the-state-of-a-button) in HA contains the date it was last pressed.

Using a template button in esphome:
- when the button is pressed __from the UI (or via a service call)__, the button's state is updated with the current date
- but when the button is pressed __locally__ in the ESP node, via [button.press](https://esphome.io/components/button/index.html#button-press-action), its state is not updated.

To be consistent, and allow the state updates to trigger automatios, the state should also be updated when the button is pressed locally.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
button:
- platform: template
  name: "my_button"
  id: _button
  on_press:
    - logger.log: "Pressed"

binary_sensor:
- platform: gpio
  pin:
    number: GPIO14
    mode: INPUT_PULLUP
    inverted: true
  on_press:
    button.press: _button   # press the button locally
```

mattster98, 2023-02-07:
@chatziko - Did you find a workaround for this, or just choose a different component to convey this info?  I have reproduced this - just trying to send a button press from ESPHome to Home Assistant.  Tried calling it in a lambda as well as with just button.press

chatziko, 2023-02-07:
I just send an event when the button is pressed and catch the event on the HA side. A bit less elegant, but it works.

```
button:
- platform: template
  on_press:
    - homeassistant.event:
        event: esphome.my_button.pressed
```

mattster98, 2023-02-07:
Nice.  I reached out on the ESPHome discord and they pointed out what I missed in the docs that the Button component is essentially read-only (only a trigger FROM HA), not something you can push TO HA.    I pivoted to binary sensor which will also work