esphome/issues #3968: GPIO output not working

kiasarecool, 2022-12-26:
### The problem

I am building a 2nd bang bang thermostat to replace another; used exact same config only changing entity names and when installed the gpio pins do not respond, tried every available pin on and all remain high besides d7 on boards  d1mini 8266 & nodemcu v3 8266
also tried ob esp3- wroom-32d and same result

After many unsuccessful attempts (using the same hardware as before when failed and same pin) I uploaded a firmware I created on 12-11-22 to a d1 mini and the gpio pins trigger a relay as expected.

My HA is one core update behind current, but esphome is up to date

### Which version of ESPHome has the issue?

13.0.2 (current) 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.7

### What platform are you using?

ESP8266

### Board

Nodemcu v3 8266

### Component causing the issue

GPIO

### Example YAML snippet

```yaml
failing yaml 
esphome:
  name: relay-temp-4

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: x

ota:
  password: x

wifi:
  ssid: "Plant Factory ATT "
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Relay-Temp-4"
    password: ""

captive_portal:
    
web_server:
  port: 80
  include_internal: true 

time:
  - platform: homeassistant
    id: ntp

switch:
  - platform: gpio
    name: "Heater GH2W"
    id: "relay2"
    pin: 13

i2c:
    sda: 4
    scl: 5
    scan: true
sensor:
  - platform: bme280
    temperature:
      name: "Temperature GH2W"
      id: temp
      oversampling: 16x
    pressure:
      name: "Pressure GH2W"
    humidity:
      name: "Humidity GH2W"
      oversampling: 8x
    address: 0x76
    update_interval: 10s

climate:
  - platform: bang_bang
    name: "GH2w Thermostat"
    sensor: temp
    default_target_temperature_low: 61 °F
    default_target_temperature_high: 64 °F

    heat_action:
      - switch.turn_on: relay2
    idle_action:
      - switch.turn_off: relay2

WORKING YAML (same yaml? just created/ installed weeks apart)
esphome:
  name: relay-temp-4

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "8Gp5TRLrzWBn0o1PM+DH7K/GE5lZpyk2jalq0uiZ15U="

ota:
  password: "ae236c6dc9d694ea5aec8f919ea99344"

wifi:
  ssid: "Plant Factory ATT "
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Relay-Temp-4"
    password: ""

captive_portal:
    
web_server:
  port: 80
  include_internal: true 

time:
  - platform: homeassistant
    id: ntp

switch:
  - platform: gpio
    name: "Heater GH2W"
    id: "relay2"
    pin: 13

i2c:
    sda: 4
    scl: 5
    scan: true
sensor:
  - platform: bme280
    temperature:
      name: "Temperature GH2W"
      id: temp
      oversampling: 16x
    pressure:
      name: "Pressure GH2W"
    humidity:
      name: "Humidity GH2W"
      oversampling: 8x
    address: 0x76
    update_interval: 10s

climate:
  - platform: bang_bang
    name: "GH2w Thermostat"
    sensor: temp
    default_target_temperature_low: 61 °F
    default_target_temperature_high: 64 °F

    heat_action:
      - switch.turn_on: relay2
    idle_action:
      - switch.turn_off: relay2
```

### Anything in the logs that might be useful for us?

```txt
No errors
```

### Additional information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.15.76-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "openmediavault": {
      "version": "1.3.0",
      "requirements": []
    },
    "samsungtv_tizen": {
      "version": "1.6.1",
      "requirements": [
        "websocket-client>=0.56.0",
        "wakeonlan>=2.0.0",
        "numpy>=1.19.2"
      ]
    },
    "average": {
      "version": "2.3.0",
      "requirements": []
    },
    "wyzeapi": {
      "version": "0.1.15",
      "requirements": [
        "wyzeapy==0.5.14"
      ]
    },
    "ui_lovelace_minimalist": {
      "version": "v1.1.6-hotfix",
      "requirements": [
        "aiofiles==0.8.0",
        "aiogithubapi>=22.2.4"
      ]
    },
    "hacs": {
      "version": "1.29.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "esphome",
    "name": "ESPHome",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/esphome",
    "requirements": [
      "aioesphomeapi==13.0.2"
    ],
    "zeroconf": [
      "_esphomelib._tcp.local."
    ],
    "dhcp": [
      {
        "registered_devices": true
      }
    ],
    "codeowners": [
      "@OttoWinter",
      "@jesserockz"
    ],
    "after_dependencies": [
      "bluetooth",
      "zeroconf",
      "tag"
    ],
    "iot_class": "local_push",
    "integration_type": "device",
    "loggers": [
      "aioesphomeapi",
      "noiseprotocol"
    ],
    "is_built_in": true
  },
  "data": {
    "config": {
      "entry_id": "cf92a55cf8b4986bcc180e367d29bc05",
      "version": 1,
      "domain": "esphome",
      "title": "esp-bme280-1",
      "data": {
        "host": "192.168.68.56",
        "port": 6053,
        "password": "",
        "noise_psk": ""
      },
      "options": {},
      "pref_disable_new_entities": false,
      "pref_disable_polling": false,
      "source": "zeroconf",
      "unique_id": "esp-bme280-1",
      "disabled_by": null
    },
    "storage_data": {
      "device_info": {
        "uses_password": false,
        "name": "esp-bme280-1",
        "mac_address": "**REDACTED**",
        "compilation_time": "Dec  4 2022, 22:55:44",
        "model": "esp01_1m",
        "manufacturer": "Espressif",
        "has_deep_sleep": false,
        "esphome_version": "2022.11.4",
        "project_name": "",
        "project_version": "",
        "webserver_port": 0,
        "bluetooth_proxy_version": 0
      },
      "services": [],
      "api_version": {
        "major": 1,
        "minor": 7
      },
      "sensor": [
        {
          "object_id": "bme280-1_temperature",
          "key": 1119871574,
          "name": "BME280-1 Temperature",
          "unique_id": "esp-bme280-1sensorbme280-1_temperature",
          "disabled_by_default": false,
          "icon": "",
          "entity_category": 0,
          "device_class": "temperature",
          "unit_of_measurement": "\u00b0C",
          "accuracy_decimals": 1,
          "force_update": false,
          "state_class": 1,
          "last_reset_type": 0
        },
        {
          "object_id": "bme280-1_pressure",
          "key": 2587786769,
          "name": "BME280-1 Pressure",
          "unique_id": "esp-bme280-1sensorbme280-1_pressure",
          "disabled_by_default": false,
          "icon": "",
          "entity_category": 0,
          "device_class": "pressure",
          "unit_of_measurement": "hPa",
          "accuracy_decimals": 1,
          "force_update": false,
          "state_class": 1,
          "last_reset_type": 0
        },
        {
          "object_id": "bme280-1_humidity",
          "key": 3626762559,
          "name": "BME280-1 Humidity",
          "unique_id": "esp-bme280-1sensorbme280-1_humidity",
          "disabled_by_default": false,
          "icon": "",
          "entity_category": 0,
          "device_class": "humidity",
          "unit_of_measurement": "%",
          "accuracy_decimals": 1,
          "force_update": false,
          "state_class": 1,
          "last_reset_type": 0
        }
      ],
      "text_sensor": []
    }
  }
}

kiasarecool, 2022-12-26:
Issue Resolved........
There was a esp board plugged into my tower I was not aware of..... This is embarrassing.

unhold, 2022-12-26:
Happens to the best.
You can click close below to make this issue disappear.

closed 2022-12-27 by kiasarecool